Avnu/OpenAvnu #500: simple talker and listener connect with avb switch

stujeffrey016, 2016-10-14:
Hi guys,
I tried to do some experience on my avb switch, and there is some questions i want to ask.
Did anyone try this structure?
[ubuntu 14.04 simple talker intel i210] <------> AVB switch <------> [intel i210 ubuntu 14.04 simple listener ]
In my case gptp was working, but listener didn't receive any audio frame. 
I want to know did i miss anything? 
Or simple talker and listener can't work in this situation?
[image]

andrew-elder, 2016-10-14:
Am I correct in assuming 2 physical i210s are residing the same PC?

stujeffrey016, 2016-10-14:
Hi Andrew,
No, two I210 on two different PCs.

eric-mann, 2016-10-14:
Two dumb questions:
a) which tag/branch are you using?
b) how do you know you aren't receiving packets?

stujeffrey016, 2016-10-15:
Hi Eric,
a) I'm using Open-AVB-Next.
b) Because my listener didn't save the .wav file.
I thought my problem is that when Listener receive the request packet from Talker, it return the indication packet back. 
And when switch receive the indication message from Listener, it change some information in the indication packet, caused the Talker couldn't handle the indication packet.
I guess that's the reason why my talker terminal display "unhandled from mrpd".
I'll use wireshark to catch the packet and analysis the problem in this situation.

eric-mann, 2016-10-16:
Could you add the output after enabling 'DEBUG' in simple_listener. 

Also beware - you won't see packets with wireshark on the talker side since simple_talker bypasses the network stack and goes directly to the HW. With simple_listener, you should be able to see packets with wireshark.

pinealservo, 2016-10-17:
Some of the "unhandled" messages are just because they're related to attributes that simple_talker/simple_listener don't care about, like MVRP stuff.  It looks a bit suspicious that you typed a ^C the talker to get it to make progress; it says it found a listener, but the listener doesn't appear to have actually sent a listener ready, so I'm guessing a reservation wasn't actually established and the switch is therefore dropping your streaming packets on the floor.

Sending ^C to either the talker or listener is probably going to break things; don't do that unless you're trying to just get the talker to spew packets over a non-AVB network. If things aren't working after you start the talker and listener, try to diagnose what's going on without hitting ^C. Check your switch's AVB reporting to see whether both endpoints have successfully joined the SRP domain and whether there are active and matching talker/listener registrations on the endpoint ports. You should also be able to use the mrp client command line tool to see what your endpoints have declared locally.

stujeffrey016, 2016-10-18:
Hi Eric,
What do you me add the output after enabling 'DEBUG'?

stujeffrey016, 2016-10-18:
Hi pinealservo,

Thanks for your apply!
These informations are very helpful.
In the earlier case, I only used Simple_talker and Simple_listener without Switch as my platform; and I found out that they couldn't always connect every time, so I type ^C tried to end the process, but somehow my Simple_Talker said "got a Listener"; and I thought this situation is a common bug.
I'll find out the problem first and do the Switch experiment.
Thank again!

Regards
Jeffrey.

pinealservo, 2016-10-18:
In case it helps, I'll list out the interactions that are supposed to take place when you start simple_talker and simple_listener:

1. AVB bridge sends SRP Domain declaration to endpoints
2. Endpoints register SRP Domain declaration and declare the same attribute in return.
3. AVB bridge registers the endpoints' SRP Domain declarations and marks them as part of the SRP Domain
4. The talker declares a SRP Talker Advertise attribute
5. The AVB bridge registers the SRP Talker Advertise attribute from the talker endpoint's port and declares SRP Talker Advertise on the listener endpoint's port.
6. The listener registers the SRP Talker Advertise and declares a SRP Listener Ready attribute for the same stream.
7. The AVB bridge registers the SRP Listener Ready attribute on the listener endpoint's port. It then matches the stream ID with the established SRP Talker Advertise. It determines that there's enough bandwidth to make the reservation and sets up its forwarding tables for it. It declares the SRP Listener Ready on the talker endpoint's port.
8. The talker registers the SRP Listener Ready attribute, indicating that the reservation for a listener has been established.
9. The talker starts transmitting its stream
10. The AVB bridge forwards the stream from the talker port to the listener port thanks to the forwarding table setup from step 7.
11. The listener starts receiving the stream.

I've left out the steps involving ensuring that both talker and listener have joined the VLAN that the talker is using for the stream; I don't remember whether they're part of what simple_talker/simple_listener are programmed to do right now, but if everything else seems okay and you're still not getting the traffic through, you might check that they're not dropped due to VLAN misconfiguration.

You should be able to either peek the SRP traffic or monitor the SRP attributes in the switch to ensure that the correct interactions are happening there.

stujeffrey016, 2016-10-19:
Hi pinealservo,
I have another problem now. I tried to build the simple_talker/listener without switch on two PCs with intel I210 as my platform.
At first, I ran the gptp and mrpd program on both simple_talker/listener side; after that I ran the simple_listener(first) and simple_talker(second), but they show same message "Msg: MSRP:Empty" on both terminal. I want to know what's wrong in this situation. 

These picture show the message which display on talker and listener.

Talker
[image]

Listener
[image]

Regards
Jeffrey.

eric-mann, 2016-10-19:
Do you have more than one Ethernet interface in either the talker or listener sides?

stujeffrey016, 2016-10-19:
Hi Eric,
Yes, will it affect this platform?
I have an interface use to go on online, and the other use to do the avb experiment.

Regards
Jeffrey.

eric-mann, 2016-10-19:
I suspect that simple_talker is selecting ‘the wrong interface’ … it is a common problem if you have multiple I210’s installed for example.

stujeffrey016, 2016-10-19:
Hi Eric, 

Thanks for your apply, I just disconnect my online Ethernet interface and tried the experiment again, but it still couldn't work.
I thought this isn't the problem. Because I type the interface eth1 in my command. I think Talker won't choose the wrong interface.

Regards
Jeffrey.

eric-mann, 2016-10-19:
It isn’t that easy … inside of simple_talker is a ‘pci_connect()’ function which looks up all of the installed i210s and picks the 1st one it finds – based on PCI bus/device/function information - to use for AVB. This may or may not correspond to the ‘eth1’ or ‘eth0’ you specify.

There’s a proposed fix to use an ethtool IOTCL on the ethX interface to return the B/D/F information instead of using libpci … but no one has had the bandwidth to make the change.

stujeffrey016, 2016-10-19:
Hi Eric,
Thanks for your information. In my case, I didn't use I210 as my online Ethernet interface. It means that I only have one I210 card in each PC. Did 'pci_connect()' still picks the wrong interface in this situation?
If I remove the online ethernet interface, can I fix this problem?
Regards
Jeffrey.

dmlarson4, 2016-10-20:
I'd like to piggyback on this thread since I have a similar issue. 

I've been successful in sending messages with simple_talker and simple_rx point to point. My next step is connecting the two through a Netgear GS716T switch. PC1 and PC2 are both running CentOS7 with an i210 NIC. 

When I connect the two, I get a similar output similar to Jeffrey's. I've checked the switch logs and the issue is with MSRP. The switch stores MSRP statistics which shows that over three quarters of frames received are not formatted correctly. Other logs show a consistent "msrpPduRead failed" message.

I’m not sure if this is an issue with the configuration of the Netgear switch or something else. I’ve tried to find documentation online to hand check the packets sent from the PC to the switch. 

I guess my questions are:
1.	What switch configuration parameters do I need to be aware of for AVB?
2.	Is there documentation out there that specifies structures of MSRP messages field-by-field? 

From wireshark:
[image]

Regards,
dmlarson4

pinealservo, 2016-10-21:
When connecting two endpoints back-to-back, unfortunately the simple_talker/simple_listener behavior of waiting for a domain attribute registration won't work. They both wait for the other to declare it first, and so nothing happens. In this case, since there's no reservation necessary, it's okay to ^C them to move along to actual streaming.

The MSRP packets that include all attribute types are the periodic leave-all packets that MRP requires. There's no useful information in them; they just prompt the peer to re-declare any attributes it wants to keep alive.

The SRP protocol is specified in 802.1Q, as is the MRP protocol that it is based on. It is available for free from the IEEE site. For an easier introduction to SRP, see the AVnu document here: http://avnu.org/wp-content/uploads/2014/05/AVnu_Stream-Reservation-Protocol-v1.pdf

stujeffrey016, 2016-10-31:
Hi guys,
I change the packet data and my switch topology work.
thanks for your helping.

andrew-elder, 2016-10-31:
@stujeffrey016 - I'm glad you got it working. Can you provide some additional details so that the next person who runs in to this issue has a clearer idea of the path to a solution. It would be much appreciated.

stujeffrey016, 2016-10-31:
I change the LLDP information, and make switch can forward the msrp packet. After that the listener got advertise and send the ready packet. 
In the earlier experiment, I find that talker send 1722 packet in every situation, no matter it receive the ready packet or not. I'm very confused with this issue now.

closed 2017-04-13 by stujeffrey016