Avnu/OpenAvnu #502: cpputest - c++11 updates causes Visual Studio issues

andrew-elder, 2016-10-14:
Updates to cmake config files cause compile errors in Visual Studio 2013 and 2015.

Update to the lastest code on cpputest master branch seems to fix the issue.
Do we want to do that?

christopher-s-hall, 2016-10-14:
I vote yes

pinealservo, 2016-10-17:
I don't have a problem with it in principle; it's probably best to keep that updated occasionally anyway. The discussion in the issue that prompted the cmake config change suggested that updating cpputest caused some other problem on the systems that were having issues without the cmake change, though. I couldn't reproduce it, but I couldn't reproduce the one I was trying to fix either.

Out of curiosity, what compile errors are you getting in Visual Studio? Is it being pickier about some new C++ stuff than gcc is?

andrew-elder, 2016-10-17:
CMAKE output of the issue (during generate step) looks like the below...
```
Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE) 
Try C++11 flag = [ ]
Performing Test CXX11_FLAG_DETECTED
Performing Test CXX11_FLAG_DETECTED - Failed
Try C++11 flag = [-std=c++11]
Performing Test CXX11_FLAG_DETECTED
Performing Test CXX11_FLAG_DETECTED - Failed
Try C++11 flag = [/Qstd=c++0x]
Performing Test CXX11_FLAG_DETECTED
Performing Test CXX11_FLAG_DETECTED - Failed
CMake Error at C:/Program Files/CMake/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find CXX11 (missing: CXX11_FLAGS)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  thirdparty/cpputest/cmake/Modules/FindCXX11.cmake:48 (find_package_handle_standard_args)
  thirdparty/cpputest/cmake/Modules/CppUTestConfigurationOptions.cmake:48 (find_package)
  thirdparty/cpputest/CMakeLists.txt:32 (include)

Configuring incomplete, errors occurred!
See also "C:/build/oavb/CMakeFiles/CMakeOutput.log".
See also "C:/build/oavb/CMakeFiles/CMakeError.log".
```

pinealservo, 2016-10-17:
Apparently that test is part of cpputest's repo and its visual studio support for detecting C++11 was busted. See https://github.com/cpputest/cpputest/commit/80cf68c993704fcc9df1b5f6dd27ef6bb9016824 for the fix they added.

So, I guess upgrading cpputest is the only reasonable way forward, but we should be able to keep the C++11 enabling in place.