Avnu/OpenAvnu #505: Ethernet AVB- Listener failed on 82579IM based controller

SwapnaAnoopkumar, 2016-10-21:
I am trying to create an Ethernet AVB demo setup with the following configuration - source code taken from Open AVB in Github

Talker running in a PC, with Intel i210 - Linux 14.04
Listener running in a PC with Intel 82579 LM based Network Controller - Linux 16.04
I am able to build the code for talker and listener and talker application is running.

The listener application terminates after receiving a few packets from talker. It seems the driver used in listener is not AVB compatible. We used the e1000e driver, since we found that the intel card used in the listener supports IEEE 802.1AS and IEEE 802.1Qav protocols which are the bare minimum support needed in a listener.

Can anybody help me to resolve the issue? Should I get another driver or is the network card not expected to work in AVB?

pinealservo, 2016-10-21:
Currently only the i210 is supported with the existing OpenAVB code. It should be entirely possible to create a listener for any NIC that supports gPTP, but the codebase currently assumes the OpenAVB igb driver and igb_lib are used. They use a "backdoor" approach to manage the descriptor rings used for stream transmit and receive in userspace, bypassing the kernel networking stack entirely.

closed 2016-10-21 by pinealservo

SwapnaAnoopkumar, 2016-10-25:
Thanks for the response.
We will proceed with i210 card in the listener side also.