Avnu/OpenAvnu #506: problem live stream listener and talker

zerbo88, 2016-10-21:
hi, I have two intel i210T two debian pc, in the "live stream test" with two PCs connected to each other everything works pretty well, but when I connect the i210 to the switch AVB dsp4you the listener pipeline remains in constant prerolling, and the GPTP not work, goes on to say "not AsCapable"

someone can help me?

pinealservo, 2016-10-21:
Please supply us some more information about your configuration:

1. Which version of Open-AVB are you using and which components? Please include the branch you checked out along with the commands you're issuing to start it.
2. What are the exact errors that you get? Is there more information about why gPTP issues "not AsCapable"?

zerbo88, 2016-10-25:
I'm using Opan_avb master branch, my real problem is when i connect the two debian pc with i210 to the switch, if they connect to each other and the connection examples are successful.
I'm using the live_stream test.
I wanted to know if altrri have had the same problem

pinealservo, 2016-10-25:
You haven't given enough information for anyone to say whether they have had the same problem or not. I don't recall anyone reporting problems with the live_stream example, but I have never tried it myself, so I'm not sure what issues it might have. I don't remember anyone reporting anything about the dsp4you switch either, but I don't have one so I can't do any testing with it.

From your initial report, it sounds like the problem is due to gPTP not working, but you didn't include any more information to help diagnose that. In order for AVB traffic to get through the switch, you'll have to get gPTP and SRP working properly with a stream reserved for at least the amount of bandwidth you're using for your live_stream configuration. In a back-to-back configuration, live_stream's talker and listener don't care about gPTP, and I don't think mrpd does either.

wgoossens, 2016-12-28:
Hi zerbo88,

I have experienced (and commented about) this exact issue about a year ago with the then current OpenAVB version. I never got gPTP stable. I did _some_ investigations then and found (using wireshark) that there was a big mismatch between the number of pdelay requests and responses. This could have something to do with either operating at 100m or something with the AVB4YOU switch.
The xmos modules (at 100m) work well with the AVB4YOU switch but the i210 never worked for me using that switch.
Using a LabX  Titanium-411 switch or direct connected crosscable works perfectly well for me so I'm not 100% sure the error is with the OpenAVB stuff.
I don't have access to any other 100Mb avb switches so I can't compare. We have Arista Networks and Extreme Networks switches at work but they are 10G/1G only.
I haven't really found any continued development (after the june-2015 update) of the AVB4YOU switch (they had some "future use" ports on it) so I'm afraid it's a bit of a dead-end.
Hope this helps anyone further.

Walter Goossens