Avnu/OpenAvnu #507: Enabling hardware time stamping in Kernel

SwapnaAnoopkumar, 2016-10-24:
Hi,

Do I need to recompile the kernel enabling hardware timestamp support (PTP) to work with Intel i210 nic ?

I am trying to set up demo of Live stream talker application. I have i210 card which is running on Ubuntu 14.04 (Kernel version - 3.19.0-69-generic) with latest openAVB source downloaded from git.

Followed the openAVB documentation and build igb_avb driver and is successfully loaded and ptpd is started.

Please comment...

Thanks & Regards,
Swapna

pinealservo, 2016-10-25:
The default Ubuntu 14.04 kernel should have PTP support available without recompiling. The gPTP daemon should work both with both the igb driver shipping with the kernel and the igb_avb driver.

Please give a much more detailed report of what errors you are seeing. Not just a description, but the commands you're issuing and the output they produce. Also, describe the specific hardware you're using besides the i210 and Ubuntu PC.

SwapnaAnoopkumar, 2016-10-26:
Thanks for your response.
Currently we have only one i210 card in which we were setting the talker application. Hence the demo is not completely functional. 
The query is posted in view of setting up the complete system so that we could directly proceed with the AVB demo once we receive the next i210 card (which is on transit)
We shall revert you if we stuck with any errors while setting up the demo.

Thanks & Regards,
Swapna

closed 2017-02-04 by pinealservo