Avnu/OpenAvnu #511: mrp_connect() understanding

baijayantimala, 2016-11-02:
Hi,

I am unable to understand why we have the below unused code snippet in the mrp_connect() in Open-AVB/blob/master/examples/common/talker_mrp_client.c file.

	memset(&addr, 0, sizeof(addr));
	addr.sin_family = AF_INET;
	addr.sin_port = htons(MRPD_PORT_DEFAULT);
	inet_aton("127.0.0.1", &addr.sin_addr);
	memset(&addr, 0, sizeof(addr));

I am not seeing anywhere the addr structure is used. Please help me to understand this.

regards,
Baijayanti

andrew-elder, 2016-11-02:
You are referring to this function
https://github.com/AVnu/Open-AVB/blob/open-avb-next/examples/common/talker_mrp_client.c#L350
?

Looks like a bug to me. One of those ones with no negative side-effects! Feel free to submit a pull request with a fix in it.

andrew-elder, 2016-11-02:
https://github.com/AVnu/Open-AVB/blob/open-avb-next/examples/common/talker_mrp_client.c#L366
doesn't look correct either. Multiple sequence returns make no sense!

eric-mann, 2016-11-02:
definitely the "addr", setting various fields and the memset() is unnecessary code ... and it goes back virtually to the intial checkin of simple_talker - *blush* ... 

I don't understand the multiple returns either ...