Avnu/OpenAvnu #517: Is the time interval for SYNC and PDelay_Req  fix?

amiguptaa2000, 2016-11-15:
Hi All,
I wanted to know that the time interval for SYNC/Pdelay_Req messages are fix?
Suppose the time interval between two SYNC messages is 125ms .
In my design there might be some data that is being transferred is in the middle but the
sync time interval(125ms between two SYNC message) comes, in this case what will happen?
Will it interrupt the transferring the current data and SYNC message will be transferred?
or It will complete the current data and then SYNC message will be transferred?

Please reply me asap.
Thanks and Regards
--Amit Gupta

jdkoftinoff, 2016-11-15:
The Sync interval can be adjusted when needed using management messages or switch settings.  The Sync can be delayed by up to 50% without affecting sync quality. (See IEEE 802.1AS Corrigendum).  

Default is 125 milliseconds. 

What specific message are you thinking may interfere with the 125 milliseconds interval? On 100baseT the maximum time a single packet can take on the wire is around 0.122 milliseconds. On GigE, the maximum time a single packet can take on the wire is around 0.0122 milliseconds.  If your system puts Sync messages on the best effort queue and your best effort queue is very large  then you may encounter problems when you have a single process sending a solid block of packets - For example 1028 packets on 100baseT or 10280 packets on GigE.

I personally make the recommendations that ideally one would use 4 separate priority queues in order to make sure that best effort packets do not cause queue congestion on the transmitting device itself:

Highest Priority: #1 802.1AS messages
Next Priortity : #2 IEEE 1722 media packet messages
Next Priority : #3 IEEE 802.1Qat (SRP) stream reservation messages
Lowest priority : #4 all other control messages (IEEE 1722.1, MAAP) and IPv4 and IPv6.

Notes / References:

IEEE 802.1AS-2011 Clause 10.6.2.3 time-synchronization event message transmission interval
http://tsn.statusbar.com/page/developer-faq/#can-fqtss-be-implemented-in-pure-software <http://tsn.statusbar.com/page/developer-faq/#can-fqtss-be-implemented-in-pure-software>

amiguptaa2000, 2016-11-24:
Hi Jeff Koftinoff ,
Thank you very much . 
Now I have other question regarding time of Follow_Ups messages.
After SYNC message, with-in how much time FOLLOW_UP message should be transmitted/received for 10Mbps/100Mbps,1G.
Similarly after Pdelay_Req message , with-in how much time Pdelay_Resp and PDelay_Resp_Follow_up messages should be transmitted/received.

If these follow_ups are not received in the given time then what signal should be issues i.e. timeout

Thanks and Regards
--Amit Gupta

jdkoftinoff, 2016-11-25:
Keep in mind that if the time between Sync and FollowUp is more than 10 milliseconds, then inaccuracies will happen. However a slave port will accept the FollowUp as long as it is received within 1.5 times the Sync interval from when the Sync was supposed to be sent.  Typically the Sync interval is 125 milliseconds.

Please see Annex B.2.2 and B.2.3. 
B.2.2 Residence time

The residence time (see 3.17) of a time-aware system, measured relative to the TAI frequency (see 8.2), shall be less than or equal to 10 ms. 

B.2.3 Pdelay turnaround time

The pdelay turnaround time is the duration of the interval between the receipt of a Pdelay_Req message by a port of a time-aware system, and the sending of the corresponding Pdelay_Resp message.

The pdelay turnaround time of a time-aware system, measured relative to the TAI frequency (see 8.2), shall be less than or equal to 10 ms.

amiguptaa2000, 2016-11-25:
Hi Jeff Koftinoff ,
Thank you very much for the explanation.
Now I have below 2 questions:
1.
I would like to know that I will send gPTP message as payload in standard ethernet (Preamble, SFD, D.A, S.A, Ethertype , gptp_messages, FCS) as given in specification. In my normal MAC implementation of standard Ethernet I am not having queuing system, then how can I assume that gPTP messages will be having higher priority. Please help me if I am wrong .

2.

My Second question is that how big IPG can be allowed in any packet, my question is for how long SYNC message can wait if a packet with very bit IPG is in progress ?

amiguptaa2000, 2016-12-26:
Hi Jeff Koftinoff ,

I have a question on timeout . what is the value supported for SYNC and Pdelay_Req message in gPTP for timeout?
Thanks and Regards
--Amit Gupta

jdkoftinoff, 2016-12-26:
I don’t understand your question.  Are you saying that your ethernet adapter does not have queues?

How long is your IPG? Sync messages by default are to be sent every 125 milliseconds.

Jeff

jdkoftinoff, 2016-12-26:
Hello Amit

I think I need clarification about your question.

Sync messages are sent by default every 125 milliseconds. This is called the sync interval.

Pdelay_req needs to be responded to within 10 milliseconds. This is called the maximum pdelay request turnaround time.

Regards,
Jeff

amiguptaa2000, 2017-01-23:
Hi Jeff,
I have another question regarding wall-clock implementation. I am trying to write code in HDL for wall-clock implementation. The timing used in gPTP is upto sub nano seconds(16bit). Now can you please let me know how to update the clock for sub-nanosecond when my clock on which this time will increase is of 1pico-second time period? If I try to go close to 1 PS in 16bit representation then there is some inaccuray . how to handle this problem
Thanks in advance.
Regards
--Amit Gupta

jdkoftinoff, 2017-01-23:
Hello Amit

I’m sorry I don’t have enough context to understand your question.

Regards,
Jeff

amiguptaa2000, 2017-01-24:
Hello Jeff,
My question is that fractional nano second time is send in correction field . I want to know how this sub nano second time is achieved. 
I feel there will be some clock whose precision is sub nano second , that's why when SYNC message is time-stamped then second and nano second are send in precision origin time field of follow_up and fractional nano second in correctionfield . I wanted to know how this sub nano second is achieved.
Thanks in advance.
Regards
Amit Gupta