Avnu/OpenAvnu #518: Did anyone know how to change the SR-class ID ?

stujeffrey016, 2016-11-15:
Hi guys,

I find that, although simple_talker and live_stream example set the packet transmit time as 125us,  the msrp advertise packet still use SR-class as 0. Means that we can't fully control the SRP part. So I want to change that part to validation this function.
Did anyone know how to change the SR-class ID in msrp.c file ?

Regards,
Jeffrey.

andrew-elder, 2016-11-15:
The Talker Advertise does not contain a SR-class. It instead contains a vlan id. See https://github.com/AVnu/Open-AVB/blob/0c8a455ec1d84ff09608dcd74cfe5726da59385a/examples/mrp_client/mrpValidate.c#L198

The domain is used to specific the SR-class. In any case, you will not need to modify msrp.c, as all parameters are passed in via a UDP message. Nothing is hardcoded at all.

I hope that helps.

- Andrew

andrew-elder, 2016-11-16:
MRSP packets can contain 4 different attribute types. Domain, Listener, Talker Advertise and Talker Failed types. The Talker Attribute does not "contain" the Domain. They just happen to be in the same packet. As to why the Domain is sometimes there and sometimes not, that is dictated by the MRP state machine in IEEE 802.1Q. Have you read http://avnu.org/wp-content/uploads/2014/05/AVnu_SWAPIs-v1.3-SWAVI-Updates_Final_Approved_Clean.pdf and http://avnu.org/wp-content/uploads/2014/05/AVnu_Stream-Reservation-Protocol-v1.pdf from the Avnu website? They provide very useful background.

stujeffrey016, 2016-11-16:
Hi Andrew, 

Thanks for your information, it's really helpful.
I'll read the PDF and analysis the packet again.

Regards,
Jeffrey.

closed 2016-11-21 by stujeffrey016