Avnu/OpenAvnu #524: MP3 playback

sachindes, 2016-11-21:
We are able to execute wav, mp4 files with open avb IGB setup on PC. However referring existing examples of open harness( README), we are not yet able to play mp3 files.  Do anyone have an idea about this?

pinealservo, 2017-05-25:
This is a very old question; I'm going to supply an answer for posterity and then close it.

If you can get the gstreamer example to work, it ought to work just as well for mp3 as mp4 audio. Alternatively, you could decode to wav and play that, or set up a Jack network to decode and play them through the jack talker, etc.

closed 2017-05-25 by pinealservo

sachindes, 2017-05-26:
Hello pinealservo ,

Using AAF_talker, I could play mp3 files as well.
Audio files played with rythm box, vlc player or even you tube and routed to ALSA customized device.
AAF talker configured for that ALSA Dev as input.
Effectively here too, audio is decoded to uncompressed audio.