Avnu/OpenAvnu #525: gPTP synchronization between two systems

ranjanBalegara, 2016-11-22:
I cant see any functions being called to synchronize time between two systems in gptp code. How the time synchronization is happening between two systems?? Which function in gptp is called to synchronize the time between two systems running open-avb gptp?

pinealservo, 2017-02-04:
I don't understand what you're asking.  What do you mean, specifically, when you say "synchronize the time"?  The point of the gPTP daemon is not to synchronize the real-time clock of your computer to another one, but to provide a way to relate high-precision timestamps taken on different machines to one another in order to coordinate events and measure frequency/phase offsets of media clocks.  If this doesn't answer your question, maybe it can help you ask again in a way that I can understand how to answer.

andrew-elder, 2017-02-13:
No response - closing.

closed 2017-02-13 by andrew-elder