Avnu/OpenAvnu #533: No Hardware Timestamping on Path_Delay_Request Message

pavan-pan, 2016-12-18:
Hello All,

I am trying to implement a gptp slave on ARM Cortex M4 Micro-controller (LPC 4357). As a first step, I am trying to get a hardware time-stamp on Path_Delay_Request Message. But there is no hardware timestamp on those messages.

I used ppsi ptpv2 (ptp over UDP) which is just another ptp daemon to verify. I can see that there is hardware timestamp on those messages. Has any one implemented gptp slave ? Has anyone observed a hardware timestamps on micro controllers? 

I have also enabled time stamping on ptp over ethernet. If anyone has any suggestions, they are greatly welcome. 

Thank you,
Pavan.

andrew-elder, 2016-12-19:
I assume you have a driver running on the ARM that you think will provide timestamps?
If this is not running on Linux, you may need to write your own HAL layer to extract timestamps from where ever they are generated.

pinealservo, 2017-02-04:
Yes, you should be able to get that MCU to work just fine.  See table 619 on page 839 of http://www.nxp.com/documents/user_manual/UM10503.pdf. Path_Delay_Request is going to be PDelay_Request in that table, not Delay_Request, so that narrows your choices considerably.

If you're doing a gPTP slave, you can't do PTP over UDP.  It's got to be PTPv2, PTP in Ethernet Frames, don't set the event-only bit, and use peer-to-peer transparent clock mode.

I wish you good luck getting it to work, but this isn't really the forum to discuss it in the future, as we would like to keep it focused on the software in this repository, which is not going to run on your Cortex M4.

closed 2017-02-04 by pinealservo