Avnu/OpenAvnu #534: OpenAVB livestream delay issue

AshlinVimal, 2016-12-30:
I'm using Opan_avb master branch.
I have connected 2 PCs directly (PC <--> PC).
i210 cards are inserted in both PCs. OpenAVB stack is running on both PCs.

A USB camera is connected to Talker PC.

When I try to do a live video streaming using the livestream test application, I am getting a delay in the recieved streames.

	* Start AVB talker:
	
		gst-launch-1.0 v4l2src do-timestamp=true ! queue ! videoscale ! video/x-raw, width=640,height=480,framerate=30/1 ! jpegenc ! queue ! fdsink | ./talker eth2 1024

   
	* Start AVB listener:
	 
		sudo ./listener eth1 1024 | gst-launch-1.0 fdsrc fd=0 ! queue ! decodebin ! queue ! xvimagesink
		

I am seeing the delay while sending the frames from talker itself. That delay is appeared in the listener.

Does anybody face similar issue?

Please help me on this.

Thanks in advance.

AshlinVimal, 2016-12-30:
I am suspecting the issue is with the igb_clean process. 
When I added log messages, I found that the below part of code is called repeatedly.

cleanup:
		igb_clean(&igb_dev, &cleaned_packets);
		while (cleaned_packets) {
			tmp_packet = cleaned_packets;
			cleaned_packets = cleaned_packets->next;
			tmp_packet->next = free_packets;
			free_packets = tmp_packet;
		}
		  goto start;
	  }
	  
But the code flow is not entering inside the while loop. After several attemps ( around 100-300) of igb_clean,  we are getting cleaned_packets and continuing the flow. 
The time taken to get a cleaned_packets is coming as a delay in streaming.

This delay is occuring in almost every second, and the delay last for 0.5s

pinealservo, 2017-02-04:
That behavior suggests to me that your tx queue is stuck; it's not sending anything, so you've queued up all your descriptors, and you keep looping trying to reclaim some of them.  The descriptors have a launch time programmed in them, and the resolution for the launch time is such that only a small wall time window is available for scheduling in the near future.  If your timestamp calculations fall outside that window, past values could map to future values and vice-versa, leading to delays.

It's also possible that something else is breaking the Tx queue; are there any messages in the kernel log about the NIC getting stuck and reset periodically? I've seen this before, although it's been a long time and I don't remember the details around it.

pinealservo, 2017-04-10:
There has been no activity on this issue in a long time; please re-open if you have additional problems related to this.

closed 2017-04-10 by pinealservo