Avnu/OpenAvnu #537: Using the open AVB stack for IMX6 Sabre AI board

ajav88, 2017-01-08:
Hi,
   Can this open source stack be used to implement AVB on an IMX6 Sabre AI board provided by NXP?

jdkoftinoff, 2017-01-08:
Have you looked at 

https://community.nxp.com/docs/DOC-95578

and 

http://www.nxp.com/products/software-and-tools/run-time-software/professional-services-software-technology/audio-video-bridging-software:AVB-SOFTWARE

I believe that in the iMX6 quad the time stamping capability of the Ethernet is not truly perfect for 802.1AS implementation and the imx solo 6x is better.

pinealservo, 2017-02-04:
The imx6SoloX has a newer version of the ENET core; all the imx6 family seem to have the same 1588 timestamping and clock adjustment mechanism (the older FEC w/1588 core had a different adjustment mechanism that produced more jitter, but I don't think any imx6 parts have that), but the SoloX has two-class credit-based shaping and launch-time based shaping capability built in, along with some other niceties like frame alignment adjustment, etc.

All the imx7 and imx8 parts seem to have the same ENET core as the imx6SoloX, as do their newer Power-based MCUs.

closed 2017-02-04 by pinealservo