Avnu/OpenAvnu #538: Using the open AVB stack for the NXP imx6 sabre AI board

ajav88, 2017-01-11:
Hi,
   I am currently using a imx 6 sabre AI board to run an AVB  demo. I read that all the code apart from the igb driver files can be ported to the imx6 board. So my question is, is it possible to configure the code for the specific driver used in the imx6 board. If so, where can I find the required documentation for that?

jdkoftinoff, 2017-01-11:
Which specific driver are you expecting to use?

ajav88, 2017-01-11:
A driver for the Qualcomm Atheros Gigabit Ethernet PHY card.

andrew-elder, 2017-01-11:
Is that the AR8031?

andrew-elder, 2017-01-11:
This thread indicates there is no driver available yet for AS8031 https://www.mail-archive.com/search?l=linuxptp-users@lists.sourceforge.net&q=subject:%22%5C%5BLinuxptp%5C-users%5C%5D+AR8031+%5C(or+other+1+GigE%5C)+PHC+driver+availability%22&o=newest&f=1

ajav88, 2017-01-12:
Okay then is there some other way I can implement this open AVB on my imx6 sabre ai?

closed 2017-02-04 by pinealservo