Avnu/OpenAvnu #539: simple talker listener packets seems not sent

olcydvt, 2017-01-12:
Hi everyone

We tried to run simple talker listener example. We use two separate ubuntu pc with springwell NIC. 
 
Firstly, we  launched sudo ./startup.sh eth0 in kmod/igb to load the required driver for the AVB interface.
Secondly, we started avb environment by GPTP time synchronization daemon. 
As a last stage of AVB environment we start stream reservation daemon
Until here everything is ok.
But in last stage when we try to start AVB talker example, as we see from wireshark capture, there is no packet traffic. I think there should be some packet traffic from talker to listener over stream reservation deamon. We may need some trig to start packet transfer I dont know. How can we understand there is packet transfer or not? Also, Do we need any stage to start AVB talker example ?

Thanks

deshpanv, 2017-01-17:
Hello,

You cannot capture packets through wireshark. When you send packets using AVB, the simple talker application bypasses the network stack and write directly to the Buffer Queue.

However, on the listener's side you can see AVB Packets through wireshark. Also,check the size of the media file created, it will increase with time if AVB Communication is really happening.

olcydvt, 2017-01-17:
Thanks for your reply

I will check from listener side but I didnt understand that aren't  outgoing packets sent over network interface card? If it is I think I must have capture the packets from sender side. Please give more information about it.
 
As I understand we don't need to any trigger to start packet transfer aside from starting talker and listener applcations. right ? Just I need check the packet size of listener side.

I will look forward your reply?

olcydvt, 2017-01-17:
I understood why we can not capture packets from sender side

Thanks

closed 2017-01-20 by olcydvt