Avnu/OpenAvnu #541: Getting Hardware Timestamp

deshpanv, 2017-01-17:
Hello,

I have successfully tested simple talker and listener for AVB and it is working. I was also able to change the class of the packets sent and observe the change.

I just want to get a timestamp (hardware) when receiving a packet. I need guidance for the same. I had made a program in c (for normal ethernet) to get hardware timestamp. It is working. When I use gPTP, the timestamp changes to the date and time of the switch.

So any guidance to adapt my program for AVB Simple listener to get hardware timestamp is highly appreciated. ( I have attached my program as well  
[receive_ts.txt](https://github.com/AVnu/Open-AVB/files/710731/receive_ts.txt)
)

varun-v-deshpande, 2017-01-26:
Hello,

I think I solved the problem by launching `daemon_cl ethX -S`. to synchronize the local slave clock to that of the master.

I then edited the code of `linux_hal_generic.cpp`

> while( cmsg != NULL ) {
> if
> ( cmsg->cmsg_level == SOL_SOCKET &&
> cmsg->cmsg_type == SO_TIMESTAMPING ) {
> Timestamp latency( delay->gb_rx_phy_delay, 0, 0 );
> struct timespec *ts_device, *ts_system;
> Timestamp device, system;
> ts_system = ((struct timespec *) CMSG_DATA(cmsg)) + 1;
> system = tsToTimestamp( ts_system );
> ts_device = ts_system + 1; device = tsToTimestamp( ts_device );
> device = device - latency;
> //Then I printed the value of device. This is the hardware timestamp which I wanted.
> gtimestamper->pushRXTimestamp( &device );
> break;
> }
> cmsg = CMSG_NXTHDR(&msg,cmsg);
> }

closed 2017-01-26 by varun-v-deshpande

Shuchish, 2021-08-25:
Hello Dhruv, 
Read your post regarding timestamping, am also facing some GPTP issue in RX side.

"Received an event packet but cannot retrieve timestamp".

If it is a known issue, some suggestions would be highly appreciated.

Thanks.