Avnu/OpenAvnu #545: gPTP not Working

deshpanv, 2017-01-23:
Hello,

When I check gPTP Frames through wireshark, it catches the frames and I could verify that gPTP was working. Even on the switch I can see the same.

But when I print the hardware timestamps from the linux_hal_generic.cpp, I see that the time is not updated to what it is on switch. So is this normal or I am printing wrong timestamp ? I have printed the 3rd timestamp structure of cmesg as others return only zero.

Any guidance is highly appriciated.

christopher-s-hall, 2017-01-23:
gPTP doesn’t, by default, update the system or network device time. The timestamp you see in linux_hal_generic.cpp is the network device time. The third zero timestamp that you mention is a deprecated kernel feature where the network device timestamp is translated to system time. I’m not aware of it being used anywhere.

gPTP outputs an adjustment (see the IPC class) that is used to transform a system clock value (CLOCK_REALTIME) to network time.

deshpanv, 2017-01-25:
Hello,

Thank You for the kind guidance. Basically, what I want to do is launch AVB communication at a specific time. For that, I have to have reference of AVB  Grandmaster Clock (In this case, the switch).

So, how can I access that clock ? For non-AVB communication, I had used ptp4l (linux ptp) to synchronize  with Master Clock and NIC. Then I had Hardware timestamp for reference.

For AVB, I used the same apporach but with your gPTP but as you said, it doesn't update the time on NIC, is there any other way to do the same ?

Waiting for your kind guidance.

andrew-elder, 2017-01-25:
Are you asking how to obtain the correct 802.1AS time to put in the timestamp field of an AVTP packet that you are going to send (talker operation) on the network?

deshpanv, 2017-01-25:
Hello,

I think I solved the problem by launching `daemon_cl ethX -S`. to synchronize the local slave clock to that of the master.

I then edited the code 
`while( cmsg != NULL ) {
			if
				( cmsg->cmsg_level == SOL_SOCKET &&
				  cmsg->cmsg_type == SO_TIMESTAMPING ) {
				Timestamp latency( delay->gb_rx_phy_delay, 0, 0 );
				struct timespec *ts_device, *ts_system;
				Timestamp device, system;
				ts_system = ((struct timespec *) CMSG_DATA(cmsg)) + 1;
				ts_system2 = ((struct timespec *) CMSG_DATA(cmsg)) + 2;
				system = tsToTimestamp( ts_system );
				ts_device = ts_system + 1; device = tsToTimestamp( ts_device );
				device = device - latency;
                                 //Then I printed the value of device. This is the hardware timestamp which I wanted.
				gtimestamper->pushRXTimestamp( &device );
				break;
			}
			cmsg = CMSG_NXTHDR(&msg,cmsg);
		}`
So did I do it correctly to retrieve the time of switch ?

andrew-elder, 2017-01-25:
That is one way, but most implementations do cross time stamping. See
http://avnu.org/wp-content/uploads/2014/05/AVnu_SWAPIs-v1.3-SWAVI-Updates_Final_Approved_Clean.pdf for a description.

The idea is that you obtain a set of parameters to convert a local clock to 802.1AS GM time. Then you read the local clock and run the time conversion to obtain GM time.

closed 2017-01-26 by varun-v-deshpande