Avnu/OpenAvnu #550: Use phc2sys with gPTP

varun-v-deshpande, 2017-01-30:
Hello,

I want to use phc2sys command with gPTP. Can you please guide me how should I use it. I tried running gPTP daemon_cl using -S option and then phc2sys but I get the following error.

> phc2sys[20197.268]: Waiting for ptp4l...
> phc2sys[20197.268]: uds: sendto failed: No such file or directory

Waiting for kind guidance.

christopher-s-hall, 2017-01-30:
I’m not an expert in using phc2sys, but my reading of the man page is that waiting on the ptp4l daemon is not the default behavior. Make sure you’re using phc2sys from git and don’t use the –w flag.

varun-v-deshpande, 2017-01-30:
Hello,

I followed your advise but it ask

> ./phc2sys -s /dev/ptp0 
> time offset must be specified using -w or -O

Can you please guide me on what time offset I need to use ?

pinealservo, 2017-01-30:
A better way to get the current NIC time from userspace is to use /dev/ptp0 as a clock for clock_gettime(); you can find in the kernel Documentation directory some sample code that illustrates how to turn an fd from open("/dev/ptp0") into a clock ID that you can pass to clock_gettime().  It's weird and not well documented, but it works, and I think it's more likely to get you what you want than trying to sync your system clock to the NIC clock.

I went looking for the documentation I referenced, but apparently the info isn't in the kernel documentation anymore; someone moved the test program that showed how to do it somewhere else. Meanwhile, you can see it here: http://lxr.free-electrons.com/source/Documentation/ptp/testptp.c?v=4.8

varun-v-deshpande, 2017-02-02:
Hello,

The approach you mentioned and the approach I mentioned, both have a latency of around 5 microseconds. The less latency approach is to use CLOCK_MONOTONIC or CLOCK_REALTIME (latency around 100 nanoseconds) which is synchronized to /dev/ptp0 using phc2sys. 

So, is there any way to do that ? Right now I am able to use phc2sys to synchronize both but the offset is tobe provided during run time and most importantly, the offset is constant. Can you please guide me on something like phc2sys which has less latency ((sub nano second level)?

pinealservo, 2017-02-03:
The gPTP daemon provides a shared-memory IPC interface through which you can get several values that will be useful to you. There are grandmaster-to-local (NIC) clock phase and frequency offset values and local-to-system clock phase and frequency offset values, along with the current value of the local clock. Using these values and a CLOCK_MONOTONIC timestamp from the system high-res timer, you can translate the timestamp to either the local clock or grandmaster clock time domain yourself.  There's some example code that does this in the lib/common/avb.c file, and it's how most of the examples as well as the avtp_pipeline code work with timestamps.

I don't know of any way of reliably doing anything with software on a PC with sub-nanosecond latency; from what I've seen just reading the TSC via assembly instructions will take tens of nanoseconds on average, or up to ~100ns if it needs to be synchronized with CPUID beforehand. Assuming your kernel is configured well and is able to use the TSC efficiently, using CLOCK_MONOTONIC is only a bit more expensive.

From my understanding of phc2sys and PTP, the offset you're supposed to provide is the number of current seconds offset between TAI and UTC. The former isn't adjusted for leap seconds, so there's an offset between the two that increases by whole seconds when leap seconds occur. This value is currently 37 seconds.  Since the offset you see when running phc2sys is on the order of years rather than tens of seconds, your NIC local clock is clearly not synchronized to TAI.

This could be because your grandmaster is not synchronized to TAI either; if you look at its clock information in the announce packets, you can find the clock class and time source fields. These *can be* but *might not be* traceable to TAI; one of the options for time source is INTERNAL_OSCILLATOR which can have an arbitrary epoch rather than the standard PTP one.  The epoch for the Local Clock is often *not* the PTP epoch, and if your grandmaster uses its Local Clock instead of a ClockMaster, it could have an essentially random epoch. If you've verified that the gPTP timestamps on the wire are based on the same epoch as the ones you read from the NIC Local Clock, then you probably have a grandmaster that's out-of-sync with TAI.  Either adjust priority1 values so that a grandmaster is selected that is synchronized to TAI or don't try to synchronize your system clock to gPTP time. :)

closed 2017-02-13 by varun-v-deshpande