Avnu/OpenAvnu #552: Porting the open AVB stack to other platforms other than intel

ajav88, 2017-02-03:
Hi,
   I am interested in using the Open AVB stack on the imx6 NXP Sabre Ai board. Going through the code I see that there is a heavy dependence on the igb driver, used for the Intel Ethernet Controllers. But the board currently supports an Atheros Enet phy card, and a broadcom BCM89810 Phy card. I am able to run the gptp and mrpd daemons on the board, but I am unable to run any application due to its dependence on the igb driver. Are there any references that I could look through to go about porting the Open avb code on the above mentioned platforms?

pinealservo, 2017-02-04:
There is code in the avtp_pipeline section of the repository, in platform/Linux/rawsock, that provides different ways of sending/receiving raw ethernet frames. The ones besides the igb one use standard kernel interfaces. The downside is that the code needs to be cleaned up a bit to allow other methods to be selected.  The avtp_pipeline code is fairly well-documented and modular, although there's a lot of it to dig through.

The Phy cards for the SabreAI board only determine what physical layer connection you can use; neither of them supply any special AVB-related functionality. The Sabre AI has an imx6 Solo on it, if I understand correctly, and that has an older version of the Freescale ENET MAC core in it.  This doesn't have any special traffic shaping capability, so you're going to have to rely on software to do traffic shaping, which can work in some circumstances but is not going to be great when combined with the Linux kernel and general-purpose driver overhead.

As far as Freescale/NXP parts go, the imx6 SoloX is a newer part that has a better MAC core in it that has traffic shaping and launch-time support built-in, so it's roughly as capable for AVB as the Intel i210, but there's no Linux support for taking advantage of it (aside from PTP) yet. The imx7 parts have full AVB support as well, and the imx8 parts are also listed as having AVB support although their reference manuals aren't published on their site yet so I don't know if it's changed from the imx6SoloX and imx7 MAC core.

ajav88, 2017-02-06:
Hi,
   I am very much new to this. Could you possibly mention how I go about it, as I am not even able to make the files in the board.

andrew-elder, 2017-02-06:
Did you try the steps here https://github.com/AVnu/Open-AVB/tree/master/lib/avtp_pipeline ? Was there a specific step that failed?

pinealservo, 2017-02-06:
@andrew-elder There's currently no provision in the avtp_pipeline code (that's checked in here, at least) to build on platforms besides Linux x86 w/i210 NIC.

@ajav88 At present, we do not officially support building for that system. The code is not far from being able to build for other Linux-based systems, and we may support that in the future, but it's not supported now.

By "not supported", I mean that we can answer specific questions that might help someone along who is doing development work to enable support, but we can't teach someone how to do it from scratch. You've asked the same basic question here numerous times over the course of a few months; I'm going to have to request that you stop opening issues until you have a specific issue for us to address.

We would really like to encourage people to use the software here and help develop it, but as provided now it's in a "some assembly required" state and is not suitable for everyone. If you need a polished, complete solution right now, or if you don't have the software development resources and AVB knowledge to put the pieces together and adapt them to your needs, this is not the project for you, and you should be looking at commercially-supported solutions.

closed 2017-02-06 by pinealservo

andrew-elder, 2017-02-06:
Oops - my bad!