Avnu/OpenAvnu #557: simple_talker PKT_SZ not changing

varun-v-deshpande, 2017-02-16:
Hello,

In simple talker application, I tried to change PKT_SZ to 200 but on wireshark at receiving end I still see only 98 Bytes.

Is it normal to happen ?

pinealservo, 2017-02-17:
Changing PKT_SZ should change the length of the frames. Are you sure the code change got into the binary you are running?

varun-v-deshpande, 2017-02-17:
Hello,

Yes, I am sure of that.
However, I would like to point few observations which I made.
Increasing PKT_SZ does increase the reserved bandwidth. I checked that on switch.

On the receiving side, When I checked with wireshark,  no matter what is my packet size, IPG, I always see AVB Transportation Protocol Packet 1722 of 98 Bytes.

On further observation I found that this 98 is nothing but 
18 + sizeof(*l4_headers) + (L4_SAMPLES_PER_FRAME * CHANNELS * L4_SAMPLE_SIZE );

varun-v-deshpande, 2017-02-20:
Hello,

Any explanation for the observation I made ?

closed 2017-04-10 by pinealservo

up0617, 2021-02-01:
Hi, PKT_SZ is for SRP BW Reservation at igb_set_class_bandwidth() in in igb.c 
to increase packet size sent, you should increase L2_SAMPLES_PER_FRAME or L4_SAMPLES_PER_FRAME if you use RTP