Avnu/OpenAvnu #559: simple_talker Explaination

varun-v-deshpande, 2017-02-17:
Hello,

Can anyone please guide me about this code snippet from simple_talker ?

      

>      update_8021as = td.local_time - td.ml_phoffset;
> 	delta_local = (unsigned)(now_local - td.local_time);
> 	delta_8021as = (unsigned)(td.ml_freqoffset * delta_local);
> 	now_8021as = update_8021as + delta_8021as;
> 	last_time = now_local + XMIT_DELAY;
> 	time_stamp = now_8021as + RENDER_DELAY;
> 	rc = nice(-20);

pinealservo, 2017-02-17:
The fields in `td` are from the gPTP daemon and they include the information necessary to do clock domain transformations so that the timestamp at some point in the future can be given as it will appear on both the gPTP grandmaster clock and on the local NIC clock.

We just got the NIC's local time stored in `now_local`; the first step is to figure out how much time has elapsed on the NIC's clock since the gPTP information was refreshed; that becomes `delta_local`. We can then multiply that by the gPTP master-to-local frequency offset to determine how many gPTP-corrected nanoseconds that time period was, which becomes `delta_8021as`.

To do the phase correction, we have to translate by the gPTP master-to-local phase offset. The code does that in two steps; the first is in the first line you pasted, where the NIC's local timestamp at the last refresh time is translated to become `update_8021as`, and then we add our calculated `delta_8021as` to get the `now_8021as` that corresponds to the NIC-local timestamp we started with.

We need to store a gPTP timestamp in the frames as a presentation time, and we need to use a NIC-local clock timestamp to schedule the launch time.  `last_time` is going to be used to schedule periodic launch times and we're pretending that we scheduled one for right now plus some delay to account for the time it would take to get it to the NIC. `time_stamp` is going to go in as the sample presentation time, so we add the `RENDER_DELAY` to account for transmission delay.

Each time we prepare another frame, we just increment both timestamps by a fixed amount. This simulates a media clock that's synchronized to a fixed multiple of the grandmaster clock; whether that's reflected in the actual rate we're sending the samples will depend on whether we had gPTP syntonize the NIC clock to the grandmaster clock or not. Please note that having the grandmaster syntonized with either the local clock or the media clock is by no means guaranteed by AVB; AVB just gives you the means to recover the clocks you need and do time domain translation. In other words, you shouldn't rely on how simple_talker behaves here, and depending on how gPTP is configured, it might even be wrong.

Our APIs are currently a little weak in the area of clock translation and media clock measurement and adjustment, largely because there isn't much in the way of standardized OS interfaces (or hardware interfaces!) for such things. On Linux, ALSA has some reasonable interfaces for doing media clock cross-timestamping with the system timer, but we don't currently have any way to adjust the physical sampling clock, so a portable clock recovery module would have to do sample skipping/stuffing or resampling in order to keep the buffer levels constant. If you feed the listener stream into Pulseaudio, it will automatically resample it so you'll get consistent playback without having to do actual clock recovery, but you'll lose some audio quality and also won't have synchronized playback.

varun-v-deshpande, 2017-02-17:
Hello,

Thank You for the explaination. It is more clear now for me.
I have a small query, can I send just a normal packet (no audio/video) using simple talker ?

As the documentation for simple talker does not explain much about its working, in order to understand clearly how does avb mechanism work to send a simple packet and how it is received, is there any such simple example in public domain which you can refer to me ?

Simple_talker and listener is not that pretty simple for beginners.
Thank you once again for the explanation on the code.

pinealservo, 2017-02-21:
You can send any data you'd like with the igb library that simple_talker uses to send its packets.  You should try writing your own program to do it; either delete all the code from simple_talker that doesn't look relevant to what you want to do, or copy code over into a new program that looks like it will be necessary. Then adjust until it works.

Everything you need to understand exactly what happens is there in the code for simple_talker along with the common library, the igb library, and the igb kernel module from this repository.  You should be able to trace the code through with a bit of dedication and some simple software tools (grep or an editor 'search' feature).  If you need to know more about how the i210 hardware works, the manual for it is available.

The state of Linux AVB support is such that there is no unified "avb mechanism" that could be explained by a simple example. Getting a full AVB system working properly will require a lot of additional code and effort beyond what's provided in this repository. If you have time constraints such that you can't spend the time to really understand what simple_talker is doing, you probably need to look at commercially-supported options for whatever your project is. None of the current developers here has a great deal of time available for beginner-level support.

However, if you're willing to dig into things yourself and you have the time available, I don't want to discourage you from experimenting with and maybe contributing to the project. I just want you to have realistic expectations about the state of things.

closed 2017-03-06 by pinealservo