Avnu/OpenAvnu #564: Timestamp on Packet Reception -> Simple Listener

varun-v-deshpande, 2017-03-06:
Hello,

I want to get a timestamp when I receive AVB Packet. Can anybody guide me in this regard ?

Right Now I am using following code :

> clock_gettime(CLOCK_REALTIME, &mytime);
>fprintf(stdout,"%ld.%09ld\n", (long)mytime.tv_sec, (long)mytime.tv_nsec);

I have inserted this code in pcap_callback of simple_listener. But this doesn't seems to work as I have only askfailed state on talker side.

I waiting for guidance in this regards.

pinealservo, 2017-03-06:
For normal AVB operation, precise timestamps are only needed for reception of certain gPTP packets, and the gPTP daemon already takes care of this.

If for some non-AVB reason you need a Rx timestamp, libpcap should already be getting kernel-level timestamps for packet Rx time, and you can even configure libpcap to request hardware timestamps from the NIC if it is capable of supplying them. You shouldn't need to take your own via clock_gettime, which is not a good method even if you have synchronized the system clock to gPTP time due to variable amount of time between the NIC's Rx time and when it gets handed to your userspace application by the kernel.

Your explanation for why your timestamp method doesn't work doesn't make any sense to me. What do listener timestamps have to do with "askfailed state on talker side"? Is the problem that you aren't getting timestamps, or that you aren't getting *packets* to timestamp?

andrew-elder, 2017-03-06:
What is your talker/listener/switch configuration?

varun-v-deshpande, 2017-03-09:
Hello,

@pinealservo , I mean that when I insert the code in listener application, then I am unable to communicate between talker and listener. On the talker side I get aksfailed state. When I remove that code from listener, everything works fine again.

@andrew-elder My configuration is Talker -> Switch ->Listener (Class A)

varun-v-deshpande, 2017-03-09:
Hello @pinealservo ,

Also, with libpcap, I can timestamp packet when I use "pcap_create" to sniff a device but in the simple listener, "pcap_open_live". So can you please guide me how can we proceed further with this ?

Can I launch another pcap instance in a separate program on the same interface to get the timestamps ? Will there be any performance issues ?

varun-v-deshpande, 2017-03-09:
Hello @pinealservo,

I have figured out why I get askfailed state on talker side when ever I make many changes on listener application. The reason is the gptp daemon stops working automatically. I think it crashes but I am unaware what causes it to crash ! Is it a known issue to your team ?

varun-v-deshpande, 2017-03-09:
Hello @pinealservo ,

Following your approach, I am able to get the hardware timestamp on reception by making an edit in the listener program. Can you please guide me what can I do for transmit timestamp (hardware) ? 

Thank You !

pinealservo, 2017-03-09:
@varun-v-deshpande 

Apparently starting a program that runs libpcap on the same interface (an overlapping filter might be required, but I'm not sure) as gPTP breaks packet reception for gPTP.  I'm pretty sure this shouldn't happen, and I'm not sure if it's a new thing, but we're seeing it now too.

The problem doesn't occur when our other raw socket program uses a PF_PACKET socket directly instead of going through libpcap.  It may involve either the RX_RING option that libpcap uses or the BPF filters that it uses.

As for how to get transmit timestamps, you'll have to look at the kernel source tree at the Documentation/networking/timestamping.txt, and you can also look at the gPTP daemon source in the Linux-specific directory for examples. That will also show you how to get receive timestamps without libpcap.

varun-v-deshpande, 2017-03-09:
Hello  @pinealservo 

I am aware of that approach. I have already implemented by programs to have hardware timestamping. The only problem is how do I port it to the simple talker application.

For example, with reference to the extract from my code :

> sendto(sock,(void *)buff,sizeof(buff),0,(struct sockaddr *)&destination_ip,sizeof(destination_ip));
>recvmsg(sock, &msg, MSG_ERRQUEUE);      	
>cmsg = CMSG_FIRSTHDR(&msg);
>stamp = (struct timespec *)CMSG_DATA(cmsg);
>stamp+=2;
>printf("TX_TSTAMP_HW:= %ld s %09ld ns\n", (long)stamp->tv_sec, (long)stamp->tv_nsec);

We cannot have TX timestamp directly. For TX Timestamp we have to read MSG_ERRQUEUE. So, in this case I call recvmsg to have cmsg structure and then I extract the timestamp. Note that this is called after sendto command. Also, before doing this, "setsockopt" is also to be called on the socket to enable Hardware TX Timestamping.

As in the simple talker application, there is no use of sendto command and also the socket, I am bit confused as calling "setsockopt" wont work as I have no socket to pass as an argument. Consequently recvmsg to read ERRQUEUE also won't work. So , I cannot extract the TX Timestamp for AVB Packet.

Can you please guide me with this ? I am trying an approach to modify simple listener just to run pcap with adjusting filter (like src mac) to print timestamp. I am not sure if it will work. Do you think this approach is good ? Any suggestions ?

pinealservo, 2017-03-10:
Ah, for programs using the igb library, you'll have to look at the igb library code.  Look at the header here: https://github.com/AVnu/OpenAvnu/blob/master/lib/igb/igb.h

You can see there's a field called `dmatime` in the same structure as the `attime` field that you use to specify the launch time.  The library routine that fills in the descriptor from that structure appears to be setting the flag that will cause a timestamp for when the DMA writeback of the descriptor header occurs; this gets copied into the `dmatime` field when you do the `igb_clean` call after a transmit.

This means that in the little routine after the cleanup label in simple_talker.c, you can compare the launch time you set in `attime` with the recorded writeback time in `dmatime` for each of the transmitted packets.  I'm not sure why you want to do this, or what delays/latency this DMA writeback timestamp accounts for, but it should give you some sort of timestamp.  Of course, you should already have a pretty good timestamp which you set in the attime field, assuming it's working correctly.

andrew-elder, 2017-03-10:
@varun-v-deshpande - might I ask what the end goal is here? Why do you wish to look at the receive packet time? Have you considered putting an Ethernet tap in the network and investigating the packet timing that way.

varun-v-deshpande, 2017-03-10:
Hello @andrew-elder ,

I want to calculate the exact time a packet takes from the time it is sent to the time it is received in different scenarios (Like comparing it with loaded and unloaded network and also comparing it with non AVB Packet Time).

Regarding your approach for investigating packet timing, No, I haven't considered putting Ethernet TAP. I will investigate this approach as well. If you have some guides, It can be very helpful for me.

pinealservo, 2017-03-13:
Precise timestamping of media packet transmit/reception is not required for AVB/TSN functionality. The precise transmit time is unimportant as long as it is not excessively delayed before transmit and as long as it meets the traffic shaping requirements of FQTSS. The timestamps you need for media playback are contained within the media packets and are based on timestamping media clock edges, not packet transmit. If the network is functioning properly you will receive the packet before the time in the AVTP presentation timestamp occurs, and the presentation timestamp plays its role in media clock recovery without any reference to transmit or receive time.

Measuring the end-to-end latency of AVB media packets might be an interesting problem, but it's not a supported use case of the OpenAvnu software, and we really don't have particular insight beyond what we've already shared into how to best measure it with whatever tools you might have available. If you need support for a tool that's designed to measure this sort of thing, I'm sure there are such tools and support contracts for them available that will get you timely and comprehensive answers.

closed 2017-03-13 by pinealservo