Avnu/OpenAvnu #567: Weird Results for Total Packet Transmission Time

varun-v-deshpande, 2017-03-13:
Hello,

I am trying to calculate the time required by the packet to travel from PC1 to PC2. I am using modified simple talker and listener.

On the Talker Side, I have inserted this code before `igb_xmit` in `simple_talker.c` :

> sleep(1);	
> clock_gettime(CLOCK_REALTIME, &mytime);			
> err = igb_xmit(&igb_dev, 1, tmp_packet);
> printf("%ld.%09ld\n", (long)mytime.tv_sec, (long)mytime.tv_nsec);
> 

My `CLOCK_REALTIME` is synchronised with the NIC clock using `phc2sys`.

On the Listener Side, I have changed the following code in `simple_listener.c` :

Line 97
>       eth_header = (struct ethernet_header*)(packet);
>   +  fprintf(stdout,"%ld.%ld\n", packet_header->ts.tv_sec, packet_header->ts.tv_usec);

Line 286
>    - glob_pcap_handle = pcap_open_live(dev, BUFSIZ, 1, -1, errbuf);
>    - if (NULL == glob_pcap_handle)
>    - {
>    -    fprintf(stderr, "Could not open device %s: %s\n", dev, errbuf);
>    -	   return EXIT_FAILURE;
>    -	}

>  + glob_pcap_handle = pcap_create(dev, errbuf);
>  + pcap_set_snaplen(glob_pcap_handle, BUFSIZ);
>  + pcap_set_promisc(glob_pcap_handle, 1);
>  + pcap_set_timeout(glob_pcap_handle, -1);
>  + pcap_set_tstamp_type(glob_pcap_handle, PCAP_TSTAMP_ADAPTER_UNSYNCED)
>  + pcap_set_tstamp_precision(glob_pcap_handle, PCAP_TSTAMP_PRECISION_NANO)
>  + pcap_activate(glob_pcap_handle);

With this I am able to get the time at which packet is sent and the time it is received (Hardware)
I have also calculated the latency for `igb_xmit()` which is around 2.5 microseconds.

So, with this the total time taken by packet to will be => Timestamp_at_listener - Timestamp_at_talker- xmit_latency(2.5 microseconds).

With this approach, I get the total time as 220 milliseconds on class A and 212 millisecond on class B when PKT_SZ is 500.

When PKT_SZ is 1000, I get 214 milliseconds on class A and 213 milliseconds on class B.
When PKT_SZ is 1500, I get 275 **micro**seconds on class A and 250 **micro**seconds on class B.

Can anybody guide me with these weird results ? AVB guarantees delays of less than 2 ms on class A over 7 hops and 5 ms on class B. These results which I got are way beyond expectation. 

Thank you for guiding me on this.

andrew-elder, 2017-03-13:
Where exactly are you trying to measure this time? The actual time required on the wire is returned as the neighborDelay from gPTP. This it the time from one RJ45 to another one. With a switch in between you would look at the MSRP propagation delay. It seems you are trying to determine times somewhere in the Linux packet transmit and receive stack?

varun-v-deshpande, 2017-03-13:
Hello @andrew-elder ,

I want to measure the time a packet takes from talker to listener. To be specific, From the time it is written to NIC SEND_BUFFER and the time it is received in RECEIVE_BUFFER.

Can you please guide me why I am receiving these weird results ?

andrew-elder, 2017-03-13:
How accurate do you want the measurements to be?

pinealservo, 2017-03-13:
@varun-v-deshpande I appreciate that you're trying to get something done, but this is really outside the scope of what these support issues are meant for.  As far as we're aware, the OpenAvnu software works properly for AVB media transport when used according to design. Precise end-to-end timestamping of media packets is not a supported use case of the software; AVB usage doesn't require precise media packet transmit or receive timestamping at all. Maybe you can coax the timestamping capability out of the combination of hardware and software you have, but we haven't focused on that since it's not necessary for the protocols to work.

If *you* can find where the time issues you are having are coming from and it turns out to be a problem in OpenAvnu-maintained software, we would appreciate you opening an issue to report it. You're the one with the particular set of hardware and software that's giving you strange results, though, so you're in a far better position to diagnose the issue than anyone else.

closed 2017-03-13 by pinealservo