Avnu/OpenAvnu #570: gptp daemons is not loading due to this erro : GPTP [17:42:25:660] Failed to configure timestamping: Operation not supported

iamdharm20, 2017-03-29:
Hi,

i am not able to load the gptp daemons due to the below errors:

root@KOR1064342VM:/gitweb/OpenAvnu-open-avb-next/daemons/gptp/linux/build# /home/dus5cob/gitweb/OpenAvnu-open-avb-next/daemons/gptp/linux/build/obj/daemon_cl eth1 > ptp.log &
[4] 13897
root@KOR1064342VM:/gitweb/OpenAvnu-open-avb-next/daemons/gptp/linux/build# INFO : GPTP [17:42:25:659] gPTP starting
ERROR : GPTP [17:42:25:660] Failed to configure timestamping: Operation not supported
ERROR : GPTP [17:42:25:660] post_init failed

ERROR : GPTP [17:42:25:660] failed to initialize port
^C

this issue is already resolved by this patch https://review.tizen.org/gerrit/24785 but still i am facing the same problem.

setup : 
ubuntu 14lts
intel 210-T1 gbi NIC network adapter :

eth1 Link encap:Ethernet HWaddr 08:00:27:e3:7c:d9
inet addr:172.17.0.8 Bcast:172.17.0.255 Mask:255.255.255.0
inet6 addr: fe80::a00:27ff:fee3:7cd9/64 Scope:Link
UP BROADCAST RUNNING MULTICAST MTU:1500 Metric:1
RX packets:24 errors:0 dropped:0 overruns:0 frame:0
TX packets:898 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000
RX bytes:2208 (2.2 KB) TX bytes:83495 (83.4 KB)

i am following this procedure to compile the source code:  

cloned this branch : OpenAvnu-open-avb-next
make EXTRA_CFLAGS="-DIGB_PTP" 

then i have loaded all these module.
sudo modprobe i2c_algo_bit
sudo modprobe dca
sudo modprobe ptp
sudo insmod /home/dus5cob/gitweb/OpenAvnu-open-avb-next/kmod/igb/igb_avb.ko

finaly i run this gptp daemons  /home/dus5cob/gitweb/OpenAvnu-open-avb-next/daemons/gptp/linux/build/obj/daemon_cl eth1 > ptp.log & 
but i face this error 
: ERROR : GPTP [17:42:25:660] Failed to configure timestamping: Operation not supported
ERROR : GPTP [17:42:25:660] Failed to configure timestamping: Operation not supported
ERROR : GPTP [17:42:25:660] post_init failed

could you please guide me what is the problem and what should be the solution for this ?

Regrads,
Dharmender sharma

christopher-s-hall, 2017-03-29:
Did you unload all of the drivers (e.g. igb) managing that interface? If that doesn't work, please post the dmesg output generated when the driver is loaded.

iamdharm20, 2017-03-30:
yes i  have unloaded all the drivers properly.

iamdharm20, 2017-03-30:
sorry i am not able to upload the dmesg file but please find the logs :

[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.13.0-112-generic (buildd@lcy01-02) (gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3) ) #159-Ubuntu SMP Fri Mar 3 15:26:07 UTC 2017 (Ubuntu 3.13.0-112.159-generic 3.13.11-ckt39)
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-3.13.0-112-generic root=/dev/mapper/kor1030612--vg-root ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000dffeffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dfff0000-0x00000000dfffffff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000002e45fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.5 present.
[    0.000000] DMI: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] No AGP bridge found
[    0.000000] e820: last_pfn = 0x2e4600 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR variable ranges disabled:
[    0.000000] x86 PAT enabled: cpu 0, old 0x7040600070406, new 0x7010600070106
[    0.000000] CPU MTRRs all blank - virtualized system.
[    0.000000] e820: last_pfn = 0xdfff0 max_arch_pfn = 0x400000000
[    0.000000] found SMP MP-table at [mem 0x0009fff0-0x0009ffff] mapped at [ffff88000009fff0]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [ffff880000099000] 99000 size 24576
[    0.000000] init_memory_mapping: [mem 0x00000000-0x000fffff]
[    0.000000]  [mem 0x00000000-0x000fffff] page 4k
[    0.000000] BRK [0x01fe3000, 0x01fe3fff] PGTABLE
[    0.000000] BRK [0x01fe4000, 0x01fe4fff] PGTABLE
[    0.000000] BRK [0x01fe5000, 0x01fe5fff] PGTABLE
[    0.000000] init_memory_mapping: [mem 0x2e4400000-0x2e45fffff]
[    0.000000]  [mem 0x2e4400000-0x2e45fffff] page 2M
[    0.000000] BRK [0x01fe6000, 0x01fe6fff] PGTABLE
[    0.000000] init_memory_mapping: [mem 0x2e4000000-0x2e43fffff]
[    0.000000]  [mem 0x2e4000000-0x2e43fffff] page 2M
[    0.000000] init_memory_mapping: [mem 0x280000000-0x2e3ffffff]
[    0.000000]  [mem 0x280000000-0x2e3ffffff] page 2M
[    0.000000] BRK [0x01fe7000, 0x01fe7fff] PGTABLE
[    0.000000] init_memory_mapping: [mem 0x00100000-0xdffeffff]
[    0.000000]  [mem 0x00100000-0x001fffff] page 4k
[    0.000000]  [mem 0x00200000-0xdfdfffff] page 2M
[    0.000000]  [mem 0xdfe00000-0xdffeffff] page 4k
[    0.000000] init_memory_mapping: [mem 0x100000000-0x27fffffff]
[    0.000000]  [mem 0x100000000-0x27fffffff] page 2M
[    0.000000] BRK [0x01fe8000, 0x01fe8fff] PGTABLE
[    0.000000] RAMDISK: [mem 0x357f6000-0x36bf2fff]
[    0.000000] ACPI: RSDP 00000000000e0000 000024 (v02 VBOX  )
[    0.000000] ACPI: XSDT 00000000dfff0030 00003C (v01 VBOX   VBOXXSDT 00000001 ASL  00000061)
[    0.000000] ACPI: FACP 00000000dfff00f0 0000F4 (v04 VBOX   VBOXFACP 00000001 ASL  00000061)
[    0.000000] ACPI: DSDT 00000000dfff0480 001B96 (v01 VBOX   VBOXBIOS 00000002 INTL 20100528)
[    0.000000] ACPI: FACS 00000000dfff0200 000040
[    0.000000] ACPI: APIC 00000000dfff0240 000064 (v02 VBOX   VBOXAPIC 00000001 ASL  00000061)
[    0.000000] ACPI: SSDT 00000000dfff02b0 0001CC (v01 VBOX   VBOXCPUT 00000002 INTL 20100528)
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] No NUMA configuration found
[    0.000000] Faking a node at [mem 0x0000000000000000-0x00000002e45fffff]
[    0.000000] Initmem setup node 0 [mem 0x00000000-0x2e45fffff]
[    0.000000]   NODE_DATA [mem 0x2e45f2000-0x2e45f6fff]
[    0.000000]  [ffffea0000000000-ffffea000b9fffff] PMD -> [ffff8802d8a00000-ffff8802e3bfffff] on node 0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x00001000-0x00ffffff]
[... 462 lines cut ...]
[   28.145587] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   28.148590] e1000: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[   28.148856] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   28.581551] vboxdrv: Found 3 processor cores.
[   28.583242] vboxdrv: fAsync=0 offMin=0x18b3 offMax=0x14fd3a
[   28.583383] vboxdrv: TSC mode is 'synchronous', kernel timer mode is 'normal'.
[   28.583385] vboxdrv: Successfully loaded version 4.3.10_Ubuntu (interface 0x001a0007).
[   28.595737] vboxpci: IOMMU not found (not registered)
[   28.701183] vboxsf: Successfully loaded version 4.3.20 (interface 0x00010004)
[   29.105973] init: samba-ad-dc main process (1267) terminated with status 1
[   29.147356] symap_custom_3_13_0_112_generic_x86_64: module license 'Proprietary' taints kernel.
[   29.147360] Disabling lock debugging due to kernel taint
[   55.890461] audit_printk_skb: 39 callbacks suppressed
[... cut ...]
[62838.013387] e1000: eth1 NIC Link is Down
[62844.024468] e1000: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[75669.666376] dca service started, version 1.12.1
[75676.316711] pps_core: LinuxPPS API ver. 1 registered
[75676.316715] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[75676.321414] PTP clock support registered
[75683.318369] Intel(R) Gigabit Ethernet Network Driver - version 5.3.2_AVB
[75683.318375] Copyright (c) 2007-2015 Intel Corporation.

iamdharm20, 2017-03-30:
followed steps :+1: 
unloaded all the driver. then loaded all the drivers

christopher-s-hall, 2017-04-11:
Good catch @pinealservo. The only way this might work is directly assigning the HW device (I210) to the VM. I don't know if that's possible with your configuration/VMM. There could also be performance issues with TSC virtualization. That depends upon the VM configuration.

As noted above, VMs are not supported at this time.

closed 2017-05-25 by pinealservo