Avnu/OpenAvnu #572: About CXX11_FLAG_DETECTED - Failed

kingdjf, 2017-03-31:
-- Try C++11 flag = [ ]
-- Performing Test CXX11_FLAG_DETECTED
-- Performing Test CXX11_FLAG_DETECTED - Failed
-- Try C++11 flag = [-std=c++11]
-- Performing Test CXX11_FLAG_DETECTED
-- Performing Test CXX11_FLAG_DETECTED - Failed
-- Try C++11 flag = [/Qstd=c++0x]
-- Performing Test CXX11_FLAG_DETECTED
-- Performing Test CXX11_FLAG_DETECTED - Failed
CMake Error at /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:91 (MESSAGE):
  Could NOT find CXX11 (missing: CXX11_FLAGS)
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:252 (_FPHSA_FAILURE_MESSAGE)
  thirdparty/cpputest/cmake/Modules/FindCXX11.cmake:48 (find_package_handle_standard_args)
  thirdparty/cpputest/cmake/Modules/CppUTestConfigurationOptions.cmake:48 (find_package)
  thirdparty/cpputest/CMakeLists.txt:32 (include)

-- Configuring incomplete, errors occurred!
gcc version 4.8.1 (Ubuntu 4.8.1-2ubuntu1~12.04) 
how can i set the CXX11_FLAGS?

kingdjf, 2017-03-31:
ok,i work!

andrew-elder, 2017-03-31:
I'm glad you fixed it! Can you share how - in case someone else has the same issue.

kingdjf, 2017-04-02:
yes.this is gcc version issue. CXX11_FLAGS,u must update gcc verstion to 4.8.when u under ubuntu u can use apt-get install gcc++-4.8. this issue under ubuntu 12.04 32bit,kernel 3.13.0 version.

closed 2017-04-03 by andrew-elder