Avnu/OpenAvnu #577: Error in cmake

zeefpt, 2017-04-10:
Hi Folks,

Im trying to build the OpenAvnu with cmake using the readme help:

-------------------
Starting from the OpenAvnu dir, one would go::

    mkdir tmp
    cd tmp
    cmake .. -G "Unix Makefiles"
    make

to build in OpenAvnu/tmp

---------------------

But  cmake gives me this error:
"Make Error at CMakeLists.txt:6 (add_subdirectory):
  The source directory

    /home/jp/Desktop/OpenAvnu/thirdparty/cpputest

  does not contain a CMakeLists.txt file.  "

and looking into folder really doesn have anything.

Anyhelp?
Regards from Portugal

andrew-elder, 2017-04-10:
Did you go
git submodule update --init --recursive
?

zeefpt, 2017-04-10:
hi,
nop,
downloaded from site and untar..
I will try with your suggestion :)

zeefpt, 2017-04-10:
Worked but got a error in make :
"
Scanning dependencies of target CppUTest
[  0%] Building CXX object thirdparty/cpputest/src/CppUTest/CMakeFiles/CppUTest.dir/CommandLineArguments.cpp.o
[  1%] Building CXX object thirdparty/cpputest/src/CppUTest/CMakeFiles/CppUTest.dir/MemoryLeakWarningPlugin.cpp.o
/home/jp/projetos/openavnu/OpenAvnu/thirdparty/cpputest/src/CppUTest/MemoryLeakWarningPlugin.cpp:224:6: error: the program should also define ‘void operator delete(void*, std::size_t)’ [-Werror=sized-deallocation]
 void operator delete(void* mem) UT_NOTHROW
      ^~~~~~~~
/home/jp/projetos/openavnu/OpenAvnu/thirdparty/cpputest/src/CppUTest/MemoryLeakWarningPlugin.cpp:244:6: error: the program should also define ‘void operator delete [](void*, std::size_t)’ [-Werror=sized-deallocation]
 void operator delete[](void* mem) UT_NOTHROW
      ^~~~~~~~
cc1plus: error: unrecognized command line option ‘-Wno-c++98-compat’ [-Werror]
cc1plus: all warnings being treated as errors
thirdparty/cpputest/src/CppUTest/CMakeFiles/CppUTest.dir/build.make:86: recipe for target 'thirdparty/cpputest/src/CppUTest/CMakeFiles/CppUTest.dir/MemoryLeakWarningPlugin.cpp.o' failed
make[2]: *** [thirdparty/cpputest/src/CppUTest/CMakeFiles/CppUTest.dir/MemoryLeakWarningPlugin.cpp.o] Error 1
CMakeFiles/Makefile2:1033: recipe for target 'thirdparty/cpputest/src/CppUTest/CMakeFiles/CppUTest.dir/all' failed
make[1]: *** [thirdparty/cpputest/src/CppUTest/CMakeFiles/CppUTest.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
"
Anyideia?
Regards,

andrew-elder, 2017-04-10:
What version of gcc?

zeefpt, 2017-04-10:
gcc version 6.2.0 20161005 (Ubuntu 6.2.0-5ubuntu12)

pinealservo, 2017-04-10:
The version of cpputest we're checking out out doesn't support gcc 6 yet; if you were to go to the cpputest directory and check out the master branch instead, it would most likely build fine afterward.

We're still using the default Travis setup, which gives a rather old set of tools, and we've had a hard time trying to find a happy configuration that will build there, on up-to-date Linux, and also on Windows.

zeefpt, 2017-04-11:
Hi pinealservo,

Thnks for anwser me.

It necessary the cpputest to use the tools to generate audio and video packets?

 I did not quite understand how to check and build the master branch. Nothing appears to me...

Regards,
José

pinealservo, 2017-04-11:
No, it's not necessary to build the tests. In Linux, there's an alternate set of Makefiles you can use instead of cmake.  Just type 'make' from the top level directory to see the options, then 'make <target>' to build the specific things you want.

zeefpt, 2017-04-17:
Ok, nice,
Thanks again,
Regards,

closed 2017-04-17 by andrew-elder