Avnu/OpenAvnu #578: TSN and repository doubt

zeefpt, 2017-04-11:
Hi again folks,

I have 2 boards with Ethernet and want to include the TSN using this OpenAvnu repository. 
Want to Clog with garbage packages my network and see if with TSN only video/audio packages are delivery just in timing and respecting an certain criterias. Then enable/disable the tsn and see the diferences.

Do you have anyguide or examples of how to implement it using this repository?

Regards,

pinealservo, 2017-04-11:
If you build avtp_pipeline along with the gPTP and MRP daemons, you should be able to use the combination to set up streams across AVB-capable switches. Configuring them to continue working when you turn off AVB/TSN support in the bridges should be possible, but things don't work that way by default; disabling the protocols in the network just shuts things down cleanly rather than allowing interfering traffic to produce poor streaming performance.

The general principles upon which AVB/TSN is based have been pretty thoroughly simulated, tested, and modeled, and most of the people contributing here have moved on to building systems with it instead of re-running old experiments. You are certainly welcome to perform any experiments you'd like, and we will try to help out with specific problems you may encounter, but we don't have guides prepared for that kind of experiment. There's enough example code, working protocol implementations, and documentation available in the repository for a self-motivated implementer to figure out how to do what you describe, but sadly we lack enough time to provide a lot of detailed assistance in the process. Good luck!

zeefpt, 2017-04-17:
Hi again pinealservo,
Thnks again for response,
I will use the examples that repository provide and documentation and if doubts emerge i will ask you guys for help/suggestion. :)
Regards from Portugal,

zeefpt, 2017-04-17:
1st doubt. :)
From main makefile i can compile gptp and mrpd daemon.

Now im trying to make avtp_pipeline but gives me one error:

make[1]: Entering directory '/home/jp/projetos/openavnu/OpenAvnu/lib/common'
cc -O2 -g -Wall -Wextra -Wno-parentheses -I../igb -c avb.c
avb.c:34:21: fatal error: pci/pci.h: No such file or directory
 #include <pci/pci.h>

Where can i find the pci.h ?

Regards

andrew-elder, 2017-04-17:
Are you working on branch open-avb-next ?

zeefpt, 2017-04-17:
Hi,

I cloned the repository:

git clone https://github.com/AVnu/OpenAvnu
git submodule update --init --recursive

and doned make from here.

Regards

andrew-elder, 2017-04-17:
go

andrew-elder, 2017-04-17:
go
git checkout -b open-avb-next
and use that.

zeefpt, 2017-04-17:
Hi,
Installed sudo apt-get install libpci-dev and solved the problem. :)
You recommend that instead of master?

zeefpt, 2017-04-17:
Now i can compile daemons_all , lib , examples_all but in   avtp_pipeline gives me that error:

-- IGB launch time enabled
-- Native build for Linux-4.8.0-46-generic on x86_64
-- Checking for modules 'gstreamer-app-0.10;gstreamer-rtp-0.10'
--   No package 'gstreamer-app-0.10' found
--   No package 'gstreamer-rtp-0.10' found
CMake Error at platform/Linux/CMakeLists.txt:152 (MESSAGE):
  Aborting: gstreamer library not found
Call Stack (most recent call first):
  CMakeLists.txt:37 (include)

-- Configuring incomplete, errors occurred!
See also "/home/jp/projetos/openavnu/OpenAvnu/lib/avtp_pipeline/build/CMakeFiles/CMakeOutput.log".
avtp_pipeline.mk:19: recipe for target 'build/Makefile' failed
make[1]: *** [build/Makefile] Error 1
Makefile:126: recipe for target 'avtp_pipeline' failed
make: *** [avtp_pipeline] Error 2

Probably beacuse gstreamer not found.

i already tryed:
sudo apt-get install libgstreamer0.10-dev
sudo apt-get install libgstreamer-plugins-base0.10-dev

but linux is unable to locate that package. 

thks

zeefpt, 2017-04-18:
Hi,

Just installed Ubuntu 14.04 on Vbox and all compile problems were solved. Have doned make all :)
Now im testing the gptp daemon with  sudo ./run_gptp.sh eth1   like suggested in readme and got this:

"INFO     : GPTP [12:56:54:696] gPTP starting
ERROR    : GPTP [12:56:54:697] Failed to configure timestamping: Operation not supported
ERROR    : GPTP [12:56:54:697] post_init failed

ERROR    : GPTP [12:56:54:697] failed to initialize port
"

Using first script  sudo ./run_igb.sh eth1 the igb_avb mod was succefully installed.

My Dmesg is : http://pasted.co/29647e02

Anyidea what can I search/do?

Regards

andrew-elder, 2017-04-18:
You aren't running from the Vbox are you? You need to run on the host.

zeefpt, 2017-04-18:
Yes Vmware. Ok i see other thread you saying that. I will try.
Regards

pinealservo, 2017-04-18:
If you look at the avtp_pipeline.mk file in `lib/avtp_pipeline`, you'll see there are some variables you can override in the environment to change how it's built. You can either do:

    export AVB_FEATURE_GSTREAMER=0
    make avtp_pipeline

Or you can just set it on the same command line:

    CMAKE_TOOLCHAIN_FILE=generic make avtp_pipeline

You can see the options for toolchain files in `lib/avtp_pipeline/platform/Linux` as the `.cmake` files in there. And setting the gstreamer variable to 0 should keep it from searching for gstreamer, although if you build it without gstreamer you won't be able to do video streaming.

zeefpt, 2017-04-19:
Hi pinealservo,
Thnks for the tip :)
Regards

pinealservo, 2017-04-19:
If you try this and cmake gives you an error about a TEMP variable, see my comment on #579 about how you can fix it by commenting out a couple of lines.

zeefpt, 2017-05-04:
Hi again pinealservo,

Now i need the gstream to run :)
Im using buildroot and enable gstreamer 0.1 and 1 packages in makemenuconfig.
But when i do make all give me the same error:

- IGB launch time enabled
-- Native build for Linux-4.8.0-49-generic on x86_64
-- Checking for modules 'gstreamer-app-0.10;gstreamer-rtp-0.10'
--   No package 'gstreamer-app-0.10' found
--   No package 'gstreamer-rtp-0.10' found
CMake Error at platform/Linux/CMakeLists.txt:160 (MESSAGE):
  Aborting: gstreamer library not found
Call Stack (most recent call first):
  CMakeLists.txt:37 (include)

Can i change something for that in compilation aponts to my librarys?

Regards,

pinealservo, 2017-05-05:
Probably you can, but you're mostly on your own at this point. You will probably have to look into cmake's cross-compilation support and how to use its toolchain files to specify the search paths it should use. You may be using buildroot, but it's still doing a native x86_64 build and is probably looking in your system paths rather than buildroot's paths.

zeefpt, 2017-05-05:
Thanks for anwser.
Any tip where path should i look?
Regards

zeefpt, 2017-05-08:
anyhelp? :)

andrew-elder, 2017-05-08:
CMakeCache.txt is not the correct file to edit. You should edit the parent https://github.com/AVnu/OpenAvnu/blob/open-avb-next/lib/avtp_pipeline/CMakeLists.txt file, or another CMakeLists.txt file that it includes.

zeefpt, 2017-05-08:
Thks andrew-elder i will explore what you suggest :)
Regards

zeefpt, 2017-05-08:
So i have to create a /home/jp/projetos/openavnu/OpenAvnu/lib/avtp_pipeline/platform/Linux /x86_i210_linux.cmake custom for pointing to what i want?

Best regards,

zeefpt, 2017-05-08:
Hi the command cmake -**DCMAKE_TOOLCHAIN_FILE**  doesnt apeears me.
To compile all the stuff i have to use cmake or editing only the makefiles should do the work?
Regards,

zeefpt, 2017-05-08:
worked :) but i doned changing the avtp_pipeline.mk to:

PLATFORM_TOOLCHAIN ?= x86_i210_linux_test
and in the x86_i210_linux_test.cmake i change :

set ( GSTREAMER_1_0 1 )

I dont know if it is the best manner because i think it has sniffed my system and not the toolchain of buildroot. Can i do this too: 
#set ( LINUX_KERNEL_DIR "home/jp/projetos/advantech_eth_qos_minimal/buildroot-2017.02.1/output/staging/usr/include" )

I have this erros too:

Now i have this erros:
-- Pcap include dirs set to /usr/include
FATALPcap library cannot be found
-- Looking for pcap_breakloop
-- Looking for pcap_breakloop - not found
-- Looking for pcap_datalink_name_to_val
-- Looking for pcap_datalink_name_to_val - not found
-- Looking for pcap_datalink_val_to_name
-- Looking for pcap_datalink_val_to_name - not found
-- Looking for pcap_findalldevs
-- Looking for pcap_findalldevs - not found
-- Looking for pcap_freecode
-- Looking for pcap_freecode - not found
-- Looking for pcap_get_selectable_fd
-- Looking for pcap_get_selectable_fd - not found
-- Looking for pcap_lib_version
-- Looking for pcap_lib_version - not found
-- Looking for pcap_list_datalinks
-- Looking for pcap_list_datalinks - not found
-- Looking for pcap_open_dead
-- Looking for pcap_open_dead - not found
-- Looking for pcap_set_datalink
-- Looking for pcap_set_datalink - not found
-- Rawsock PCAP enabled
-- Rawsock IGB enabled
-- common IGB library included

I installed in buildroot the lib that faults but still the same error.

Really glad for your help,
Regards,
Jose

pinealservo, 2017-05-08:
Try this, from the top level directory:

    $ PLATFORM_TOOLCHAIN=x86_i210_linux_test make avtp_pipeline

The only build system for avtp_pipeline is based on cmake; most of the other subsystems in the repository have an alternate Makefile as well, but not avtp_pipeline.

In general, you should not run cmake directly from the top-level of a project. It expects to be run from a build directory and to be given the path of the source directory. The top-level Makefile rule for avtp_pipeline takes care of this for you.

Failing to find the pcap library in your buildroot is a tricky one, and my understanding of how cmake works is not perfect with respect to cross-compiling, but I think you will be able to fix this one by setting up some search paths in your `x86_i210_linux_test.cmake` file. Specifically, you'll need to set up the following variables:

    CMAKE_FIND_ROOT_PATH
    CMAKE_FIND_ROOT_PATH_MODE_PACKAGE
    CMAKE_FIND_ROOT_PATH_MODE_PROGRAM
    CMAKE_FIND_ROOT_PATH_MODE_LIBRARY
    CMAKE_FIND_ROOT_PATH_MODE_INCLUDE

All but the first one should (possibly?) be set to `ONLY` and the first one should be set to the paths that you want it to do library searches in.

Hopefully that will give you some pointers to where to look in the cmake documentation at least.

zeefpt, 2017-05-09:
Hi pinealservo,

Thanks for guiding me. :)
Now everything compiled.
My file now is:

# and another kernel sources
#set ( LINUX_KERNEL_DIR "/usr/src/kernel" )

# this one is important
SET(CMAKE_SYSTEM_NAME Linux)
#this one not so much
SET(CMAKE_SYSTEM_VERSION 1)

SET(CMAKE_SYSTEM_PROCESSOR x86_64)

# specify the cross compiler
SET(CMAKE_C_COMPILER   /home/jp/projetos/advantech_eth_qos_minimal/buildroot-2017.02.1/output/host/usr/bin/x86_64-linux-gcc)
SET(CMAKE_CXX_COMPILER /home/jp/projetos/advantech_eth_qos_minimal/buildroot-2017.02.1/output/host/usr/bin/x86_64-linux-c++)

# where is the target environment

set(CMAKE_FIND_ROOT_PATH /home/jp/projetos/advantech_eth_qos_minimal/buildroot-2017.02.1/output/staging/)

#set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)

# search for programs in the build host directories
set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)

# for libraries and headers in the target directories
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)

#MESSAGE (  "Passei aqui" )

# build configuration
set ( OPENAVB_HAL      "x86_i210" )
set ( OPENAVB_OSAL     "Linux" )
set ( OPENAVB_TCAL     "GNU" )
set ( OPENAVB_PLATFORM "${OPENAVB_HAL}-${OPENAVB_OSAL}" )

# Platform Additions
set ( PLATFORM_INCLUDE_DIRECTORIES 	
	${CMAKE_SOURCE_DIR}/platform/x86_i210/include 
	${CMAKE_SOURCE_DIR}/../igb 
	${CMAKE_SOURCE_DIR}/openavb_common
	${CMAKE_SOURCE_DIR}/../../daemons/common
	${CMAKE_SOURCE_DIR}/../../daemons/mrpd
)

set ( PLATFORM_LINK_DIRECTORIES
	${CMAKE_SOURCE_DIR}/../igb
)

set ( PLATFORM_LINK_LIBRARIES
	igb
)

# TODO_OPENAVB : need this?
# Set platform specific define
#set ( PLATFORM_DEFINE "AVB_DELAY_TWEAK_USEC=15" )

set ( GSTREAMER_1_0 1 )
set ( AVB_FEATURE_PCAP 1 )

-----------------------------
I have questions :)
1- If i put the set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER) to ONLY the program gives me that error:

CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:52 (find_package_handle_standard_args)
  platform/Linux/CMakeLists.txt:134 (find_package)
  CMakeLists.txt:37 (include)

Resolved puting NEVER!

2- Again, the original makefile of  /home/jp/projetos/openavnu/OpenAvnu/Makefile is overwriten at end of avtp_pipeline build is because of what you said: ?

Try this, from the top level directory:

$ PLATFORM_TOOLCHAIN=x86_i210_linux_test make avtp_pipeline

right?

2.1 - Using **make all** builds all the stuff ok but in the end overwrites the toplever Makefile again :) is normal?

3 - Can i change the cmake to install the stuff in my buildroot target system using sysroot? (http://docs.w3cub.com/cmake~3.5/variable/cmake_sysroot/#variable:CMAKE_SYSROOT)

Cmake install the binaries in 
-- Installing: /home/jp/projetos/openavnu/OpenAvnu/lib/avtp_pipeline/build/lib/libintf_echo.a

Many thnks to all,
Best Regards

pinealservo, 2017-05-10:
I am really not an expert on cmake; I am cross-compiling with a system based on bitbake/openembedded tools rather than buildroot, and the cmake suggestions I gave were some stuff that worked for me that I found via mostly trial and error. I'm glad you're making progress, but I'm afraid I can't offer much more help than I already have!

zeefpt, 2017-05-10:
Hi thaks for response.

I discover where in buildroot have sysroot :  lrwxrwxrwx   1 jp jp  117 abr 20 13:23 staging -> /home/jp/projetos/advantech_eth_qos_minimal/buildroot-2017.02.1/output/host/usr/x86_64-buildroot-linux-uclibc/sysroot/

so i insert in my toolchainfile:

SET(CMAKE_SYSROOT /home/jp/projetos/advantech_eth_qos_minimal/buildroot-2017.02.1/output/staging)

but when i do make avtp_pipeline cmake are installing in :  (instead of my sysroot dir)

[100%] Linking C executable openavb_harness
[100%] Built target openavb_harness
Install the project...
-- Install configuration: "Release"
-- Installing: /home/jp/projetos/openavnu/OpenAvnu/lib/avtp_pipeline/build/lib/libavbTl.a
-- Installing: /home/jp/projetos/openavnu/OpenAvnu/lib/avtp_pipeline/build/lib/libmap_ctrl.a
-- Installing: /home/jp/projetos/openavnu/OpenAvnu/lib/avtp_pipeline/build/lib/libmap_mjpeg.a
-- Installing: /home/jp/projetos/openavnu/OpenAvnu/lib/avtp_pipeline/build/lib/libmap_mpeg2ts.a
-- Installing: /home/jp/projetos/openavnu/OpenAvnu/lib/avtp_pipeline/build/lib/libmap_null.a
-- Installing: /home/jp/projetos/openavnu/OpenAvnu/lib/avtp_pipeline/build/lib/libmap_pipe.a
-- Installing: /home/jp/projetos/openavnu/OpenAvnu/lib/avtp_pipeline/build/lib/libmap_aaf_audio.a
-- Installing: /home/jp/projetos/openavnu/OpenAvnu/lib/avtp_pipeline/build/lib/libmap_uncmp_audio.a

etc...

Best Regards,

zeefpt, 2017-05-15:
Hi Folks,
Another doubt.
I try to test in my target system the gb_avb.ko module but give me that error:
igb_avb: version magic '4.8.0-51-generic SMP mod_unload modversions ' should be '4.11.0-rc6 SMP mod_unload '
igb_avb: version magic '4.8.0-51-generic SMP mod_unload modversions ' should be '4.11.0-rc6 SMP mod_unload '
---
Because im not crosscompiling right :)
So analysing the kmod/igb makefile i susbtitute the KSRC for:
KSRC=/home/jp/projetos/advantech_eth_qos_minimal/buildroot-2017.02.1/output/build/linux-vETH20170418

and now trying to compile gives me that erro:

jp@JP:~/projetos/openavnu/OpenAvnu/kmod/igb$ make 
make -C /home/jp/projetos/advantech_eth_qos_minimal/buildroot-2017.02.1/output/build/linux-vETH20170418 SUBDIRS=/home/jp/projetos/openavnu/OpenAvnu/kmod/igb modules
make[1]: Entering directory '/home/jp/projetos/advantech_eth_qos_minimal/buildroot-2017.02.1/output/build/linux-vETH20170418'
  CC [M]  /home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.o
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:222:13: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
   .fault  = igb_vm_fault
             ^~~~~~~~~~~~
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:222:13: note: (near initialization for ‘igb_mmap_ops.fault’)
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:299:15: error: variable ‘igb_miscdev’ has initializer but incomplete type
 static struct miscdevice igb_miscdev = {
               ^~~~~~~~~~
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:300:3: error: unknown field ‘minor’ specified in initializer
   .minor = MISC_DYNAMIC_MINOR,
   ^
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:300:12: error: ‘MISC_DYNAMIC_MINOR’ undeclared here (not in a function)
   .minor = MISC_DYNAMIC_MINOR,
            ^~~~~~~~~~~~~~~~~~
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:300:12: warning: excess elements in struct initializer
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:300:12: note: (near initialization for ‘igb_miscdev’)
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:301:3: error: unknown field ‘name’ specified in initializer
   .name = "igb_avb",
   ^
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:301:11: warning: excess elements in struct initializer
   .name = "igb_avb",
           ^~~~~~~~~
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:301:11: note: (near initialization for ‘igb_miscdev’)
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:302:3: error: unknown field ‘fops’ specified in initializer
   .fops = &igb_fops,
   ^
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:302:11: warning: excess elements in struct initializer
   .fops = &igb_fops,
           ^
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:302:11: note: (near initialization for ‘igb_miscdev’)
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c: In function ‘igb_init_module’:
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:394:2: error: implicit declaration of function ‘misc_register’ [-Werror=implicit-function-declaration]
  misc_register(&igb_miscdev);
  ^~~~~~~~~~~~~
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c: In function ‘igb_exit_module’:
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:419:2: error: implicit declaration of function ‘misc_deregister’ [-Werror=implicit-function-declaration]
  misc_deregister(&igb_miscdev);
  ^~~~~~~~~~~~~~~
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c: At top level:
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:2351:21: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .ndo_set_vf_vlan = igb_ndo_set_vf_vlan,
                     ^~~~~~~~~~~~~~~~~~~
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:2351:21: note: (near initialization for ‘igb_netdev_ops.ndo_set_vf_vlan’)
/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.c:299:26: error: storage size of ‘igb_miscdev’ isn’t known
 static struct miscdevice igb_miscdev = {
                          ^~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:294: recipe for target '/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.o' failed
make[2]: *** [/home/jp/projetos/openavnu/OpenAvnu/kmod/igb/igb_main.o] Error 1
Makefile:1492: recipe for target '_module_/home/jp/projetos/openavnu/OpenAvnu/kmod/igb' failed
make[1]: *** [_module_/home/jp/projetos/openavnu/OpenAvnu/kmod/igb] Error 2
make[1]: Leaving directory '/home/jp/projetos/advantech_eth_qos_minimal/buildroot-2017.02.1/output/build/linux-vETH20170418'
Makefile:266: recipe for target 'default' failed
make: *** [default] Error 2

zeefpt, 2017-05-16:
Resolved the problem.
Porting the functions from old kernel to new kernel.
Now im ready for teste hehe
Thnks averyone

closed 2017-05-16 by andrew-elder

zeefpt, 2017-05-19:
Hi again Folks,

I have all the stuff compiled, ready for testing. :) But now i have some problems.

I have 2 cards in my embebed system:

I211 : use igb driver
I219: use e1000e driver

I remove the igb original driver and put the igb_avb. He is inserted but after that nothing happens. Interface is not up.

# rmmod igb
igb 0000:02:00.0: removed PHC on eth1_i211
# modprobe i2c_algo_bit
modprobe: module i2c_algo_bit not found in modules.dep
#
# modprobe dca
modprobe: module dca not found in modules.dep
# modprobe ptp
modprobe: module ptp not found in modules.dep
# ls
igb_avb.ko
# insmod igb_avb.ko
igb_avb: loading out-of-tree module taints kernel.
Intel(R) Gigabit Ethernet Network Driver - version 5.3.2_AVB
Copyright (c) 2007-2015 Intel Corporation.
# ethtool -i eth1_i211
Cannot get driver information: No such device
# ethtool -i eth1_i219
Cannot get driver information: No such device
# ethtool -i eth2_i219
driver: e1000e
version: 3.2.6-k
firmware-version: 0.8-4
expansion-rom-version:
bus-info: 0000:00:1f.6
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: no

igb_avb driver is compatible with I211 card?

Could i mount a talker and listener without the module or it mandatory that have to insert that?

Best regards,