Avnu/OpenAvnu #579: Openavb on i.MX6Q sabresd NIC issue

stujeffrey016, 2017-04-13:
Hi there, 
I'm using i.MX6Q and trying to port openavb on my platform. First, I'm confused that what's different between the original igb driver( 4.1.15 kernel source code ) and igb_avb. The Qav(FQTSS) part is built in hardware or software(igb_main.c) ? Second, I knew that libigb is only for I210 NIC. It means that I need to build a new API specific for imx6q NIC, and connect it to the user space application(openavb_harness). I want to know what's different between I210 and my NIC card, do I need to modify my driver code in the kernel source? Adding the Qav part or something like that. If you have any experience or information, please that me know, thanks.

pinealservo, 2017-04-14:
The igb_avb driver is based on a much older version of the igb driver. The primary feature it offers now is that it exposes the management of one of its hardware queues directly to userspace. This means that you can write directly to the control descriptors, completely bypassing the kernel stack for streaming traffic. It also is set up to use the built-in FQTSS shaper as well as a launch-time-based shaper. The launch-time shaper allows you to load a bunch of packets into the streaming queue at once, but have them launch at intervals of your choosing rather than in a burst. See the simple_talker and simple_listener examples and the igb library code for examples, although they won't be very helpful on an i.MX6Q board.

I have recently merged some changes to the avtp_pipeline library that allow you to build it without using the i210 support it previously required to be present. There is no special driver support for streaming, but depending on what you need to do, the Linux stack's built-in traffic shaping can do a reasonable job itself. You should be able to cross-compile and run the open-avb-next branch code as it is; you can definitely get a talker and listener based entirely on avtp_pipeline and the OpenAvnu gptp and mrp daemons running back-to-back or across a standards-compliant switch. You may have to tweak a bit in various places to get it running *well*, though, especially against a device that expects precisely-timed presentation timestamps.

Several people seem to be wanting to implement AVB on the i.MX6 family lately; unfortunately, only the iMX6SoloX has their newer Ethernet module that has full AVB support. The older ones only have 1588 timestamping support, while the newer ones (i.MX6SoloX, i.MX7, etc.) include a FQTSS shaper as well as launch-time shaping, similar to the capabilities of the i210. There's not currently any support for those features in any mainline Linux kernel drivers that I'm aware of.

Let me know if you have any trouble getting things built.

stujeffrey016, 2017-04-16:
Hi @pinealservo ,
  Thanks for your helping. You mean that I don't need to compiler the /lib/igb to build Talker/Listener on my platform?

pinealservo, 2017-04-18:
That's right.  From the top-level, run:

    PLATFORM_TOOLCHAIN=generic make avtp_pipeline

For a particular cross-compile environment, you may have to set up your own cmake toolchain file.

Oh; you may also have to remove the 'lib' dependency in the top-level Makefile for the avtp_pipeline build, or just run the same command it does manually. The 'lib' build tries to build the igb library, which probably won't work in a cross-build environment, but is required for the default i210 build.

stujeffrey016, 2017-04-19:
Hi @pinealservo ,
I set the -DCMAKE_TOOLCHAIN_FILE to /platform/Linux/generic.cmake in avtp_pipeline.mk file, and run the generic make avtp_pipeline, here comes the error message:

> make[1]: Leaving directory `/home/linaro/OpenAvnu-open-avb-next/lib/common'
make -s -C lib/avtp_pipeline -f avtp_pipeline.mk
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Build type is Release
-- CMAKE_SOURCE_DIR              : /home/linaro/OpenAvnu-open-avb-next/lib/avtp_pipeline
-- AVB_SRC_DIR                   : /home/linaro/OpenAvnu-open-avb-next/lib/avtp_pipeline
-- AVB_HAL_DIR                   : /home/linaro/OpenAvnu-open-avb-next/lib/avtp_pipeline/platform/generic
-- AVB_OSAL_DIR                  : /home/linaro/OpenAvnu-open-avb-next/lib/avtp_pipeline/platform/Linux
-- AVB_INSTALL_BIN_DIR           : /home/linaro/OpenAvnu-open-avb-next/lib/avtp_pipeline/build/bin
-- AVB_INSTALL_LIB_DIR           : /home/linaro/OpenAvnu-open-avb-next/lib/avtp_pipeline/build/lib
-- SDK_INSTALL_SDK_INTF_MOD_DIR  : /home/linaro/OpenAvnu-open-avb-next/lib/avtp_pipeline/build/sdk_intf_mod
-- SDK_INSTALL_SDK_MAP_MOD_DIR   : /home/linaro/OpenAvnu-open-avb-next/lib/avtp_pipeline/build/sdk_map_mod
-- SDK_INSTALL_SDK_EAVB_DIR      : /home/linaro/OpenAvnu-open-avb-next/lib/avtp_pipeline/build/sdk_eavb
-- IGB launch time enabled
-- Native build for Linux-4.1.15 on armv7l
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'glib-2.0'
--   found glib-2.0, version 2.40.2
-- checking for modules 'gstreamer-app-0.10;gstreamer-rtp-0.10'
--   found gstreamer-app-0.10, version 0.10.36
--   found gstreamer-rtp-0.10, version 0.10.36
-- checking for module 'alsa'
--   found alsa, version 1.0.27.2
-- TL with Endpoint
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.6") 
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
TEMP
    linked by target "rawsock_rx" in directory /home/linaro/OpenAvnu-open-avb-next/lib/avtp_pipeline
    linked by target "rawsock_tx" in directory /home/linaro/OpenAvnu-open-avb-next/lib/avtp_pipeline
    linked by target "openavb_harness" in directory /home/linaro/OpenAvnu-open-avb-next/lib/avtp_pipeline/platform/Linux/avb_host
    linked by target "openavb_host" in directory /home/linaro/OpenAvnu-open-avb-next/lib/avtp_pipeline/platform/Linux/avb_host
-- Configuring incomplete, errors occurred!
See also "/home/linaro/OpenAvnu-open-avb-next/lib/avtp_pipeline/build/CMakeFiles/CMakeOutput.log".
make[1]: *** [build/Makefile] Error 1
make: *** [avtp_pipeline] Error 2

This error show in both PC platform and i.mx6q platform. What does it mean? Which cmake file I should modified?

pinealservo, 2017-04-19:
Hmm, it looks like that's another change I need to propagate over.  Find the following lines in `platform/Linux/CMakeLists.txt` and comment them out (they should be 169-170):

    find_library(TEMP NAMES ${PLATFORM_LINK_LIBRARIES} PATHS ${PLATFORM_LINK_DIRECTORIES} )
    set ( PLATFORM_LINK_LIBRARIES ${TEMP} )

After that, it should build in the host environment, but we may need to set up a platform-specific toolchain file for you for the i.MX6 build if it can't find some of the libraries it needs. Let me know if there are more errors and we'll tackle it from there.

stujeffrey016, 2017-04-20:
Hi @pinealservo ,
I comment them out and re-compile again, but still got the error message in both host and i.mx6 platform. Those c files(openavb_ether_hal.c, openavb_time_osal.c, etc.) are not in the path which CMakeFile.txt expected, and they all include the igb.h file, some of them seems like the core of openavb. What should I need to do? 

> mkdir -p lib/igb && make  subdir=lib/igb  -C lib/igb 
make[1]: Entering directory `/home/jeffrey/OpenAvnu-open-avb-next/lib/igb'
cc -c   igb.c
rm -f libigb.a
ar rcs libigb.a igb.o
ranlib libigb.a
make[1]: Leaving directory `/home/jeffrey/OpenAvnu-open-avb-next/lib/igb'
mkdir -p lib/common && make  subdir=lib/common  -C lib/common 
make[1]: Entering directory `/home/jeffrey/OpenAvnu-open-avb-next/lib/common'
cc -O2 -g -Wall -Wextra -Wno-parentheses -DAVB_FEATURE_IGB=1   -c -o avb_avtp.o avb_avtp.c
cc -O2 -g -Wall -Wextra -Wno-parentheses -DAVB_FEATURE_IGB=1   -c -o avb_gptp.o avb_gptp.c
cc -O2 -g -Wall -Wextra -Wno-parentheses -DAVB_FEATURE_IGB=1 -I../igb  -c -o avb_igb.o avb_igb.c
make[1]: Leaving directory `/home/jeffrey/OpenAvnu-open-avb-next/lib/common'
make -s -C lib/avtp_pipeline -f avtp_pipeline.mk
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Build type is Release
-- CMAKE_SOURCE_DIR              : /home/jeffrey/OpenAvnu-open-avb-next/lib/avtp_pipeline
-- AVB_SRC_DIR                   : /home/jeffrey/OpenAvnu-open-avb-next/lib/avtp_pipeline
-- AVB_HAL_DIR                   : /home/jeffrey/OpenAvnu-open-avb-next/lib/avtp_pipeline/platform/generic
-- AVB_OSAL_DIR                  : /home/jeffrey/OpenAvnu-open-avb-next/lib/avtp_pipeline/platform/Linux
-- AVB_INSTALL_BIN_DIR           : /home/jeffrey/OpenAvnu-open-avb-next/lib/avtp_pipeline/build/bin
-- AVB_INSTALL_LIB_DIR           : /home/jeffrey/OpenAvnu-open-avb-next/lib/avtp_pipeline/build/lib
-- SDK_INSTALL_SDK_INTF_MOD_DIR  : /home/jeffrey/OpenAvnu-open-avb-next/lib/avtp_pipeline/build/sdk_intf_mod
-- SDK_INSTALL_SDK_MAP_MOD_DIR   : /home/jeffrey/OpenAvnu-open-avb-next/lib/avtp_pipeline/build/sdk_map_mod
-- SDK_INSTALL_SDK_EAVB_DIR      : /home/jeffrey/OpenAvnu-open-avb-next/lib/avtp_pipeline/build/sdk_eavb
-- IGB launch time enabled
-- Native build for Linux-4.2.0-42-generic on x86_64
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'glib-2.0'
--   found glib-2.0, version 2.40.2
-- checking for modules 'gstreamer-app-0.10;gstreamer-rtp-0.10'
--   found gstreamer-app-0.10, version 0.10.36
--   found gstreamer-rtp-0.10, version 0.10.36
-- checking for module 'alsa'
--   found alsa, version 1.0.27.2
-- TL with Endpoint
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configuring done
CMake Error at platform/Linux/CMakeLists.txt:260 (add_library):
  Cannot find source file:
    /home/jeffrey/OpenAvnu-open-avb-next/lib/avtp_pipeline/platform/generic/openavb_ether_hal.c
  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx
Call Stack (most recent call first):
  CMakeLists.txt:37 (include)
-- Build files have been written to: /home/jeffrey/OpenAvnu-open-avb-next/lib/avtp_pipeline/build
make[1]: *** [build/Makefile] Error 1
make: *** [avtp_pipeline] Error 2

pinealservo, 2017-04-20:
Ah, I'm sorry, it turns out that the changes to support the "generic" toolchain build are in my fork under the "task_optional_deps" branch and not yet in the "open-avb-next" branch. I have been staging the pull requests in chunks and forgot there was one left to do! You can pull it from there now, or wait a few days until I get the pull request ready and it gets merged.

stujeffrey016, 2017-04-21:
Hi @pinealservo ,
Thanks for your helping!
If I have any question, I'll let you know.

stujeffrey016, 2017-04-23:
Hi @pinealservo ,
I compiled the generic branch successfully, and got some questions:
1. The gptp part is depend on 1588 clock. If my NIC got the hardware support, it will work fine, right?
2. Should I modify the Makefile in the top-level? Cause I saw that it still compiled the lib/igb which should not be needed.
> mkdir -p lib/igb && make  subdir=lib/igb  -C lib/igb 
make[1]: Entering directory `/home/jeffrey/OpenAVB-NTPU/lib/igb'
cc -c   igb.c
rm -f libigb.a
ar rcs libigb.a igb.o
ranlib libigb.a
make[1]: Leaving directory `/home/jeffrey/OpenAVB-NTPU/lib/igb'
mkdir -p lib/common && make  subdir=lib/common  -C lib/common 
make[1]: Entering directory `/home/jeffrey/OpenAVB-NTPU/lib/common'
cc -O2 -g -Wall -Wextra -Wno-parentheses -DAVB_FEATURE_IGB=1   -c -o avb_avtp.o avb_avtp.c
cc -O2 -g -Wall -Wextra -Wno-parentheses -DAVB_FEATURE_IGB=1   -c -o avb_gptp.o avb_gptp.c
cc -O2 -g -Wall -Wextra -Wno-parentheses -DAVB_FEATURE_IGB=1 -I../igb  -c -o avb_igb.o avb_igb.c
make[1]: Leaving directory `/home/jeffrey/OpenAVB-NTPU/lib/common'
make -s -C lib/avtp_pipeline -f avtp_pipeline.mk

3. Now I want to port the openavb-generic to the i.mx6q platform, how to set up a platform-specific toolchain file? Which CMakeFile I need to modify?

andrew-elder, 2017-04-23:
1. above - will depend on software driver support as well as hardware support.
3. above - is beyond the level of support volunteers replying to messages should be expected to provide. I assume you have already asked your silicon vendor? Someone may be able to provide hints, but I would not expect a complete working solution. I assume you have set up a cross platform toolchain already?

stujeffrey016, 2017-04-24:
Hi @andrew-elder ,
Thanks for your information. I removed the lib command in Makefile and got the second question's answer. I'll check the information of cross compiler issue.

stujeffrey016, 2017-04-27:
Hi @pinealservo ,
This week I put the execution file on my i.mx6q. It built up successfully, but there are some error message show up:

> INFO     : GPTP [00:17:55:232] gPTP starting
ERROR    : GPTP [00:17:55:237] Using clock device: /dev/ptp0
STATUS   : GPTP [00:17:55:237] Starting PDelay
STATUS   : GPTP [00:17:55:272] AsCapable: Enabled
EXCEPTION: GPTP [00:17:56:000] PDelay Response Receipt Timeout
STATUS   : GPTP [00:17:56:000] AsCapable: Disabled
ERROR    : GPTP [00:17:56:001] >>> Received PDelay FUP has different seqID than the PDelay request (41/42)
ERROR    : GPTP [00:17:56:002] >>> Received PDelay followup but no REQUEST exists
EXCEPTION: GPTP [00:17:56:007] PDelay Response Receipt Timeout
STATUS   : GPTP [00:17:56:020] AsCapable: Enabled
STATUS   : GPTP [00:17:56:046] Switching to Slave
STATUS   : GPTP [00:17:56:046] New Grandmaster "00:51:00:FF:FE:10:00:00" (previous "00:00:00:00:00:00:00:00")
EXCEPTION: GPTP [00:17:56:390] PDelay Response Receipt Timeout
STATUS   : GPTP [00:17:56:390] AsCapable: Disabled
ERROR    : GPTP [00:17:56:400] >>> Received PDelay FUP has different seqID than the PDelay request (64/65)
ERROR    : GPTP [00:17:56:401] >>> Received PDelay followup but no REQUEST exists
EXCEPTION: GPTP [00:17:56:405] PDelay Response Receipt Timeout
STATUS   : GPTP [00:17:56:407] AsCapable: Enabled
EXCEPTION: GPTP [00:17:57:567] PDelay Response Receipt Timeout
STATUS   : GPTP [00:17:57:567] AsCapable: Disabled
ERROR    : GPTP [00:17:57:571] >>> Received PDelay FUP has different seqID than the PDelay request (132/133)
ERROR    : GPTP [00:17:57:572] >>> Received PDelay followup but no REQUEST exists
EXCEPTION: GPTP [00:17:57:577] PDelay Response Receipt Timeout
STATUS   : GPTP [00:17:57:590] AsCapable: Enabled
EXCEPTION: GPTP [00:17:58:321] PDelay Response Receipt Timeout
STATUS   : GPTP [00:17:58:321] AsCapable: Disabled
STATUS   : GPTP [00:17:58:322] AsCapable: Enabled
Segmentation fault

Do you know what's going on? Why the AsCapable message goes up and down? I'll check the source code and try to fit my platform. If you got some information about this issue, please let me know. Thanks.

andrew-elder, 2017-04-27:
What is the imx6 Ethernet port connected to?

stujeffrey016, 2017-04-27:
Hi @andrew-elder 
An AVB standard switch. It works fine with the OpenAVB on PC platform.

pinealservo, 2017-04-27:
Try setting your announce, sync, and delay request intervals to reasonable values on the command line. If you look at the sequence ID numbers and compare with the timestamps, you'll see they are currently going a bit fast. There must be something odd going on with the defaults.

andrew-elder, 2017-04-27:
Maybe the system timers are running at a weird rate?

andrew-elder, 2017-04-27:
Or, could gPTP be using a timer in a mode that isn't implemented on this platform?

christopher-s-hall, 2017-04-27:
gPTP is seeing a sequence ID in the response that doesn't match that of the request. You can print out the sequence id that is being received. The error log should give you a pretty good idea of where is the code that comparison is made. You might also use wireshark - with port mirroring if necessary. The second error line just "confirms" that the response was discarded and doesn't provide any additional information.

@andrew-elder could be correct as well - meaning there's some race condition resulting from too fast messaging. Wireshark would be useful there also.

stujeffrey016, 2017-04-28:
Hi Everyone,
Thanks for your information, I'll check the timer interval setting and sequence ID. Use Wireshark with port mirroring on my PC platform. If I got any information, I'll let you know.

stujeffrey016, 2017-05-11:
Hi @pinealservo & @andrew-elder & @christopher-s-hall,

I finally set up the interval of pdelay message, I use `./daemon_cl eth0 -INITPDELAY 0.5` command, and the sequence ID problem fixed, but there's another issue. My fec NIC has the 1588 protocol capability, but the error message show up like this:

> groupadd: group 'ptp' already exists
INFO     : GPTP [00:31:53:575] gPTP starting
ERROR    : GPTP [00:31:53:581] Using clock device: /dev/ptp0
STATUS   : GPTP [00:31:53:581] Starting PDelay
ERROR    : GPTP [00:31:53:616] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [00:31:54:625] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [00:31:55:625] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
STATUS   : GPTP [00:31:56:582] *** Announce Timeout Expired - Becoming Master
STATUS   : GPTP [00:31:56:582] New Grandmaster "00:C0:08:FF:FE:88:B2:DE" (previous "00:00:00:00:00:00:00:00")
ERROR    : GPTP [00:31:56:626] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [00:31:57:627] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [00:31:58:643] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [00:31:59:644] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [00:32:00:644] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [00:32:01:645] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [00:32:02:645] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [00:32:03:646] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [00:32:04:646] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [00:32:05:647] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
^CERROR    : GPTP [00:32:06:024] Exiting on 2

I monitor the switch port to another port, and use wireshark to catch the packet. I saw that my i.mx6q sent the pdelay_request, pdelay_response and pdelay_response_followup packet correctly. But there's no announce, sync and follow_up packet. Do you know something about that? How can I fix the AsCapable issue?

Very appreciate your helping.
Jeffrey.

andrew-elder, 2017-05-11:
I would modify 
```
ERROR : GPTP [00:31:53:616] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
```
to print out the neighborPropDelayThresh to make sure it initialized properly. Also the link delay of 0 does not seem to be correct. You need to turn on logging of the pDelay transaction.

stujeffrey016, 2017-05-15:
Hi @andrew-elder ,

Thanks for your helping, but how to turn on the logging of the pDelay transaction? 
By the way, I use `./damon_cl eth0 -V` to run the program, it shows that my device switching to slave, but didn't print the GM address. I also use Wireshark to check my packet, it seems like the program only do the pDelay part, and ignore the BMCA (best master clock algorithm) and rateRate part. How to make it run the whole process of IEEE 802.1AS?

Jeffrey.

andrew-elder, 2017-05-15:
I would look in the source code and figure out  where the logging was and turn it on. The code does run IEEE 802.1AS by default, so there is nothing to turn on there.

pinealservo, 2017-05-15:
Until your device believes its link partner is ASCapable, it is not supposed to do BMCA or send Sync packets. So bb you'll have to get past the neighborPropDelayThresh problem before you'll see the other behavior.

The neighborPropDelayThresh is 800ns unless you have set up an ini file for configuration, in which case it will use the one you set there. If it's using the ini file, you should see all the values read from the config file at startup. If you have a config file but haven't set some of the values in it, that might cause some problems.

The `-V` flag is not for verbosity; it enables the automotive profile mode. There's no BMCA at all in that mode, and you'll have to use other flags to set it in master or slave mode. To see additional log levels, you'll have to turn them on by modifying `daemons/gptp/common/gptp_log.hpp` to enable the extra log levels you want.

One last thing you may or may not already know: the arguments like INITPDELAY are expressed as exponents (because they're expressed that way in the protocol definition), so that the actual interval will be 2^exp seconds. Passing 0.5 apparently does something, but it's not really a valid value--instead of half a second, you're requesting the square root of 2 seconds, and it's probably just being read as 0 anyway. Instead, if you want less than a second, you should use a negative exponent, such as -1 for 2^-1 or 0.5 seconds.

stujeffrey016, 2017-05-18:
Hi @pinealservo & @andrew-elder,

Did the error message `ERROR : GPTP [00:31:53:616] Link delay 0 beyond neighborPropDelayThresh; not AsCapable` means that my neighborPrppDelayThresh use the wrong value? Or I got somethng wrong in the hardware device? How to know what value I should set in ini file?
I also turn on the debug_log to see what's wrong. This is what I got:
> INFO : GPTP [08:05:15:619] gPTP starting
INFO : GPTP [08:05:15:625] priority1 = 248
INFO : GPTP [08:05:15:625] announceReceiptTimeout: 3
INFO : GPTP [08:05:15:625] syncReceiptTimeout: 3
INFO : GPTP [08:05:15:625] phy_delay_gb_tx: 184
INFO : GPTP [08:05:15:625] phy_delay_gb_rx: 382
INFO : GPTP [08:05:15:625] phy_delay_mb_tx: 1044
INFO : GPTP [08:05:15:625] phy_delay_mb_rx: 2133
INFO : GPTP [08:05:15:625] neighborPropDelayThresh: 2133
INFO : GPTP [08:05:15:626] syncReceiptThreshold: 8
ERROR : GPTP [08:05:15:626] Using clock device: /dev/ptp0
DEBUG : GPTP [08:05:15:626] [../../common/ieee1588port.cpp:518] Received POWERUP/INITIALIZE event
STATUS : GPTP [08:05:15:626] Starting PDelay
DEBUG : GPTP [08:05:15:627] [../src/linux_hal_common.cpp:251] Interface eth0 is up
DEBUG : GPTP [08:05:15:659] [../../common/ieee1588port.cpp:882] PDELAY_INTERVAL_TIMEOUT_EXPIRES occured
DEBUG : GPTP [08:05:15:660] [../../common/ieee1588port.cpp:910] *** Sent PDelay Request message
DEBUG : GPTP [08:05:15:660] [../../common/ieee1588port.cpp:937] PDelay Request message, Timestamp 1494938804 (sec) 974312134 (ns), seqID 0
DEBUG : GPTP [08:05:15:660] [../../common/ieee1588port.cpp:967] Schedule PDELAY_RESP_RECEIPT_TIMEOUT_EXPIRES, PDelay interval 129, wait_time 0, timeout 5000000
DEBUG : GPTP [08:05:15:660] [../../common/ptp_message.cpp:288] *** Received PDelay Response message, Timestamp 1494938804 (sec) 974311568 (ns), seqID 0
DEBUG : GPTP [08:05:15:661] [../../common/ptp_message.cpp:347] *** Received PDelay Response FollowUp message
ERROR : GPTP [08:05:15:661] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
DEBUG : GPTP [08:05:15:665] [../../common/ieee1588port.cpp:882] PDELAY_INTERVAL_TIMEOUT_EXPIRES occured
DEBUG : GPTP [08:05:15:677] [../../common/ieee1588port.cpp:910] *** Sent PDelay Request message
DEBUG : GPTP [08:05:15:677] [../../common/ieee1588port.cpp:937] PDelay Request message, Timestamp 1494938804 (sec) 974312134 (ns), seqID 1
DEBUG : GPTP [08:05:15:677] [../../common/ieee1588port.cpp:967] Schedule PDELAY_RESP_RECEIPT_TIMEOUT_EXPIRES, PDelay interval 129, wait_time 0, timeout 5000000
DEBUG : GPTP [08:05:15:678] [../../common/ptp_message.cpp:288] *** Received PDelay Response message, Timestamp 1494938804 (sec) 974311568 (ns), seqID 1
DEBUG : GPTP [08:05:15:678] [../../common/ptp_message.cpp:347] *** Received PDelay Response FollowUp message
ERROR : GPTP [08:05:15:678] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
DEBUG : GPTP [08:05:15:682] [../../common/ieee1588port.cpp:882] PDELAY_INTERVAL_TIMEOUT_EXPIRES occured
DEBUG : GPTP [08:05:15:694] [../../common/ieee1588port.cpp:910] *** Sent PDelay Request message
DEBUG : GPTP [08:05:15:694] [../../common/ieee1588port.cpp:937] PDelay Request message, Timestamp 1494938804 (sec) 974312134 (ns), seqID 2
DEBUG : GPTP [08:05:15:694] [../../common/ieee1588port.cpp:967] Schedule PDELAY_RESP_RECEIPT_TIMEOUT_EXPIRES, PDelay interval 129, wait_time 0, timeout 5000000
DEBUG : GPTP [08:05:15:695] [../../common/ptp_message.cpp:288] *** Received PDelay Response message, Timestamp 1494938804 (sec) 974311568 (ns), seqID 2
DEBUG : GPTP [08:05:15:695] [../../common/ptp_message.cpp:347] *** Received PDelay Response FollowUp message
ERROR : GPTP [08:05:15:695] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
DEBUG : GPTP [08:05:15:700] [../../common/ieee1588port.cpp:882] PDELAY_INTERVAL_TIMEOUT_EXPIRES occured
DEBUG : GPTP [08:05:15:711] [../../common/ieee1588port.cpp:910] *** Sent PDelay Request message
DEBUG : GPTP [08:05:15:712] [../../common/ieee1588port.cpp:937] PDelay Request message, Timestamp 1494938804 (sec) 974312134 (ns), seqID 3
DEBUG : GPTP [08:05:15:712] [../../common/ieee1588port.cpp:967] Schedule PDELAY_RESP_RECEIPT_TIMEOUT_EXPIRES, PDelay interval 129, wait_time 0, timeout 5000000
DEBUG : GPTP [08:05:15:712] [../../common/ptp_message.cpp:288] *** Received PDelay Response message, Timestamp 1494938804 (sec) 974311568 (ns), seqID 3
DEBUG : GPTP [08:05:15:712] [../../common/ptp_message.cpp:347] *** Received PDelay Response FollowUp message
ERROR : GPTP [08:05:15:712] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
DEBUG : GPTP [08:05:15:717] [../../common/ieee1588port.cpp:882] PDELAY_INTERVAL_TIMEOUT_EXPIRES occured
DEBUG : GPTP [08:05:15:729] [../../common/ieee1588port.cpp:910] *** Sent PDelay Request message
DEBUG : GPTP [08:05:15:729] [../../common/ieee1588port.cpp:937] PDelay Request message, Timestamp 1494938804 (sec) 974312134 (ns), seqID 4
DEBUG : GPTP [08:05:15:729] [../../common/ieee1588port.cpp:967] Schedule PDELAY_RESP_RECEIPT_TIMEOUT_EXPIRES, PDelay interval 129, wait_time 0, timeout 5000000
DEBUG : GPTP [08:05:15:729] [../../common/ptp_message.cpp:288] *** Received PDelay Response message, Timestamp 1494938804 (sec) 974311568 (ns), seqID 4
EXCEPTION: GPTP [08:05:15:734] PDelay Response Receipt Timeout
DEBUG : GPTP [08:05:15:734] [../../common/ieee1588port.cpp:882] PDELAY_INTERVAL_TIMEOUT_EXPIRES occured
DEBUG : GPTP [08:05:15:737] [../../common/ptp_message.cpp:347] *** Received PDelay Response FollowUp message
[... cut ...]
DEBUG : GPTP [08:05:17:312] [../../common/ieee1588port.cpp:967] Schedule PDELAY_RESP_RECEIPT_TIMEOUT_EXPIRES, PDelay interval 129, wait_time 0, timeout 5000000
ERROR : GPTP [08:05:17:312] Link delay 0 beyond neighborPropDelayThresh; not AsCapable
Segmentation fault

It sent the pDelay_request in same timestamp value again and again. My device got the 1588 capable, but it seems like I need to modify the software to porting gptp program in my device. Do you know any information about this issue?

Jeffrey.

pinealservo, 2017-05-18:
That's very strange; I'm guessing that all your local Tx/Rx timestamps are being set to the same value, but the phy delay values are making them look different.  The timestamping is a function of the kernel and the drivers, so I would take a look at the kernel configuration.

stujeffrey016, 2017-05-18:
HI @pinealservo,
But if the gptp got the request and response's timestamp, why program calculate the link_delay as 0?
I'll check the kernel configuration, thanks for your helping.

pinealservo, 2017-05-18:
The link delay calculation isn't going to do anything helpful if you don't have good timestamps, so I would not worry about it. It works well when you have good timestamps and your phy delay values are set up properly.

stujeffrey016, 2017-05-21:
Hi @pinealservo ,
I tried to find out my network interface card's phy_delay, but there's no any information about it. Instead of asking the chip supplier, is there any possible ways to find out my phy_delay value? I don't want to modify the kernel source code, that will be too complex to do this job.

andrew-elder, 2017-05-22:
PHY delay compensation is entered in the gPTP .ini file. There is no need to touch the kernel source code. As @pinealservo states, you need to be sure you are getting good timestamps first.

pinealservo, 2017-05-22:
In other words, the phy delay values being off wouldn't cause you to get timestamps that aren't changing between packets. Unless they're really far off, they might not even keep it from working once you are getting real timestamps. So don't worry about phy delay values yet.

You shouldn't have to change any kernel source code at all. You may need to change the kernel configuration though if it's not currently building the PTP support modules. I'm afraid we can't offer a lot of real answers for you though, as none of us have a build or test environment like yours, so we can only guess at what might be wrong. I have never seen the problem you're having before, even when cross-compiling for an i.MX 6SoloX board.

stujeffrey016, 2017-05-22:
Hi @pinealservo & @andrew-elder ,
Thanks for your information. After doing some researching and testing, I think my device's timestamp has some setting problem. I used other program to print the timestamp value, and it caught the same timestamp value again and again, too.  In the booting process, the message show that `ptp clock support registered`, I think my kernel configuration has enable the ptp function, but I need to set something about it. I'll keep trying. Very appreciate your helping!

ahogen, 2017-05-22:
@stujeffrey016, you said (about [a month ago](https://github.com/AVnu/OpenAvnu/issues/579#issuecomment-297694441)) that you were connected to an AVB switch. What specific switch?

I've had (what I think is) a similar issue with timing info not getting synced in the AVB network because the switch wasn't doing what it was supposed to do. AVB endpoints could see each other, but the timing was not getting communicated properly. Switching to a [Cisco Catalyst 3650 2X10G](https://www.cdw.com/shop/products/Cisco-Catalyst-3650-24PDM-L-switch-24-ports-managed-desktop-rack-m/4153867.aspx#PO) we had in house solved everything, as it actually supports AVB. You probably already know this as you seem like you have a better understanding of AVB than I do, but I just wanted to toss in my 2 cents nonetheless. Cheers!

stujeffrey016, 2017-05-23:
Hi @ahogen ,
Congratulations! I pretty sure that my switch supports AVB. I used to connect to PC endpoint and it works fine, so I think is the endpoint timestamp issue in this case.

stujeffrey016, 2017-06-18:
Hi @pinealservo , 
I already port the Openavb into my i.mx6Q now, but there is a little problem in the building process. The rawsocket_rx.c has include the libigb.a file which I didn't compile. I won't use the avtp_pipeline program so I think this issue won't be a problem for me, just mention to you guys.
[image]

Another issue is the AS part; I already open the timestamp in my platform, and there's no error message show up but I still got some strange phenomenon. In PC platform the AS program seems in a stable system. But in the i.MX6Q The Grand Master clock keep changing. Do you have any information about this issue? 

[image]
[image]

Thanks!
Jeffrey.

pinealservo, 2017-06-18:
The build works fine if you use a toolchain configuration that isn't configured for an i210, like the one that I mentioned earlier in this issue here: https://github.com/AVnu/OpenAvnu/issues/579#issuecomment-294929400

I am building and running with the "generic" toolchain configuration on both an i.MX 6SoloX and i.MX 7 board, as well as on Intel PCs with the upstream kernel driver for the i210 that doesn't support the igb avb library.

Is there any reason you're not going to use the avtp_pipeline code?

stujeffrey016, 2017-06-18:
Hi @pinealservo ,
I already done the CMakeList.txt part and use the command you have mentioned to compile the program on my I.MX6Q (not compile in PC so it won't need to setup the cross_compile environment, right?), but still got the error.
The reason I am not using avtp_pipeline is that I try to combine the AS part and mrpd to my program, which is a 1722 Listener using gstreamer and VPU on my platform. But if the avtp_pipeline compile successfully, I'll try to use it (add the VPU feature in avtp_pipeline gstreamer) and turn to focus on the NIC driver part.

Jeffrey.

stujeffrey016, 2017-06-18:
Hi @pinealservo ,
I download the newest zip file and re-compile again, it success! Sorry for bother you.
Now I can focus on the driver issue.

Thanks!
Jeffrey.

stujeffrey016, 2017-06-21:
Hi everyone,
I already put the openavb on i.mx6q platform. Thanks for your helping.

Regards,
Jeffrey.

closed 2017-06-21 by stujeffrey016