Avnu/OpenAvnu #580: MRPD: Double Free Corruption in msrp_dumptable

chris-kuhr, 2017-04-13:
Hi,

I wrote my own AVB software with 8 talkers and 8 listeners running in parallel.

However, mrpd crashes with a double free corruption in glibc, when I receive the following MSRP packets:

1. Packet:
=================================================================================
No.     Time           Source                Destination           Protocol Length Info
   5599 225.919075     ExtremeN_9c:52:0a     LLDP_Multicast        MRP-MSRP 260    Multiple Stream Reservation Protocol

Frame 5599: 260 bytes on wire (2080 bits), 260 bytes captured (2080 bits)
    Encapsulation type: Ethernet (1)
    Arrival Time: Apr 13, 2017 08:58:42.122858000 CEST
Ethernet II, Src: ExtremeN_9c:52:0a (00:04:96:9c:52:0a), Dst: LLDP_Multicast (01:80:c2:00:00:0e)
Multiple Stream Reservation Protocol
    Protocol Version: 0
    Message: Domain (4)
        Attribute Type: Domain (4)
        Attribute Length: 4
        Attribute List Length: 9
        Attribute List
            Vector Attribute
                Vector Header: 0x2002, Leave All Event: Leave All
                    001. .... .... .... = Leave All Event: Leave All (1)
                    ...0 0000 0000 0010 = Number of Values: 2
                First Value
                    SR Class ID: SR Class B (5)
                    SR Class Priority: 2
                    SR Class VID: 2
                Attribute Event: JoinMt (3)
                Attribute Event: JoinMt (3)
            End Mark: 0x0000
    Message: Talker Advertise (1)
        Attribute Type: Talker Advertise (1)
        Attribute Length: 25
        Attribute List Length: 142
        Attribute List
            Vector Attribute
                Vector Header: 0x2001, Leave All Event: Leave All
                    001. .... .... .... = Leave All Event: Leave All (1)
                    ...0 0000 0000 0001 = Number of Values: 1
                First Value
                    Stream ID: 0xc351000100000301
                    Stream DA: Ieee1722_03:c3:51 (91:e0:f0:03:c3:51)
                    VLAN ID: 0x0002
                    TSpec Max Frame Size: 56
                    TSpec Max Frame Interval: 1
                    Priority and Rank: 0x60, Priority: Traffic Class A, Rank: Emergency, Reserved: Reserved-0
                        011. .... = Priority: Traffic Class A (3)
                        ...0 .... = Rank: Emergency (0)
                        .... 0000 = Reserved: Reserved-0 (0)
                    Accumulated Latency: 141949
                Attribute Event: JoinMt (3)
            Vector Attribute
                Vector Header: 0x0001, Leave All Event: Null
                    000. .... .... .... = Leave All Event: Null (0)
                    ...0 0000 0000 0001 = Number of Values: 1
                First Value
                    Stream ID: 0xc352000200000301
                    Stream DA: Ieee1722_03:c3:52 (91:e0:f0:03:c3:52)
                    VLAN ID: 0x0002
                    TSpec Max Frame Size: 56
                    TSpec Max Frame Interval: 1
                    Priority and Rank: 0x60, Priority: Traffic Class A, Rank: Emergency, Reserved: Reserved-0
                        011. .... = Priority: Traffic Class A (3)
                        ...0 .... = Rank: Emergency (0)
                        .... 0000 = Reserved: Reserved-0 (0)
                    Accumulated Latency: 141949
                Attribute Event: JoinMt (3)
[... 257 lines cut ...]
                    SR Class ID: SR Class A (6)
                    SR Class Priority: 3
                    SR Class VID: 2
                Attribute Event: JoinIn (1)
            End Mark: 0x0000
    End Mark: 0x0000

I don't have the backtrace right now, but will post it later, the backtrace however, points to the function msrp_dumptable in msrp.c to the following code segment (l.3315). To be precise, ``free(msgbuf);`` crashes.

    free_msgbuf:
            if (regsrc)
                 free(regsrc);
            if (variant)
                 free(variant);
            if (stage)
                 free(stage);
            free(msgbuf);
            return 0;

Is this related to some other MRP bugs?
Has someone an idea, of what goes wrong?
Or should I dig myself?

BR,
Chris

andrew-elder, 2017-04-13:
Are you sending a ? query type command to mrpd?

andrew-elder, 2017-04-13:
There is this https://github.com/AVnu/OpenAvnu/issues/119

chris-kuhr, 2017-04-13:
I used the mrp clients from simple talker and listener, mostly unchanged, well refactored but processing should be the same. So I cannot tell you what I am doing exactly, yet.

pinealservo, 2017-04-14:
Unfortunately, heap corruption bugs can lead to crashes far from where the corruption occurs. I haven't run across any bugs like that, but I also haven't used the mrp daemon outside of some very simple scenarios yet.

You might try building with -fsanitize=address; this should instrument the binary in such a way that it immediately crashes with diagnostics when it performs an illegal memory operation. I haven't tried it before on mrpd, but it's helped me track down bugs in other programs before.

andrew-elder, 2017-04-14:
I've used mrpd outside of simple scenarios, and I haven't seen any issues that would point to major bugs. I don't use the query interface at all though.

chris-kuhr, 2018-10-02:
The bug was a heap corruption.
closing...

closed 2018-10-02 by chris-kuhr