Avnu/OpenAvnu #581: General question: What AVDECC Command do I use to dynamically create destroy Talker and Listener?

chris-kuhr, 2017-04-19:
Hi,

for my AVB Proxy Server, I need to communicate the ( public internet IP:Port ) tuple for a listener and talker instance to be initialized. Prior to ACMP Connect_Tx_CMD and AECP AEM START_STREAMING.

One way would be a memory object, but is there no other way? I found nothing else so far.

BR, 
Ck

jdkoftinoff, 2017-04-19:
Hi Chris - Can you describe in more detail what you are wanting to accomplish? Like who is going to consume the IP:port and what they would do with it?

However - at this point I would use a custom control type.  use CONTROL_VENDOR, and use your own OUI-24 or OUI-36 as a prefix for the vendor_eui64: see 7.3.5.2.9.   Then you can publish the eui64 and the internal blob data format so others can parse it. 
You can see an example of how I used a CONTROL_VENDOR for logging and debug console: https://github.com/jdkoftinoff/jdksavdecc-c/blob/master/include/jdksavdecc_jdks.h - and if you need a placeholder OUI-36 for your experiments you can use   70:b3:d5:ed:cf:00:00:00 which is in my personal range.

jdkoftinoff, 2017-04-19:
Oh, I completely forgot I already did an IPv4 parameters vendor specific control which may or may not be relevant for you, it is defined in the above linked header file:

JDKSAVDECC_JDKS_AEM_CONTROL_IPV4_PARAMETERS ( 70:b3:d5:ed:c0:00:00:01 )

chris-kuhr, 2017-04-19:
Hi Jeff,
thanks for that!

Well I just need some standard compliant way to tell the proxy, which ports to listen to/open in the firewall and which IPv6 address each AVB Listener should reply to.

I need IPv6, but I will have a look at your code, perhaps I can add an IPv6 version.

BR,
Ck

closed 2019-09-19 by chris-kuhr