Avnu/OpenAvnu #582: [AVB] Network Latency for 1722

MartinSeungjinHyun, 2017-04-20:
Hi, I'm about to measure network latency of igb.
I'm using I210 network interface.

I can put some log on igb_fetch_rx_buffer function on file "igb_main.c".
And I could get some logs from it.
However I couldn't get any log when I get 1722 packet from talker.
Could you tell me why I can't track 1722 data from IGB driver ?
Or tell me something wrong in a way of by me?

jdkoftinoff, 2017-04-20:
Can you explain what exact latency are you trying to measure? Between which two points? And why?

MartinSeungjinHyun, 2017-04-21:
Hi, Jeff,
I want to know latency two points between Talker's Driver's IEEE 1722 packet sending function and Listener's IGB driver. I heard that network latency between two physical network should be under 2ms.
I want to know I210 driver is able to meet. I have oscilloscope. but I couldn't find any physical point on I210 network card (Listener side). So I just do it in a way of examining driver code by changing GPIO on my board.

jdkoftinoff, 2017-04-21:
What size is your network? is it gigabit? How is the packet being sent? Is there any packets in the network driver's output queue?

A single packet being sent through an uncongested network is fast.  Quite often the biggest problem is your code getting scheduled to execute at the correct time.

Here are some rough points. See my AVB Bandwidth calculator at https://abc.statusbar.com 

Imagine an ethernet packet containing a 1722 AVTDU with 8 channels of 48 kHz audio. 
This packet is 256 bytes. Including all overhead this is 299 octet-times. At gigabit this is 2.392 microseconds.  This is how long the message takes to be transported over the wire to the switch.
The switch typically takes around 300 nanoseconds to forward it to the destination ports.  So this is 2.692 microseconds for one hop and switch backplane.  If you put this through 7 switches, this gives you 18.844 microseconds from original transmitter to final switch port.

Other people on this list can assist you to see if it is possible to use the igb time stamping features with non-802.1AS ethernet frames.

Now, once you enable class traffic shaping (FQTSS) in order to make sure that your flow won't interfere with other flows, then there can be additional latency added.  This also allows for maximum interference packets. This is where each hop can take up to around 138 microseconds - but only if there is enough interference packets actually present.  The point of AVB with SRP is that SRP will guarantee that given these traffic shapers that your media packet will have a known worst case latency through the entire network.  If you use IEEE 1722.1 for your listener, you can ask your end station for the MSRP Accumulated Latency value that the switch reports for a specific stream, and this will tell you the worst case latency possible on that network for that stream.  And even then, the actual typical latency will be much less depending on traffic, and this is why your listener needs a time compensation buffer to store these samples for later in case some were delayed by interference traffic.

Does this help?

MartinSeungjinHyun, 2017-04-25:
Small, Very local area network. Yes it is gigabit. The network driver just used on listener.. so no.
Your comment is very helpful.

I finally did it on driver function "igb_fetch_rx_buffer". measured just checking packet length (skb->len)
like this.
       /* IEEE1722 packet */
       if((skb->len == 94))
       {
               GPIO_SET(1);
       }

        /* clear contents of rx_buffer */
        rx_buffer->page = NULL;
       GPIO_SET(0);
And run pcap_loop on the user space.
I got what I want.
It is measured under 2ms.

Best Reagrds
Martin

closed 2017-04-25 by MartinSeungjinHyun