Avnu/OpenAvnu #592: MRP: handling of rMT! events

andrew-elder, 2017-05-09:
When processing an MRP PDU is it acceptable to throw away rMT! events early in the processing if the attribute does not exist locally? See below

[image]

Currently what happens is that the attribute is created then applicant and registrar statement machines are both run and the client is notified of a LV event. The attribute is then deleted. If there are many attributes, running the state machines and notifying the client can result in significant processing and memory usage. Furthermore, no useful information is conveyed to the client. I'd like to propose that the event processing at this line here
https://github.com/AVnu/OpenAvnu/blob/open-avb-next/daemons/mrpd/mvrp.c#L286
be modified so that processing for the attribute terminates in the case of MRP_EVENT_RMT when the attribute database does not contain an attribute that matches the one being processed.

Comments?

pinealservo, 2017-05-09:
I think it's fine to ignore rMT! when there is currently no applicant state for that attribute, and it's certainly fine for the registrar to ignore rMT! since it's not listed as an event in the Registrar state table. Sending a LV event to clients for each rMT! event definitely seems like the wrong thing to do, and that seems to be the only visible effect that the rMT! event currently has.

That MVRP message looks crazy. I've never seen anything send the full range of attribute values before, although I guess I can see how you could interpret it as the right thing to do in some situations. Does it always include the whole range, or only in the states with 's' actions rather than '[s]' actions?

andrew-elder, 2017-05-12:
See
https://github.com/AVnu/OpenAvnu/pull/596

pinealservo, 2017-05-15:
Closing since #596 is now merged.

closed 2017-05-15 by pinealservo