Avnu/OpenAvnu #601: igb_avb driver compatibility

zeefpt, 2017-05-19:
Hi again Folks,

I have all the stuff compiled, ready for testing. :) But now i have some problems.

I have 2 cards in my embebed system:

I211 : use igb driver
I219: use e1000e driver

I remove the igb original driver and put the igb_avb. He is inserted but after that nothing happens. Interface is not up.

#rmmod igb
igb 0000:02:00.0: removed PHC on eth1_i211
#modprobe i2c_algo_bit
modprobe: module i2c_algo_bit not found in modules.dep

#modprobe dca
modprobe: module dca not found in modules.dep
#modprobe ptp
modprobe: module ptp not found in modules.dep
#ls
igb_avb.ko
#insmod igb_avb.ko
igb_avb: loading out-of-tree module taints kernel.
Intel(R) Gigabit Ethernet Network Driver - version 5.3.2_AVB
Copyright (c) 2007-2015 Intel Corporation.
#ethtool -i eth1_i211
Cannot get driver information: No such device
#ethtool -i eth1_i219
Cannot get driver information: No such device
#ethtool -i eth2_i219
driver: e1000e
version: 3.2.6-k
firmware-version: 0.8-4
expansion-rom-version:
bus-info: 0000:00:1f.6
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: no

igb_avb driver is compatible with I211 card?

Could i mount a talker and listener without the module or it mandatory that have to insert that?

Best regards,

pinealservo, 2017-05-20:
The i211 doesn't have the FQTSS or launch-time scheduling features that the i210 has, which are the whole reason the i210 "igb" driver exists.  The i211 does have 1588/PTP support, but that ought to be usable from the mainline kernel driver.

You can't use simple_talker or simple_listener without an actual i210; I don't know of any other Intel NIC that has the required features.

The avtp_pipeline talker and listener code, if you build them with the "generic" toolchain library instead of the x86 igb stuff, can be used with non-i210 NICs with 1588 support.

zeefpt, 2017-05-22:
Hi pinealservo,
Thnks for anwser,

I have the i219 ethernet card too :) , Where you see in specs that doesn have the FQTSS features? :)

Relatively to avtp_pipeline where i find the generic toolchain that you talk, instead of my x86 cross-compile? Just compile with generic settings? :)

Regards,

pinealservo, 2017-05-22:
At the top-level of the OpenAvnu checkout, run this make command:

    $ PLATFORM_TOOLCHAIN=generic make avtp_pipeline

That will prevent it from linking to the i210 library and instead use the standard Linux network stack. The packet launch times and presentation timestamps will be far less even and I'm working on ways to lessen those problems, but you should be able to get a talker and listener built from the avtp_pipeline code to work together with some tweaking of the avtp_pipeline INI files. Be sure to read through the documentation that comes with the avtp_pipeline code.

You might also consider buying an i210 NIC if you've got a free PCI-E slot. They're not very expensive.

zeefpt, 2017-05-23:
Hi pinealservo.
Thks for anwser.
Just bought an i210 Nic.
First i have to enable MSI-X in kernel, beacause gives me one error for that. Solved.

Now when i insert the module gives me that error:

igb_avb: probe of 0000:01:00.0 failed with error -5

When i poweron the board both the i211 and i210 starts powered on and ok with generic igb driver.
When i removed de module ther error occurs.

Anyidea where shoul i look?

Regards,

pinealservo, 2017-05-23:
Those messages don't necessarily indicate that it won't work, but I guess it doesn't guarantee it is working either. It doesn't seem to be using MSI-X, but maybe that's not supported on your board for some reason? There's not enough information to say much more.

zeefpt, 2017-05-25:
Hi folks, thnks for help.

Its now solved. It was my kernel that have some functions of msi that have changed name , so correct that resolved the problem. :)

Searching the examples i dont find a readmr for how to use jackd listener and talker, can you give me some examples?

Doesn understand to, how to use the simple_rx is with simple talker?

Regards

pinealservo, 2017-05-25:
I have never used the jackd examples, but assuming you have the gPTP and MRP daemons working on the talker and listener machines, you just run the example programs. They are examples, so the source code is meant to be read in order to understand what they do.

This is not an ideal level of documentation, but it's what we've got right now. Contributions of README files,wiki pages, or other forms of documentation are welcome if you'd like to capture and share your learning experiences.

pinealservo, 2017-05-25:
I'm closing this issue, as the original topic of driver compatibility appears to have been resolved. If you have further issues, please open topic-specific ones so that we can keep them reasonably organized.

closed 2017-05-25 by pinealservo

JXter, 2018-03-07:
@zeefpt I am also facing the same problem.

igb_avb adapter->tx_ring_size 256
igb_set_interrupt_capability: Multi Q activated
TXQ: 4, RXQ: 4, NumVecs: 4
igb_set_interrupt_capability: looping through item 0
igb_set_interrupt_capability: looping through item 1
igb_set_interrupt_capability: looping through item 2
igb_set_interrupt_capability: looping through item 3
igb_set_interrupt_capability: looping through item 4
igb_avb 0000:01:00.0: Failed to initialize MSI-X interrupts. Falling back to MSI interrupts.
pps pps0: new PPS source ptp1
igb_avb 0000:01:00.0: added PHC on eth0
igb_avb 0000:01:00.0: Intel(R) Gigabit Ethernet Network Connection
igb_avb 0000:01:00.0: eth0: (PCIe:2.5GT/s:Width x1)
igb_avb 0000:01:00.0 eth0: MAC: 68:05:ca:51:32:cd
igb_avb 0000:01:00.0: eth0: PBA No: G65055-011
igb_avb 0000:01:00.0: LRO is disabled
igb_avb 0000:01:00.0: Using MSI interrupts. 1 rx queue(s), 1 tx queue(s)

Can you please share some more information about what changes you did in kernel to fix this.