Avnu/OpenAvnu #606: Question on AVTP timestamp

Shrikant-M, 2017-05-24:
Hi,
I am referring to IEEE-1722 2016 document. As per the document, avtp_timestamp is calculated as below
`avtp_timestamp = (AS_sec × 10^9 + AS_ns)mod2^32`

Since this value cannot be directly compared with gPTP time as it rolls over every 4 seconds,
I would like to know how does listener uses this value so that it can playback/present samples?

Thanks in advance

pinealservo, 2017-05-25:
If you apply the same transformation to the current gPTP timestamp, then you *can* directly compare them as integers modulo 2^32. Since a valid presentation time value represents a time that should be well under 1 second in the future vs. current gPTP wall clock time when you interpret it, that should help you deal correctly with any ambiguity between congruent timestamp values in your integers modulo 2^23.

Alternatively, you could use that same knowledge to correctly reverse the transformation from avtp_timestamp to a gPTP time, and do a comparison against gPTP times. Whatever makes the most sense for your use case.

Shrikant-M, 2017-05-26:
Thank you pinealservo.

First approach is clear.

Regarding second approach, at listener side, I assume this is what you meant
`presentation time = (((AS_sec × 10^9 + AS_ns) / 2^32) * 2^32) + avtp_timestamp `

andrew-elder, 2017-05-26:
I don't think that is correct at all for the listener side. The avtp_timestamp *is* a presentation time - it is just that the gPTP time is represented as the lower 32 bits of the gPTP time in nanoseconds. So it is incorrect to add avtp_timestamp to any other gPTP timestamp.

The approach I use is to keep a copy of the gPTP time represented as a 32-bit nanosecond number in the listener. To do that, one can use the math you have above. Something like

gPTP32 = (uint32_t)((uint64_t)AS_sec * 1000000000 + AS_ns) & 0xffffffff);

pinealservo, 2017-05-26:
If you keep track of 64 bits of nanoseconds since the gPTP epoch, the avtp timestamp will correspond to the lower 32 of those bits. Keep the upper 32 in a separate word and increment when avtp timestamp wraps. Now, shift-left the upper word by 32, add avtp_timestamp then div/mod by 10^9, and you've got seconds and nanoseconds since the last second, which you can easily convert to however you're storing gPTP timestamps.

It's unfortunate that gPTP doesn't store timestamps in a powers-of-2 format in the first place; it's far more convenient and efficient to work with them that way rather than having to deal with base-10 conversions. I wouldn't recommend implementing your time-related algorithms in terms of gPTP timestamps, but it's important to be able to translate between timestamp representations.

Shrikant-M, 2017-05-29:
Thank you andrew-elder & pinealservo for detailed explanation

closed 2017-05-29 by Shrikant-M

waltergallegog, 2020-12-30:
Because this thread comes up when googling about the avtp timestamps, I leave some details for future people looking for information, that helped me understand how the avtp timestamp relates to the gptp time, and how it is reconstructed in the listener.

The avtp timestamp can only represent 4.29 seconds (2^32 in ns).
These 4.29 seconds (relative to the current gptp time) are actually divided in 2.1 seconds in the future, 2.1 seconds in the past.

When the listener receives each avtp sample, it has to decide if the sample is in the past, or in the future w.r.t the ptp clock.
Imagine a sample that is 4 seconds in the futere w.r.t the ptp clock. It is more probable that the sample is actually .29 seconds in the past, and got delayed in the transmission.
To understand how this is done in the listener, it is easier to just take a look at the simple function **openavbAvtpTimeSetToTimestamp** in the OpenAvnu repo, and try with some values by hand:
https://github.com/Avnu/OpenAvnu/blob/master/lib/avtp_pipeline/avtp/openavb_avtp_time.c#L167

Then, if the sample is in the past, it may get purged and not be played at all.
Samples get purged if they are older than the max_stale parameter. (from the .ini files)
For this see:
https://github.com/Avnu/OpenAvnu/blob/master/lib/avtp_pipeline/mediaq/openavb_mediaq.c#L183