Avnu/OpenAvnu #607: Is the Openavnu must use I210?

liyisecond339, 2017-05-25:
Hello all:
Is the OpenAvnu must the device I210 and special switch? I did not have this device and special switch, may I run this program?
Thanks a lot!
Eric Li

andrew-elder, 2017-05-25:
OpenAvnu is a suite of programs. You will run in to many issues if you don't have and i210 and an AVB/TSN capable switch. If you are an expert developer and wish to experiment, you could try a different test configuration. You will have to figure many things out for yourself due to your non-standard setup.

ahogen, 2017-05-25:
How difficult would it be to use another NIC, provided it supports 802.1AS and 802.1Qav? How much of the i210 driver is hardware specific vs generic to those AVB/TSN standards? In other words, how difficult would it be to port the i210 driver to another compatible NIC?

pinealservo, 2017-05-26:
The problem is that there are currently no standard Linux APIs for managing FQTSS traffic shaper parameters, either at the interface between userspace and the kernel or between the kernel networking code and the individual drivers.

The custom i210 driver (along with its associated userspace library) does a few not-necessarily-related things:

1. It provides a way to set the FQTSS slope parameters of Qav from userspace.
2. It provides a way to program the special descriptors for the NIC that perform launch-time shaping
3. It gives full control of some kernel-allocated buffers and the NIC registers to a userspace task, which allows you to manage the SR-class queues (i.e. not the best-effort one, which is still managed in the kernel) entirely from userspace.

#3 is not strictly required for doing AVB, but it can make it easier to meet latency requirements in some situations as it eliminates kernel context switches and associated re-scheduling opportunities. It also prevents this driver from ever being considered for inclusion in the upstream kernel, and it accounts for most of the complexity in the igb library.

The code to do #1 is pretty trivial, at least if you're familiar with network device driver code. #2 is slightly less so, but not bad. You can get good-enough performance without #3 for many use cases, so I'd avoid doing that unless absolutely necessary. The mmapped raw-socket interface of AF_PACKET, used by libpcap, is pretty low-overhead.

The problem is that you'll have to do custom work for every NIC, and unless the work is coordinated, the way you manage the driver code from userspace will vary from implementation to implementation. If you've got the interest, inclination, and hardware to do this for something besides the i210, please let us know!

The interfaces required for 802.1AS operation are already integrated in the kernel, and there's no requirement for the i210 driver to use the gPTP daemon. Many other NICs have the requisite hardware and driver code, and they'll work just fine with the OpenAvnu gPTP daemon.

AVB is specified to require at least a 100Mb link; I think current work in TSN is relaxing that to some new 10Mb physical layers for automotive and industrial control purposes, but some current AVB solutions use 100Mb.

ahogen, 2017-05-26:
Sorry if I'm a little ignorant of things going on here. I'm basically a new college grad and haven't written much for Linux specifically. Experience has been with lower level CPU architectures and bare-metal stuff. I'm asking because I am looking into a more embedded, cheap, talker-only kind of application and I don't really have any use for Linux. I'm looking at MACs (not PCIe NICs) like [WGI21x](http://www.mouser.com/new/Intel/intel-wgi21x-controller/) or possibly even a [fido 5000](http://www.innovasic.com/products/fido5000-rem-switch). Both of which support the necessary timing protocols in hardware and leave the IEEE 1722 business up to the user code (from what I understand).

My understand is that Open Avnu probably has the most complete code base for AVB/TSN. XMOS has theirs, which works fine and great, but it isn't updated/developed nearly as frequently, and I've tended to not like their code/documentation.  Because the Avnu code targets Linux platforms, is it worth trying to dig in to and attempt to port it to a bare-metal application or would you think it'd be simpler to approach it from a different perspective and maybe just use the libraries here as a reference? 

I did skim over issues #353, #514, and #579, but those questions seemed to still have a Linux kernel involved on the target platform.

Thanks for all that info above!

pinealservo, 2017-05-26:
The avtp_pipeline code (found in the lib/avtp_pipeline directory) is a mostly-complete framework that several commercial implementations were based on; some of them were on Linux and others on other small RTOS environments. I'm actively working on getting it working better in Linux and getting the missing pieces (mostly time-related parts, like presentation time stamps and clock recovery) implemented in a way that won't be super board-specific. You'd need to write a "platform" layer for your board, and you'd need to hook up gPTP and SRP implementations there as well as provide some application-level glue code, but it's designed to keep that separate from the core AVTP implementation code.

I think you've been a bit unfair in your judgement of the XMOS solution. There's definitely a learning curve involved, but they also provide a great deal of learning material and excellent free tools. It's by far the easiest architecture to use for precisely timed interactions. Their AVB code hasn't seen a huge amount of development lately, but that's largely because the protocol implementations have been working correctly for years now. Plumbing code really shouldn't have a lot of churn once it works. They also make terrific debugging tools, since you can extend and instrument just about any part of the system. If you have access to one of the XMOS AVB dev boards, time spent getting familiar with how it works would be well-spent even if you don't plan on using it directly in a product/project.

If you want a non-XMOS platform that's got everything you need for AVB/TSN development without Linux, check out the Atmel SAMV71 Xplained board: http://www.atmel.com/tools/atsamv71-xult.aspx -- it has absolutely everything you need, including a CS2100 PLL-based clock generator for media clock generation/recovery and flexible timer/counter peripherals for controlling and measuring its output.

Everyone gets caught up looking at MAC/PHY stuff, but you've got to be able to get precise media clock edge timestamps for both talkers and listeners that deal in live audio! Media clock recovery in AVB is one of the trickiest aspects for both talkers and listeners to get right, and having the right hardware in place to support it is often overlooked. If you think you're in good shape just because you have a 1588-capable MAC, you may find yourself in a world of hurt later if you can't precisely measure your media clock or keep your transmit rate in sync with the AVTP timestamps you're sending and the rate at which you're generating media samples.

pinealservo, 2017-05-30:
The original opener hasn't responded and I think the original question has been pretty thoroughly answered at this point.  I'm going to close the issue; please open a new issue if you have any other questions or problems.

closed 2017-05-30 by pinealservo