Avnu/OpenAvnu #609: what are the scenario in which CRF streams are useful

dnpawate, 2017-05-26:
As per my understanding the AVTP_timestamp is used to present the audio on listener.

1. when will the CRF stream used
2. when are the data_field of CRF stream is used
3. comparison & role of ptp vs CRF timestamp
4. if ptp is availble to sync the devices to run on common clock, why do we need CRF time stamp.

Could you please acknowledge for above queries.

Regards,
Deepak

pinealservo, 2017-05-26:
The purpose of CRF streams is to provide a low-overhead method of distributing a "house clock" among a set of AVB participants. This can allow them to pre-synchronize their media clocks so that they are already locked before media streams are established.

Each timestamp, whether it be in a media packet's AVTP_timestamp field or an entry in a CRF data_field, represents the precise timestamp at which a media clock edge transition occurred. Knowing how many clock edges apart the timestamps are taken, you can calculate from the difference between the timestamps in a series the actual rate (and thus the frequency) of the clock being timestamped. By comparing transmitted timestamps vs locally-captured media clock timestamps of the corresponding local media clock edges, the phase offset between remote and local media clocks can be determined. These provide input to the local clock recovery mechanism.

CRF frames don't have any media samples; instead they contain in their packet data all the clock edge timestamps that occurred since the last CRF frame was transmitted. This gives them a lower transmit rate than standard media transport streams, while still conveying the same amount of clock information.

The purpose of gPTP is to provide a very accurate "wall clock" reference with which devices across the network can coordinate both the timing of events and the measurement of the frequency and phase offset of *other* unrelated clocks. This measurement forms the feedback mechanism for clock recovery, which can be done for media clocks embedded within normal streams via AVTP_timestamp fields or CRF streams via their data_field timestamps.

AES67 takes a different approach, in which all media clocks (for both talkers and listeners) must be adjusted to run at an integral ratio of the PTP wall clock rate; i.e. the clock must run exactly at its nominal rate as measured against the PTP wall clock. This means that only equipment capable of slaving its clock to a PTP grandmaster clock can serve as a "house clock" source to non-networked devices, and all network talkers must be able to do so as well.

By contrast, AVB can allow any number of arbitrary, unreleated and unsynchronized media clocks to be used on a network. An existing, non-networked house clock source can be used as a talker media clock, which makes integrating AVB to an existing Pro A/V environment with a distributed house clock easier--the AVB devices synchronize to the existing clock without requiring sample rate conversion.

dnpawate, 2017-05-30:
Thank you very much for your explanation.

Can we find any reference for CRF implementation. ?
Can CRF help in video presentation sync ?

pinealservo, 2017-05-30:
We don't currently have any implementation of CRF streams in our repository.

CRF can be used to synchronize any media clocks, so yes, they could be used to sync the video and audio clocks related to video streams. That would only be part of the story, however, and clock sync is currently a very solution-specific area so we don't have a lot of code for anything beyond 1588/gPTP clock sync.

closed 2017-06-01 by dnpawate