Avnu/OpenAvnu #612: RESTConf for TSN

syedsk, 2017-05-29:
Firstly excuse me if this is not a proper forum for this question. 
Is there a plan under OpenAvnu to develop RESTConf server for configuring TSN devices?
For 802.1Qbv and 802.1Qcc, its very important(in my view) to have some reference implementation of RESTConf and associated REST APIs to use IEEE 802.1x yang model. 
   I have forked a python based RESTConf project(jetconf) https://github.com/syedsk/jetconf/ and added 802.1Qbv yang model support(although with some minor changes to fix some issues). You can add/delete 802.1Qbv database through REST, then there is a callback/plugin(tsn_connect) that gets invoked when such a configuration change happens. 
   Any user wishing to use this on their HW(say TSN switch) can extend tsn_connect and configure their HW using ioctl calls etc.
   I wish to invest more time on this server fork and appreciate any feedback. 

Regards
-syed

andrew-elder, 2017-05-30:
@syedsk - we have talked about OpenAvnu being a place for folks to collaborate on extensions to 802.1Qxx. Up 'til now, we didn't have anyone interested in contributing code and/or getting some momentum. Would you like to host your efforts on OpenAvnu so that you get more exposure? What can we do to help?

syedsk, 2017-05-31:
I have raised a request for approval from my employer for a go ahead to officially host our TSN efforts(Qcc etc) on openAvnu.

andrew-elder, 2017-05-31:
Great. And of course the great thing about git is that you would maintain your own complete fork of the repository that you would work off.

syedsk, 2017-07-03:
@andrew-elder - I have got this response from my employer: If the TSN components we are developing are generic(like RESTconf, tsn_talker etc) and are not directly associated with our HW, it would not be hosted on our company's github. 
  So I can work with AvNU_SW group to contribute on an individual capacity for these generic TSN sw components.
  
regards
-syed