Avnu/OpenAvnu #613: gptp AsCapable doubt

zeefpt, 2017-05-30:
Hi folks,

Im starting testing the daemons and have several doubts.

1- When i start gptp with a switch netgear GS308 i have this output:

PC1:

#./daemon_cl eth0
INFO     : GPTP [15:17:54:739] gPTP starting
ERROR    : GPTP [15:17:54:739] Group ptp not found, will try root (0) instead
ERROR    : GPTP [15:17:54:739] Using clock device: /dev/ptp1
STATUS   : GPTP [15:17:54:739] Starting PDelay
EXCEPTION: GPTP [15:17:57:797] PDelay Response Receipt Timeout
EXCEPTION: GPTP [15:17:58:817] PDelay Response Receipt Timeout
EXCEPTION: GPTP [15:17:59:833] PDelay Response Receipt Timeout
ERROR    : GPTP [15:17:59:869] Link delay 19441 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [15:18:00:877] Link delay 19393 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [15:18:01:889] Link delay 19361 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [15:18:02:897] Link delay 19933 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [15:18:03:905] Link delay 19861 beyond neighborPropDelayThresh; not AsCapable
^CERROR    : GPTP [15:18:05:570] Exiting on 2

PC2:

local@pt02ipk1:~/jpereira/OpenAvnu/daemons/gptp/linux/build/obj$sudo ./daemon_cl eth3
INFO     : GPTP [15:18:02:028] gPTP starting
ERROR    : GPTP [15:18:02:029] Using clock device: /dev/ptp0
STATUS   : GPTP [15:18:02:029] Starting PDelay
STATUS   : GPTP [15:18:02:029] Link Speed: 1000000 kb/sec
ERROR    : GPTP [15:18:02:067] Link delay 19693 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [15:18:03:074] Link delay 19329 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [15:18:04:083] Link delay 19546 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [15:18:05:094] Link delay 19902 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [15:18:06:103] Link delay 19534 beyond neighborPropDelayThresh; not AsCapable
ERROR    : GPTP [15:18:07:111] Link delay 19122 beyond neighborPropDelayThresh; not AsCapable
^CERROR    : GPTP [15:18:07:296] Exiting on 2

2 -When i start the gptp directly between boards without switch i have this:

PC1:

#./daemon_cl eth0
INFO     : GPTP [15:23:39:140] gPTP starting
ERROR    : GPTP [15:23:39:140] Group ptp not found, will try root (0) instead
ERROR    : GPTP [15:23:39:140] Using clock device: /dev/ptp1
STATUS   : GPTP [15:23:39:140] Starting PDelay
STATUS   : GPTP [15:23:39:181] AsCapable: Enabled

PC2:

local@pt02ipk1:~/jpereira/OpenAvnu/daemons/gptp/linux/build/obj$sudo ./daemon_cl eth3
INFO     : GPTP [15:23:39:681] gPTP starting
ERROR    : GPTP [15:23:39:682] Using clock device: /dev/ptp0
STATUS   : GPTP [15:23:39:682] Starting PDelay
STATUS   : GPTP [15:23:39:682] Link Speed: 1000000 kb/sec
EXCEPTION: GPTP [15:23:42:716] PDelay Response Receipt Timeout
STATUS   : GPTP [15:23:42:753] AsCapable: Enabled

In first example could be switch interference?

In second example they shouldn't negotiate who is master anda slave to sync the time?
Both become AsCapable and in wireshark i can see packets.. but nothing more happens..

Best Regards,

andrew-elder, 2017-05-30:
The netgear GS308 is not AVB capable, so everything is working exactly as expected. The whole point of the asCapable flag is to detect whether a port is connected to another time aware port (or not). Your case 1 above has worked perfectly.

zeefpt, 2017-05-30:
hi andrew ,
But they shouldn t negotiate who can be the best master/slave when i connect both without switch?
It imperative to use a swith for testing?

Regards,

pinealservo, 2017-05-30:
1 - Back-to-back should work

2 - They should negotiate via BMCA which is the clock master. I don't have enough information to guess why they aren't, though. Is either of them sending Announce messages as far as you can see in Wireshark? Do you have a gPTP config file, and have you listed all the command line arguments you're using?

3 - If you mean the echo interface in the avtp_pipeline code as "echo talker/listener", then I would say there is no comparison to be made between that and the simple talker/listener code. Simple talker/listener exist to show how to talk to gPTP and MRP daemons and use the igb library to send/receive frames. I have no idea what testing scheme the echo interface was intended for, but it is probably not something you want to use. The closest to simple talker/listener would be the wav file interface with the uncmp_audio mapper.

4.2 - I'm not sure why it's printing output for every event, but it's normal for MRP to send 'JoinMt' periodically to keep a stream reservation attribute established.

4.3 - You put a dash in front of the device name. Write "-i eth3" instead of "-i -eth3"

The listener is not connected to the NIC for receiving 1722 frames due to 4.3, so it won't be able to extract audio from them.

zeefpt, 2017-05-30:
1- Now appears to work :) It was the "-i eth3". 

2- I didint configure a gPTP config file, just use ./daemon_cl eth3. Should i configure a ptp config file?

3- Yes the echo talker/listener scripts from top level folder from avtp_pipeline. Thnks for clarify.

4.2 -Ok i will investigate.

Now i will try to use the live_stream exmample. Expect questionss ahha

Many thnks

zeefpt, 2017-05-31:
Hi again folks,

When i try to revert to use the talker on PC 2 and listener in PC 1 gives that error:

local@pt02ipk1:~/jp/OpenAvnu/examples/simple_talker$ sudo ./simple_talker -i eth3 -t 2
attaching to 0000:01:00.0
attach failed! (Unknown error -6)
connect failed (No such file or directory) - are you running as root?
local@pt02ipk1:~/jpereira/OpenAvnu/examples/simple_talker$ 

What should i look, im running in root..
Regards,

pinealservo, 2017-06-08:
You shouldn't need a config file, but if you had one, it might have been set up wrong. If gPTP is still not working, you're going to have to dig into it yourself further in order for anyone else to be able to help.

You should probably not try to use the live_stream example, or anything involving gstreamer, unless you're prepared to figure out any gstreamer issues yourself.  Get an avtp pipeline tone generator stream working first, then work your way up by adding alsa source/sink modules, etc.

zeefpt, 2017-06-08:
Hi pinelservo,

Thanks for helping,
Is exactly what im doing, the tone generator are now working with talker/listener. Im now avancing to the videotestsrc of gstream :) When all work i will put all the work here :)
Thanks,

vitorroriz, 2017-06-23:
Hi @zeefpt I am connecting 2 devices directly, and both print the "asCapable" message but nothing seems to happen later, as you reported. In debug mode I could see they are exchanging some messages but apparently no Sync message and then the shared memo variable /shm/ptp is not being updated. Did you find out what was the problem?

Thanks!