Avnu/OpenAvnu #616: open-avb-next testing before merge to master

andrew-elder, 2017-06-01:
We need to have a few folks sign-off that the latest changes in the open-avb-next branch work ok. Particularly the gPTP changes. No problems are anticipated, but some feedback that all works as expected would be helpful.

andrew-elder, 2017-06-05:
Any comments? I assume @christopher-s-hall is fairly sure all is ok?

christopher-s-hall, 2017-06-05:
Everything works OK in my somewhat limited testing. The wireshark traces look good as well.

andrew-elder, 2017-06-09:
Anyone else have any comments on this subject?

bdthomsen, 2017-06-09:
I haven't seen any issues with the latest changes.

andrew-elder, 2017-06-13:
Ok, I'm going to create the pull request then.

closed 2017-06-13 by andrew-elder

CHM-me-ch, 2017-10-26:
which one is stable now, openAVB master or OpenAVB-next?

andrew-elder, 2017-10-26:
They should both be stable. I do see that we are overdue to do a merge for open-avb-next to master, so I have just created a new pull request.

CHM-me-ch, 2017-10-26:
thankyou for the reply. 
i am following the steps described in : https://github.com/AVnu/OpenAvnu/wiki/Getting-the-source to get the OpenAVB in my ubuntu 16.04

But when running the ./travis.sh command i am getting the error:
[error.txt](https://github.com/AVnu/OpenAvnu/files/1418364/error.txt)

The command "sudo insmod ./igb_avb.ko" is giving the error:
insmod: ERROR : could not load module ./igb_avb.ko: No such file or directory (I am running the command from the Open-AVB directory)
the igb_avb.ko file is not getting generated. How should i proceed?