Avnu/OpenAvnu #617: Get Tx Timestamp from gPTP

riteshjaiswal2016, 2017-06-01:
How can we print the timestamps which are transmitted in FollowUp messages?

andrew-elder, 2017-06-01:
You mean like this https://github.com/AVnu/OpenAvnu/blob/master/daemons/gptp/common/ptp_message.cpp#L1710
?

christopher-s-hall, 2017-06-01:
[look here](https://github.com/AVnu/OpenAvnu/blob/master/daemons/gptp/common/ptp_message.cpp#L906)

and change the output to VERBOSE

riteshjaiswal2016, 2017-06-02:
Thank you for guiding me. Now, I am able to get the timestamps values in FollowUp(Synch) and FollowUp(PDelay_Resp). 
One more thing, Can you please guide me on how can I print the time when a packet from simple_talker just get transmitted from the NIC? Also, gPTP time is the time which will get set on master and slaves after synchronization between them? Can you please clear the meanings of Device time, System time and gPTP time?

closed 2017-06-06 by riteshjaiswal2016