Avnu/OpenAvnu #618: I210 driver on ARM (lib/igb/libigb.a)

MartinSeungjinHyun, 2017-06-02:
Hi, I'm using I210 Ethernet card on my freescale sabre board.
I'm trying to build gptp with I210 feature. but I couldn't.
Because, file "Open-AVB/lib/igb/igb.c" include assembler code ( rdtscpll, __sync).

I couldn't use them so I just blocked it.
Is there any problem if I block these function for arm-build.
Could you tell me the way how to build these function in arm-build?

ahogen, 2017-06-02:
Refer to #353 and #514. In particular, maybe @dbol-elium has some insight.

MartinSeungjinHyun, 2017-06-05:
Ahogen,
Thanks for your comment.
I already use msi-x in kernel. so I don't think your recommendation is helpful.
(My kernel is okay and igb_avb modul is okay)
Actually I want to know exact arm assembler code for arm target.

So, please two function code. (igb.c in Open-AVB/lib/igb.c)
------------------------------------------
static inline void rdtscpll(uint64_t *val)
{
    uint32_t high, low;
    __asm__ __volatile__("lfence;"
                          "rdtsc;"
                          : "=d"(high), "=a"(low)
                          :
                          : "memory");
    *val = high;
    *val = (*val << 32) | low;
}

static inline void __sync(void)
{
    __asm__ __volatile__("mfence;"
                          :
                          :
                          : "memory");
}
------------------------------------------

1. These code for arm?
2. If answer for 1. is "no code for arm" then,
   Is it okay I block these functions like below?

------------------------------------------
static inline void rdtscpll(uint64_t *val)
{
/* uint32_t high, low;
    __asm__ __volatile__("lfence;"
                          "rdtsc;"
                          : "=d"(high), "=a"(low)
                          :
                          : "memory");
    *val = high;
    *val = (*val << 32) | low; */
}

static inline void __sync(void)
{
/*  __asm__ __volatile__("mfence;"
                          :
                          :
                          : "memory"); */
}
------------------------------------------

Thanks

pinealservo, 2017-06-08:
1. We don't currently have an ARM version of those functions.  I'm not familiar enough with why they're written the way they are to suggest how you'd write ARM versions.

2. That will almost certainly not work. The first one reads the CPU timestamp counter, presumably as a way of measuring some interval; the second one creates a memory barrier to ensure read/write ordering. The ARM memory model has weaker guarantees in that respect than x86's, so you'd be even more likely to run into memory coherency problems.

I would look for some ARM compiler intrinsics or something for reading the cycle counter and making a memory barrier and replace the inline assembly with those.

vitorroriz, 2017-06-21:
@MartinSeungjinHyun I managed to compile the gptp daemon for the imx7 sabre today, I wrote about it on: #362 I hope it helps if you were facing the same issue as me.