Avnu/OpenAvnu #622: Error (TX) timestamping PDelay request, error=-72

zeefpt, 2017-06-06:
Hi Again Folks,

I have the simple talker/listener working good, and the wav file with beep sound too.

Now im trying the audiotestsrc example with gstreamer with this:

Pc 1:
#**gst-launch-1.0 audiotestsrc ! fdsink fd=1 | ./talker eth0 512**

attaching to 000bind to iface 0000:01:00.0
checking against adapter name 0000:01:00.0
0:01:00.0
FOUND STREAM ID=6805ca5132cd0000 with state ready
detected domain Class A PRIO=0 VID=0000...
VNE 0000 R=000000000000 VN/MT
 unhandled from mrpd
advertising stream ...
awaiting a listener ...
FOUND STREAM ID=6805ca5132cd0000 with state ready
added listener
got a listener ...
frame sequence = 0
frame sequence = 1
frame sequence = 2
frame sequence = 3
frame sequence = 4
frame sequence = 5
frame sequence = 6
frame sequence = 7
and are continuos sending the frames..

PC2:
root@pt02ipk1:/home/local/jp/OpenAvnu/examples/live_stream#
 **./listener eth3 512 | gst-launch-1.0 fdsrc fd=0 ! audioconvert ! autoaudiosink**

Msg: D:C=0,P=0,V=0000,N=0 R=6805ca5132cd QA/IN
L:D=2,S=6805ca5132cd0000 R=6805ca5132cd QA/MT

Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Msg: SJO D:C=0,P=0,V=0000,N=0 R=6805ca5132cd QA/IN

Msg: VNE 0000 R=000000000000 VN/MT

Msg: VJO 0000 R=6805ca5132cd QA/IN

Msg: VJO 0000 R=6805ca5132cd QA/IN

Msg: SJO D:C=0,P=0,V=0000,N=0 R=6805ca5132cd VP/IN

Msg: VJO 0000 R=6805ca5132cd VP/IN

Msg: VNE 0000 R=6805ca5132cd QA/IN

Msg: SNE T:S=6805ca5132cd0000,A=91e0f0000e80,V=0000,Z=542,I=1,P=0,L=3900 R=6805ca5132cd VO/IN

Msg: SNE L:D=2,S=6805ca5132cd0000 R=000000000000 VN/MT

Msg: VNE 0000 R=6805ca5132cd QA/IN

Msg: SNE T:S=6805ca5132cd0000,A=91e0f0000e80,V=0000,Z=542,I=1,P=0,L=3900 R=6805ca5132cd VO/IN

frame sequence = 0
frame sequence = 1
frame sequence = 2
WARNING: from element /GstPipeline:pipeline0/GstAudioConvert:audioconvert0: not negotiated
Additional debug info:
gstbasetransform.c(2115): gst_base_transform_handle_buffer (): /GstPipeline:pipeline0/GstAudioConvert:audioconvert0:
not negotiated
frame sequence = 3
ERROR: from element /GstPipeline:pipeline0/GstFdSrc:fdsrc0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2865): gst_base_src_loop (): /GstPipeline:pipeline0/GstFdSrc:fdsrc0:
streaming task paused, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
frame sequence = 4
frame sequence = 5
frame sequence = 6
frame sequence = 7
frame sequence = 8
frame sequence = 9
Freeing pipeline ...
frame sequence = 10
frame sequence = 11
frame sequence = 12
frame sequence = 13
frame sequence = 14
frame sequence = 15
frame sequence = 16
frame sequence = 17
frame sequence = 18
frame sequence = 19
frame sequence = 20
frame sequence = 21
frame sequence = 22
frame sequence = 23
frame sequence = 24
frame sequence = 25
frame sequence = 26
frame sequence = 27
frame sequence = 28
frame sequence = 29
frame sequence = 30
frame sequence = 31
frame sequence = 32
frame sequence = 33
frame sequence = 34
frame sequence = 35
frame sequence = 36
frame sequence = 37
frame sequence = 38
frame sequence = 39
frame sequence = 40
frame sequence = 41
frame sequence = 42
frame sequence = 43
frame sequence = 44
frame sequence = 45
frame sequence = 46
frame sequence = 47
frame sequence = 48
frame sequence = 49
frame sequence = 50
frame sequence = 51
frame sequence = 52
frame sequence = 53
frame sequence = 54
frame sequence = 55
frame sequence = 56

and stops

if i revert the commands in pcs , frames are sending but got the 

GPTP [13:35:12:765] Error (TX) timestamping PDelay request, error=-72

Using:

PC1 :

#**./listener eth0 512 | gst-launch-1.0 fdsrc fd=0 ! audioparse ! alsasink**

Msg: D:C=0,P=0,V=0000,N=0 R=6805ca513381 QA/IN
L:D=2,S=6805ca5132cd0000 R=6805ca513381 VO/IN

Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Msg: SJO L:D=2,S=6805ca5132cd0000 R=6805ca513381 VO/IN

Msg: SJO D:C=0,P=0,V=0000,N=0 R=6805ca513381 VP/IN

Msg: VNE 0000 R=000000000000 VN/MT

Msg: VJO 0000 R=6805ca513381 VP/IN

Msg: SJO L:D=2,S=6805ca5132cd0000 R=6805ca513381 VO/IN

Msg: SJO D:C=0,P=0,V=0000,N=0 R=6805ca513381 QA/IN

Msg: VJO 0000 R=6805ca513381 QA/IN

Msg: SJO L:D=2,S=6805ca5132cd0000 R=6805ca513381 VO/IN

Msg: SJO D:C=0,P=0,V=0000,N=0 R=6805ca513381 VP/IN

Msg: VNE 0000 R=6805ca513381 QA/IN

Msg: SNE T:S=680bind to iface 0000:01:00.0
checking against adapter name 0000:01:00.0
5ca5133810000,A=91e0f0000e80,V=0000,Z=542,I=1,P=0,L=3900 R=6805ca513381 VO/IN

Msg: SNE L:D=2,S=6805ca5133810000 R=000000000000 VN/MT

Msg: VNE 0000 R=6805ca513381 QA/IN

Msg: SNE T:S=6805ca5133810000,A=91e0f0000e80,V=0000,Z=542,I=1,P=0,L=3900 R=6805ca513381 VO/IN

frame sequence = 0
frame sequence = 1
frame sequence = 2
frame sequence = 3
frame sequence = 4
frame sequence = 5
frame sequence = 6
frame sequence = 7
frame sequence = 8
Redistribute latency...
frame sequence = 9
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
frame sequence = 10
New clock: GstAudioSinkClock
frame sequence = 11
frame sequence = 12
frame sequence = 13
frame sequence = 14
frame sequence = 15
frame sequence = 16
frame sequence = 17
frame sequence = 18
frame sequence = 19
frame sequence = 20
frame sequence = 21
frame sequence = 22
frame sequence = 23
frame sequence = 24
frame sequence = 25
frame sequence = 26
frame sequence = 27
frame sequence = 28
frame sequence = 29
frame sequence = 30
frame sequence = 31
frame sequence = 32
frame sequence = 33
frame sequence = 34
frame sequence = 35
frame sequence = 36
frame sequence = 37
frame sequence = 38
frame sequence = 39
frame sequence = 40
frame sequence = 41
frame sequence = 42
frame sequence = 43
frame sequence = 44
frame sequence = 45
frame sequence = 46
frame sequence = 47
frame sequence = 48
frame sequence = 49
frame sequence = 50
frame sequence = 51

PC2:
**gst-launch-1.0 audiotestsrc ! fdsink fd=1 | ./talker eth3 512**

root@pt02ipk1:/home/local/jp/OpenAvnu/examples/live_stream# 
gst-launch-1.0 audiotestsrc ! fdsink fd=1 | ./talker eth3 512
attaching to 0000:01:00.0
FOUND STREAM ID=6805ca5132cd0000 with state ready
FOUND STREAM ID=6805ca5133810000 with state ready
detected domain Class A PRIO=0 VID=0000...
VNE 0000 R=000000000000 VN/MT
 unhandled from mrpd
advertising stream ...
awaiting a listener ...
FOUND STREAM ID=6805ca5132cd0000 with state ready
FOUND STREAM ID=6805ca5133810000 with state ready
added listener
got a listener ...
frame sequence = 0
frame sequence = 1
frame sequence = 2
frame sequence = 3
frame sequence = 4
frame sequence = 5
frame sequence = 6
frame sequence = 7
frame sequence = 8
frame sequence = 9
frame sequence = 10
frame sequence = 11
frame sequence = 12
frame sequence = 13
frame sequence = 14
frame sequence = 15
frame sequence = 16
frame sequence = 17
frame sequence = 18
frame sequence = 19
frame sequence = 20
frame sequence = 21
frame sequence = 22
frame sequence = 23
frame sequence = 24
frame sequence = 25
frame sequence = 26
frame sequence = 27
frame sequence = 28

... always sending..

Im really stuck on this. 

Anyhelp/tip are welcome :)
Myquetions:

1- my  gstream command are right?
2- Anyone know a example for sending the videotestsrc via listener/talker too?
3- Should i ear sound , and gstream GUI interface automatically opens too?

Best Regards,

pinealservo, 2017-06-08:
I don't know if any of the current active maintainers use any of the gstreamer examples. Learning to configure gstreamer is complex and outside of the scope of what we have time to support here.

I would suggest moving on from the code in the examples directory and looking at the code in the avtp pipeline section. I'm actively working with that code, and it's meant to actually work instead of just be an example.

Please don't spam other issues trying to get attention. We all volunteer support time here as we are able to, and we don't always have time to dig into things we're unfamiliar with. If you don't get an answer, there may not be anyone who knows or has time to help. You may have to keep digging and figure things out yourself.

zeefpt, 2017-06-08:
Hi pinealservo,

Thanks for anwser,

I will move to avtp pipeline from your suggestion.

Regards