Avnu/OpenAvnu #623: Intel i210 Hardware Timestamping with simple_talker

riteshjaiswal2016, 2017-06-06:
Hello,
I am using intel i210 board for AVB communication. My configuration is 
PC1(Simple_talker)<--->AVB Switch<--->PC2(Simple_Listener)

Since, gPTP stores the timestamp values of it's synchronization messages in register of Intel i210 board. So, Can you please guide me on how can I enable this functionality of Intel i210 board for simple_talker application so that when I use simple_talker to send packets, I can retrieve those timestamp values from registers to print.

andrew-elder, 2017-06-06:
I don't think you are going to be able to print 8000 timestamps a second. Why do you need to observe the timestamps of the talker packets?

riteshjaiswal2016, 2017-06-06:
Thanks for your response. I am working on Testbed for Realtime AVB Communication. Actually, I want to print timestamp of a specific packet transmitted from simple_talker (Suppose when 10th packet from simple_talker is get transmitted then I wanted to print it's corresponding hardware timestamp, stored in intel i210 register).

jdkoftinoff, 2017-06-07:
What exactly are you expecting to test?

The IEEE 1722 ethernet frame contains the presentation timestamp in the avtp_timestamp field. All that matters is that the media packet is received early enough to process it before this time.

Jeff

riteshjaiswal2016, 2017-06-07:
Actually, I want to measure the precise time when a particular packet is just get sent through NIC using simple_talker on first PC and the time when this packet is just get received at the NIC of other PC using simple_rx and then I will calculate the difference between those times to get the exact transmission time.
Since, gPTP is able to record the hardware timestamp values of transmitted and received packets in register of intel i210. So, I want to implement that technique in simple_talker and simple_rx so that I can measure the precise time when a packet is just get sent or received through NIC. 
I think avtp_timestamp contains presentation time which is not required in this case. So, can you please guide me on the above mentioned issue?

andrew-elder, 2017-06-07:
If I were trying to do this, I would remove the switch and try to setup to timestamp the incoming packets at the receiver. I know the i210 can be configured timestamp every incoming packet, but I'm not sure that all the interfaces are in place to pick out those timestamps. It is somewhat of an R&D project that is for sure.

Note the gPTP computes the delay (or transmission time) between two 802.1AS capable network interfaces. It is designed to do that.

Or, are you trying to time how long it takes a packet to travel through a switch?

zeefpt, 2017-06-07:
Hi folks can you please review my issues ?

#622
#619
Im really stucked.

Best Regards ;

pinealservo, 2017-06-08:
You may find some information in this thread helpful: #564

We don't intend to add anything to do the sort of measurements you're talking about; if you end up having to make some changes to accomplish your goal and you'd like to contribute them, we'll definitely consider merging them if it's clear they don't interfere with the current operation or maintenance of the code. But I think you may be able to use existing features.

I'm afraid we don't have a lot of support bandwidth to help, though, as it's a bit beyond the current project scope, and we're stretched very thin with the scope as it is.

riteshjaiswal2016, 2017-06-09:
I think I found the code where gPTP is receiving the timestamp of transmitted and received packets. In linux_hal_generic.cpp. For rx Hardware timestamps. Inside- 

net_result LinuxNetworkInterface::nrecv( LinkLayerAddress *addr, uint8_t *payload, size_t &length,struct phy_delay *delay ), there is a function call- 
err = recvmsg( sd_event, &msg, 0 ); 

When a packet is get received at NIC, a message with ancillary data is get generated which contains the Hardware timestamp of reception of that packet. Note that we need to configure the socket to work like that, which is implemented in-

bool LinuxTimestamperGeneric::post_init( int ifindex, int sd, TicketingLock *lock ),  there is a function call-
err = setsockopt( sd, SOL_SOCKET, SO_TIMESTAMPING, &timestamp_flags,sizeof(timestamp_flags) ); 

For tx Hardware timstamps, whenever a packet is get transmitted from NIC, the hardware timstamp value is reported back in MSG_ERRQUEUE so need to turn on the flag when we call the recvmsg
Inside-

int LinuxTimestamperGeneric::HWTimestamper_txtimestamp( PortIdentity *identity, uint16_t sequenceId, Timestamp &timestamp,unsigned &clock_value, bool last ) there is function call-
err = recvmsg( sd, &msg, MSG_ERRQUEUE );

@andrew-elder : Yes, I am trying to measure the precise time a packet takes after it transmitted from NIC of one PC which is running simple_talker, travels through switch and get received at NIC of other PC which is running simple_rx. From your statement-
"Note the gPTP computes the delay (or transmission time) between two 802.1AS capable network interfaces. It is designed to do that."
Can you please guide me on where can I find that transmission time computation in the code of gPTP?

Questions:
1).Please check the correctness of my understanding related with how gPTP is getting Hardware timestamps?

2).In linux_hal_generic.cpp-
Timestamp device, system;
What are they (device, system) representing here and is the difference between them? and why this modification-
device = device + latency; (In int LinuxTimestamperGeneric::HWTimestamper_txtimestamp)
device = device - latency; (In net_result LinuxNetworkInterface::nrecv)

3).what these variables inside the structure represents and for what purpose they are used in the code-
struct phy_delay
{
   int mb_tx_phy_delay;
   int mb_rx_phy_delay;
   int gb_tx_phy_delay;
   int gb_rx_phy_delay;
};

4).In net_result LinuxNetworkInterface::nrecv
( LinkLayerAddress *addr, uint8_t *payload, size_t &length,struct phy_delay *delay )
Is payload representing the received packet's payload?

andrew-elder, 2017-06-09:
I can answer 3) easily. The gPTP timestamp is supposed to be taken at the RJ45 connector. But on the i210 a received packet passes through the RJ45, then the PHY and is finally time stamped using an internal counter. The PHY delay has to be subtracted from the Rx time stamp so as to transfer the measurement location from the internal timestamping hardware to the PJ45 connector.

andrew-elder, 2017-06-09:
And 2), system time is based on an internal CPU counter (it is free running with respect to gPTP). Device time is based on the time stamp counter running on the I210. Device time is the time base used for timestamps from the I210. Neither of these is gPTP time.

riteshjaiswal2016, 2017-06-09:
@andrew-elder : Thank you for your response. Can you please explain me what is gPTP time? Is it the time which is get set after synchronization?

andrew-elder, 2017-06-09:
gPTP time is computed using math inside gPTP. The GrandMaster distributes gPTP time and all other devices do time stamping and some math to figure out how to compute gPTP time from their local I210 device time.

riteshjaiswal2016, 2017-06-09:
One more thing-
5). In linux_hal_generic.cpp, we are using "recvmsg" to receive the message which contains ancillary data having hardware timestamp. I want to ask that where is the "sendmsg" in the code which is responsible for sending these messages?

jdkoftinoff, 2017-06-09:
Inside the kernel.

riteshjaiswal2016, 2017-06-09:
@jdkoftinoff : You mean that we only need to use "recvmsg" in our code and the KERNEL will automatically send those messages (by taking required parameters from "recvmsg" of the code) which we are trying to receive using "recvmsg"??

pinealservo, 2017-06-09:
The messages are sent by the kernel to the gPTP daemon because we ask for them by setting specific socket options related to timestamping. Here is the relevant documentation: https://www.kernel.org/doc/Documentation/networking/timestamping.txt

However, none of this will help you with getting AVTP transmit timestamps when you use the igb library and kernel driver to send your packets. Look through the issue comment thread I linked to above again, especially this comment: https://github.com/AVnu/OpenAvnu/issues/564#issuecomment-285535432

andrew-elder, 2017-06-10:
@pinealservo - I reread your comment about dmatime and I seem to recall from previous study of the I210 datasheet that I concluded that the dmatime was related to when memory copy operations occurred and not the time the packet was actually put on the wire. My 2c.

riteshjaiswal2016, 2017-06-11:
@andrew-elder and @pinealservo : Hello, 
Yes, I think @andrew-elder is right. Launch time of packet, which is represented as "attime" field, will be more precise transmit time than the time represented by "dmatime". From the datasheet-
[](https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/i210-ethernet-controller-datasheet.pdf)
Page-285, 7.2.2.2.3
Launch time is the time when packet is transmitted from NIC buffer and it can be calculated using "attime" field as following-
Launch time = 32*(Launch offset+ "attime" field value)

Could you please confirm that If I add gb_tx_phy_delay to Launch time then should I get the precise time when packet got transmitted from RJ45?

andrew-elder, 2017-06-12:
Could you please confirm that If I add gb_tx_phy_delay to Launch time then should I get the precise time when packet got transmitted from RJ45?

Yes, I think so, but I don't think there is a mechanism to retrieve the launch time timestamps fast enough. 8,000 a second is rather fast to be reading the timestamp queue.

pinealservo, 2017-06-12:
@andrew-elder I think the Tx timestamp queue you're thinking of is only operational on 1588 frames. I think the DMA writeback timestamp (which should correspond closely to what you set in the launch time field, if you set it) + the LaunchOffset field in LAUNCH_OS0 and maybe something to do with FetchTimDelta is the closest you can get to 1722 transmit timestamps. I think the delay should be deterministic since each descriptor queue has its own packet buffer.

Because this value is written into the descriptor rather than a hardware queue (the way 1588 Tx timestamps are), there shouldn't be any trouble copying the timestamps out of the descriptors in the descriptor cleanup code in your talker.

I don't have any deep knowledge beyond the datasheet, so I can't confirm this interpretation is correct. I have to reiterate that this isn't a normal AVB use case and is not even a really interesting value for a test bench.

closed 2017-06-12 by pinealservo