Avnu/OpenAvnu #625: i210: Questions about the queueing architecture

davidcemin, 2017-06-08:
Hi, 

I just start reading the igb code and I'm trying to understand the libigb architecture / igb driver with regards to the TX/RX queues on the i210.

I would like to have time sensitive traffic along with normal Lan traffic and be able to control which queue I am using for performance purposes. I210 has 4 queues, and what I gathered from the datasheet (https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/i210-ethernet-controller-datasheet.pdf) is:
- Queue 0 is the one with highest priority and queue 3 is the one with lowest priority
- Queues 0/1 are the Strict Reservation (SR) queues when in Qav mode
- Queues 2/3 are used for best effort traffic when in Qav mode (Strict Priority policy)

Questions:

1. Would it be possible to configure both queues 0 and 1 to send traffic with the same priority ? (i.e. use both queues for time sensitive traffic and let the NIC decide when to put them on the wire based on the launch time for instance)
2. The function igb_select_queue defined at igb_main.c:5699 (callback for ndo_select_queue) is always returning 3. Does that mean that we are never using queue 2 for best effort traffic ?

pinealservo, 2017-06-10:
1. If you are in Qav mode and you put 0 and 1 in SR (stream reservation) mode, you are always going to get strict priority scheduling between them as well. So if both 0 and 1 are ready to transmit a packet, 0 will always go first.

If you set DataTranTIM to 1, a queue will not win arbitration if the head of its queue has a launch time set and that time is still in the future. It only looks at the head, so you have to queue them in launch time order. If you use this mode and schedule the launch time of all your SR packets, you could split them between queues 0 and 1 (assuming you set them both to SR mode) and it ought to behave pretty close to as if 0 and 1 were the same priority. This could be useful in multi-core scenarios where both cores could be generating SR traffic.

If you also have DataTranARB set to 1 (for FQTSS shaping) and split a single class's bandwidth between 0 and 1, you will probably not get the behavior you want unless you also use DataTranTIM and clever blocking of the queues with scheduled frames; I would not recommend trying to do this, but it just might be helpful in difficult cases. Or it might not.

2. See net/core/dev.c and search for ops->ndo_select_queue to see how ndo_select_queue is used by the kernel; I think it's not the only criteria used for selection, but I don't recall the details.

davidcemin, 2017-06-27:
Thanks for the detailed answer @pinealservo. If I have more question I will reopen this issue.

closed 2017-06-27 by davidcemin

joyoseller20100802, 2023-04-25:
> 1. If you are in Qav mode and you put 0 and 1 in SR (stream reservation) mode, you are always going to get strict priority scheduling between them as well. So if both 0 and 1 are ready to transmit a [...]

Thank your for your sharing ,how can i get similar material?