Avnu/OpenAvnu #629: Simple_talker and Simple_rx are bypassing the Linux Network Stack?

riteshjaiswal2016, 2017-06-18:
Hello,
My configuration is-
PC2<-->AVB Switch<--->PC2

simple_talker and simple_rx are using functions of files present in OpenAVB/lib/igb (like igb_xmit, igb_receive functions is igb.c, presented in above mentioned directory) to send/receive media packets.

Questions:
(1). So, in this way they are bypassing the linux network stack by dealing with NIC i210 DIRECTLY from user space without interacting with linux network stack present in kernel. Is this right?

(2). Are they using the concept of User Space Device Driver to bypass the linux stack?

(3). When we run gPTP using run_daemons.sh to synchronize the PCs with AVB Switch, it uses igb_avb.ko kernel driver module to interact with NIC i210 BY GOING THROUGH linux network stack. Is this right?

Thanks in advance!

pinealservo, 2017-06-19:
1) Yes, sort of.

2) I don't know exactly what you mean by "the concept of User Space Device Driver". The words sort of describe what's happening, but it looks like you mean something specific that may not be exactly what's happening.  In reality, the igb_avb.ko module is still running the best-effort queues and it exposes some special ioctls that the igb.c library uses to set up access to the SR class queues and get kernel-allocated memory page addresses, so it's not entirely managed from user space.

3) Yes, the kernel stack has support for IEEE 1588 timestamping and provides userspace APIs to access those features.  If the kernel fully supported the other AVB features of i210, we'd probably be using it for those as well.

riteshjaiswal2016, 2017-06-19:
Hello @pinealservo,
Thanks for your response! 
When we use simple_talker.c then we can not capture 1722 AVB Packets using wireshark at talker side. Reason behind this is somewhat bypassing the Linux Network Stack by using OpenAVB/lib/igb at user-space to interact directly with NIC. Am I right?

pinealservo, 2017-06-20:
Yes, 1722 packets bypass the kernel stack and thus don't show up in a Wireshark session on the talker.

closed 2017-06-20 by riteshjaiswal2016