Avnu/OpenAvnu #632: Gptp link_delay value always 0

stujeffrey016, 2017-06-20:
Hi there,
I use gptp program on the i.mx6q platform, which has the 1588 hw_timestamp. The packet seems normally but link_delay value always 0. It causes the program stuck at the Pdelay process never go to BMCA and rate ratio part.

`root@linaro-alip:/home/OpenAvnu-open-avb-next# ./run_gptp.sh eth0 -INITPDELAY 0
groupadd: group 'ptp' already exists
INFO     : GPTP [21:51:01:065] gPTP starting
ERROR    : GPTP [21:51:01:068] Using clock device: /dev/ptp0
DEBUG    : GPTP [21:51:01:068] [../../common/common_port.cpp:558] Received POWERUP/INITIALIZE event
STATUS   : GPTP [21:51:01:068] Starting PDelay
DEBUG    : GPTP [21:51:01:069] [../src/linux_hal_common.cpp:254] Interface eth0 is up
STATUS   : GPTP [21:51:01:069] Link Speed: 1000000 kb/sec
DEBUG    : GPTP [21:51:01:084] [../../common/common_port.cpp:608] ANNOUNCE_INTERVAL_TIMEOUT_EXPIRES occured
DEBUG    : GPTP [21:51:01:100] [../../common/ether_port.cpp:496] PDELAY_INTERVAL_TIMEOUT_EXPIRES occured
DEBUG    : GPTP [21:51:01:101] [../../common/ptp_message.cpp:573] TX PHY compensation: 0 0.000000184 sec
DEBUG    : GPTP [21:51:01:101] [../../common/ether_port.cpp:519] *** Sent PDelay Request message
DEBUG    : GPTP [21:51:01:101] [../../common/ether_port.cpp:536] Schedule PDELAY_RESP_RECEIPT_TIMEOUT_EXPIRES, PDelay interval 0, timeout 3000000000
DEBUG    : GPTP [21:51:01:102] [../../common/ptp_message.cpp:290] *** Received PDelay Response message, Timestamp 78661 (sec) 101879622 (ns), seqID 0
DEBUG    : GPTP [21:51:01:102] [../../common/ether_port.cpp:232] RX PHY compensation: 0 0.000000382 sec
DEBUG    : GPTP [21:51:01:102] [../../common/ptp_message.cpp:349] *** Received PDelay Response FollowUp message
DEBUG    : GPTP [21:51:01:102] [../../common/ptp_message.cpp:1664] Link delay: 0 ns
STATUS   : GPTP [21:51:01:102] AsCapable: Enabled
DEBUG    : GPTP [21:51:01:993] [../../common/ptp_message.cpp:241] *** Received PDelay Request message
DEBUG    : GPTP [21:51:01:993] [../../common/ether_port.cpp:232] RX PHY compensation: 0 0.000000382 sec
DEBUG    : GPTP [21:51:01:993] [../../common/ptp_message.cpp:1201] Process PDelay Request SeqId: 16053
DEBUG    : GPTP [21:51:01:993] [../../common/ptp_message.cpp:573] TX PHY compensation: 0 0.000000184 sec
DEBUG    : GPTP [21:51:01:993] [../../common/ptp_message.cpp:1215] *** Sent PDelay Response message
DEBUG    : GPTP [21:51:01:994] [../../common/ptp_message.cpp:1253] *** Sent PDelay Response FollowUp message
DEBUG    : GPTP [21:51:02:101] [../../common/ether_port.cpp:496] PDELAY_INTERVAL_TIMEOUT_EXPIRES occured
DEBUG    : GPTP [21:51:02:107] [../../common/ptp_message.cpp:573] TX PHY compensation: 0 0.000000184 sec
DEBUG    : GPTP [21:51:02:107] [../../common/ether_port.cpp:519] *** Sent PDelay Request message
DEBUG    : GPTP [21:51:02:107] [../../common/ether_port.cpp:536] Schedule PDELAY_RESP_RECEIPT_TIMEOUT_EXPIRES, PDelay interval 0, timeout 3000000000
DEBUG    : GPTP [21:51:02:107] [../../common/ptp_message.cpp:290] *** Received PDelay Response message, Timestamp 78662 (sec) 107451270 (ns), seqID 1
DEBUG    : GPTP [21:51:02:107] [../../common/ether_port.cpp:232] RX PHY compensation: 0 0.000000382 sec
DEBUG    : GPTP [21:51:02:107] [../../common/ptp_message.cpp:349] *** Received PDelay Response FollowUp message
DEBUG    : GPTP [21:51:02:108] [../../common/ptp_message.cpp:1664] Link delay: 0 ns
DEBUG    : GPTP [21:51:02:993] [../../common/ptp_message.cpp:241] *** Received PDelay Request message
DEBUG    : GPTP [21:51:02:994] [../../common/ether_port.cpp:232] RX PHY compensation: 0 0.000000382 sec
DEBUG    : GPTP [21:51:02:994] [../../common/ptp_message.cpp:1201] Process PDelay Request SeqId: 16054
DEBUG    : GPTP [21:51:02:994] [../../common/ptp_message.cpp:573] TX PHY compensation: 0 0.000000184 sec
DEBUG    : GPTP [21:51:02:994] [../../common/ptp_message.cpp:1215] *** Sent PDelay Response message
DEBUG    : GPTP [21:51:02:994] [../../common/ptp_message.cpp:1253] *** Sent PDelay Response FollowUp message
DEBUG    : GPTP [21:51:03:107] [../../common/ether_port.cpp:496] PDELAY_INTERVAL_TIMEOUT_EXPIRES occured
DEBUG    : GPTP [21:51:03:123] [../../common/ptp_message.cpp:573] TX PHY compensation: 0 0.000000184 sec
DEBUG    : GPTP [21:51:03:123] [../../common/ether_port.cpp:519] *** Sent PDelay Request message
DEBUG    : GPTP [21:51:03:124] [../../common/ether_port.cpp:536] Schedule PDELAY_RESP_RECEIPT_TIMEOUT_EXPIRES, PDelay interval 0, timeout 3000000000
DEBUG    : GPTP [21:51:03:124] [../../common/ptp_message.cpp:290] *** Received PDelay Response message, Timestamp 78663 (sec) 124228918 (ns), seqID 2
DEBUG    : GPTP [21:51:03:124] [../../common/ether_port.cpp:232] RX PHY compensation: 0 0.000000382 sec
DEBUG    : GPTP [21:51:03:124] [../../common/ptp_message.cpp:349] *** Received PDelay Response FollowUp message
DEBUG    : GPTP [21:51:03:124] [../../common/ptp_message.cpp:1664] Link delay: 0 ns
^CERROR    : GPTP [21:51:03:860] Exiting on 2`

Best Regards,
Jeffrey.

pinealservo, 2017-06-20:
Please turn up the debug level to "verbose" so that we can see the actual timestamps involved in the calculations. In fact, after doing so (which I know you have done before because your previous issue thread includes at least one log with the VERBOSE messages) you should be able to figure out the next step to take in investigating this. In case you are still feeling stuck, here are some things to look at:

1. Do the log timestamps show that you're sending/receiving the packets you're expecting to at the rate you're expecting them? If not, can you fix it by adjusting the values in your configuration or command line?

2. Do the 1588 timestamps captured locally and displayed in the verbose log messages seem to be increasing at roughly the same rate as the log timestamps for the packets they're associated with?

3. Does time seem to be moving forward in the series of timestamps that are collected at the end of the PDelay process and used to calculate the delay?  If not, can it be corrected by fixing the PHY delay values in your configuration?

From what's in that log, I would guess that the problem is 3, but it's impossible to say for sure without seeing the timestamps that lead to the "Link delay: 0 ns" calculation.

stujeffrey016, 2017-06-21:
Hi @pinealservo ,
 I turned on the verbose log to check the issue you had mentioned. It seems normally, my PDelay message sent once per second, and timestamp was increasing. I modified the PHY delay value as 0, but the result still the same. Here's the verbose message. By the way, when I turn on the verbose log, the link_delay value keep increasing(per pDelay message).

root@linaro-alip:/home/OpenAvnu-open-avb-next# ./run_gptp.sh eth0 -INITPDELAY 0
groupadd: group 'ptp' already exists
INFO     : GPTP [21:34:54:685] gPTP starting
ERROR    : GPTP [21:34:54:691] Using clock device: /dev/ptp0
DEBUG    : GPTP [21:34:54:691] [../../common/common_port.cpp:558] Received POWERUP/INITIALIZE event
STATUS   : GPTP [21:34:54:691] Starting PDelay
DEBUG    : GPTP [21:34:54:692] [../src/linux_hal_common.cpp:254] Interface eth0 is up
STATUS   : GPTP [21:34:54:692] Link Speed: 1000000 kb/sec
DEBUG    : GPTP [21:34:54:707] [../../common/common_port.cpp:608] ANNOUNCE_INTERVAL_TIMEOUT_EXPIRES occured
DEBUG    : GPTP [21:34:54:723] [../../common/ether_port.cpp:496] PDELAY_INTERVAL_TIMEOUT_EXPIRES occured
VERBOSE  : GPTP [21:34:54:724] [../../common/ptp_message.cpp:626] Sending Sequence Id: 0
DEBUG    : GPTP [21:34:54:724] [../../common/ptp_message.cpp:573] TX PHY compensation: 0 0.000000184 sec
DEBUG    : GPTP [21:34:54:724] [../../common/ether_port.cpp:519] *** Sent PDelay Request message
DEBUG    : GPTP [21:34:54:725] [../../common/ether_port.cpp:536] Schedule PDELAY_RESP_RECEIPT_TIMEOUT_EXPIRES, PDelay interval 0, timeout 3000000000
VERBOSE  : GPTP [21:34:54:725] [../../common/ether_port.cpp:215] Processing network buffer
VERBOSE  : GPTP [21:34:54:725] [../../common/ptp_message.cpp:118] Captured Sequence Id: 0
VERBOSE  : GPTP [21:34:54:725] [../../common/ptp_message.cpp:153] Timestamping event packet
DEBUG    : GPTP [21:34:54:725] [../../common/ptp_message.cpp:290] *** Received PDelay Response message, Timestamp 164094 (sec) 724890894 (ns), seqID 0
VERBOSE  : GPTP [21:34:54:725] [../../common/ether_port.cpp:225] Processing message
DEBUG    : GPTP [21:34:54:725] [../../common/ether_port.cpp:232] RX PHY compensation: 0 0.000000382 sec
VERBOSE  : GPTP [21:34:54:726] [../../common/ether_port.cpp:215] Processing network buffer
VERBOSE  : GPTP [21:34:54:726] [../../common/ptp_message.cpp:118] Captured Sequence Id: 0
DEBUG    : GPTP [21:34:54:726] [../../common/ptp_message.cpp:349] *** Received PDelay Response FollowUp message
VERBOSE  : GPTP [21:34:54:726] [../../common/ether_port.cpp:225] Processing message
VERBOSE  : GPTP [21:34:54:726] [../../common/ptp_message.cpp:1582] Request Sequence Id: 0
VERBOSE  : GPTP [21:34:54:726] [../../common/ptp_message.cpp:1583] Response Sequence Id: 0
VERBOSE  : GPTP [21:34:54:726] [../../common/ptp_message.cpp:1584] Follow-Up Sequence Id: 0
VERBOSE  : GPTP [21:34:54:726] [../../common/ptp_message.cpp:1655] Turn Around Adjustment 0.000000
VERBOSE  : GPTP [21:34:54:726] [../../common/ptp_message.cpp:1658] Step #1: Turn Around Adjustment 443152.000000
VERBOSE  : GPTP [21:34:54:726] [../../common/ptp_message.cpp:1659] Adjusted Peer turn around is 443152
DEBUG    : GPTP [21:34:54:727] [../../common/ptp_message.cpp:1664] Link delay: 0 ns
STATUS   : GPTP [21:34:54:727] AsCapable: Enabled
VERBOSE  : GPTP [21:34:55:033] [../../common/ether_port.cpp:215] Processing network buffer
VERBOSE  : GPTP [21:34:55:033] [../../common/ptp_message.cpp:118] Captured Sequence Id: 35903
VERBOSE  : GPTP [21:34:55:033] [../../common/ptp_message.cpp:153] Timestamping event packet
DEBUG    : GPTP [21:34:55:033] [../../common/ptp_message.cpp:241] *** Received PDelay Request message
VERBOSE  : GPTP [21:34:55:033] [../../common/ether_port.cpp:225] Processing message
DEBUG    : GPTP [21:34:55:033] [../../common/ether_port.cpp:232] RX PHY compensation: 0 0.000000382 sec
DEBUG    : GPTP [21:34:55:033] [../../common/ptp_message.cpp:1201] Process PDelay Request SeqId: 35903
VERBOSE  : GPTP [21:34:55:033] [../../common/ptp_message.cpp:626] Sending Sequence Id: 35903
VERBOSE  : GPTP [21:34:55:033] [../../common/ptp_message.cpp:1429] PDelay Resp Timestamp: 164095,32911216
DEBUG    : GPTP [21:34:55:033] [../../common/ptp_message.cpp:573] TX PHY compensation: 0 0.000000184 sec
DEBUG    : GPTP [21:34:55:033] [../../common/ptp_message.cpp:1215] *** Sent PDelay Response message
VERBOSE  : GPTP [21:34:55:034] [../../common/ptp_message.cpp:1238] Response Depart(sec): 164095
VERBOSE  : GPTP [21:34:55:034] [../../common/ptp_message.cpp:1239] Request Arrival(sec): 164095
VERBOSE  : GPTP [21:34:55:034] [../../common/ptp_message.cpp:1240] #1 Correction Field: 0
VERBOSE  : GPTP [21:34:55:034] [../../common/ptp_message.cpp:1244] #2 Correction Field: 33785422
VERBOSE  : GPTP [21:34:55:034] [../../common/ptp_message.cpp:1248] #3 Correction Field: 874206
VERBOSE  : GPTP [21:34:55:034] [../../common/ptp_message.cpp:626] Sending Sequence Id: 35903
VERBOSE  : GPTP [21:34:55:034] [../../common/ptp_message.cpp:1757] PDelay Resp Timestamp: 164095,33785422
DEBUG    : GPTP [21:34:55:034] [../../common/ptp_message.cpp:1253] *** Sent PDelay Response FollowUp message
DEBUG    : GPTP [21:34:55:725] [../../common/ether_port.cpp:496] PDELAY_INTERVAL_TIMEOUT_EXPIRES occured
VERBOSE  : GPTP [21:34:55:725] [../../common/ptp_message.cpp:626] Sending Sequence Id: 1
DEBUG    : GPTP [21:34:55:727] [../../common/ptp_message.cpp:573] TX PHY compensation: 0 0.000000184 sec
[... cut ...]
VERBOSE  : GPTP [21:34:59:036] [../../common/ptp_message.cpp:1244] #2 Correction Field: 35665390
VERBOSE  : GPTP [21:34:59:036] [../../common/ptp_message.cpp:1248] #3 Correction Field: 845910
VERBOSE  : GPTP [21:34:59:036] [../../common/ptp_message.cpp:626] Sending Sequence Id: 35907
VERBOSE  : GPTP [21:34:59:036] [../../common/ptp_message.cpp:1757] PDelay Resp Timestamp: 164099,35665390
DEBUG    : GPTP [21:34:59:036] [../../common/ptp_message.cpp:1253] *** Sent PDelay Response FollowUp message

pinealservo, 2017-06-21:
Just blindly modifying the phy delays to 0 is not likely to help. There is a transmit delay and a receive delay; they're probably not symmetric and they affect the calculation in different ways. You should look at how the t1 through t4 timestamps go into the calculation of link delay and think about how your phy delay values will affect the timestamps and thus the resulting calculations. You might also want to think about why the delay value seems to change when you have extra debugging information turned on. Look at the code so you can see what calculations all those numbers represent.

Remember, you're trying to measure the delay caused by the time it takes for electrons to travel through your cable.  If you have really short cables, the actual delay really will be near zero, and any error in your phy delay compensation values could easily make the result of the calculation negative on one end or the other.  Here's an interesting illustration of the issue: https://youtu.be/JEpsKnWZrJ8

I am not going to spend time calculating things for you if you just blindly tweak things and dump logs. Get out some paper and work the math. Add more debug statements to figure out what the numbers are if you're missing some. Look at the actual timestamp values in packets if you need to. You can figure this out without asking at every step. If you find actual evidence of a bug and not just poor calibration, try to fix it or at least give a clear explanation of what's going wrong.

stujeffrey016, 2017-06-21:
Hi @pinealservo,

Thanks for your information. I'll calculate the phy_delay value and make sure my AS workable.

Best regards,
Jeffrey.