Avnu/OpenAvnu #644: Access to the gPTP measured propagation delay.

riteshjaiswal2016, 2017-06-28:
Hello,
My configuration is-
PC2<-->AVB Switch<-->PC1

I synchronized both the PCs with AVB Switch.
Since gPTP will calculate the propagation delay between PC1 and PC2 for it's synchronization purposes. So, could you please guide me on how can I access that propagation delay value? 

Thanks in advance!

andrew-elder, 2017-06-28:
gPTP will measure the propagation delay between PC2 and the AVB swtich, not between PC1 and PC2.

I looked here https://github.com/AVnu/OpenAvnu/blob/master/daemons/gptp/common/ipcdef.hpp and is seems like the neighborPropagationDelay may not currently be published. gPTP definitely knows what it is though. Hopefully someone more familiar than I am with the code will jump in here.

bdthomsen, 2017-06-28:
The gPTP daemon copies values to shared memory, and those values are then accessible to other applications.  You might find the commit https://github.com/AVnu/OpenAvnu/commit/42f95189d782ae956282014d18a3b9a37d8d4fdf useful, as that added some new values to the shared memory.  Look at https://github.com/AVnu/OpenAvnu/blob/master/lib/common/avb_gptp.h for guidance accessing the shared values.  Note that you would need to add neighborPropagationDelay to both the daemon and the access code for it to be useful.

riteshjaiswal2016, 2017-06-28:
Thank you for the response but I didn't able to find exactly where is the data in above references which I can use to calculate the propagation delay. Could you please be more specific?

andrew-elder, 2017-06-28:
@riteshjaiswal2016 - where exactly have you looked so far? Maybe search for setLinkDelay()

pinealservo, 2017-06-29:
We aren't able to provide the level of support you're asking for; the code is available and enough information has been provided here for interested and competent parties to make the desired changes. If the information provided thus far in the issue is not sufficient for you, you'll have to find an alternative means for support.

closed 2017-06-29 by pinealservo

riteshjaiswal2016, 2017-06-29:
Thank You @andrew-elder ! Through your guidance I have found what I was looking for. Previously, I was trying to find that in 
- https://github.com/AVnu/OpenAvnu/blob/master/daemons/gptp/common/ipcdef.hpp  
- https://github.com/AVnu/OpenAvnu/blob/master/lib/common/avb_gptp.h
but it was not there.