Avnu/OpenAvnu #650: Q on SRP bandwidth reservation

JohnRama, 2017-07-08:
Hi, guys

I have a question on SRP protocol.

In my understanding the stream class affects the behavior of FQTSS in terms of queuing priority and the frequency of the message. 

My question is does the stream class affects the bandwidth reservation process ?

For example, assuming that there is a system which has already has almost full bandwidth for AVB traffic, 
let's say there are 4 stream allocated, stream A, B, C is class A and stream D is class B. 
if new AVB device is added to the network, and it tries to send talker advertisement for new stream of "Class A"(stream E), what happens ? 

stream D will be replaced with stream E because of the priority ?
of switch just send back by "talker failed" because there is no bandwidth available ?

Any feedback are really appreciated.

Jonh Rama

pinealservo, 2017-07-09:
Hi John,

The behavior depends on the topology of the network and how much bandwidth is available on each link. The decision as to whether a reservation can be made is done on a port-by-port basis during the attribute propagation process. If a Talker Advertise is registered on one port of a switch, it will try to propagate it to all the other ports belonging to the SRP domain. If the active reservations on a port don't leave enough bandwidth for the advertised stream, the normal situation (regardless of stream class) is to propagate the declaration as a Talker Failed instead, indicating in the FailureCode field that there was insufficient bandwidth on the link where it was changed.

However, there is another bit in the stream t-spec that sets the "rank" of the stream; it can be a standard non-emergency stream or an emergency stream. If a new talker declares a Talker Advertise for a stream with emergency rank, a switch will tear down existing non-emergency streams in order to ensure there is bandwidth for it. This rank parameter is completely independent of traffic class, though.

If you haven't seen it yet, you might find my SRP Tutorial document helpful:  http://avnu.org/wp-content/uploads/2014/05/AVnu_Stream-Reservation-Protocol-v1.pdf

makotoharada, 2017-07-09:
Hi, Levi

Great thanks for your reply.
> If the active reservations on a port don't leave enough bandwidth for the advertised stream, the normal situation (regardless of stream class) is to propagate the declaration as a Talker Failed [...]
Thanks what I wanted to confirm. Thank you !!
I'm fine to close this issue.

Makoto

JohnRama, 2017-07-09:
Thanks, Levi

> If the active reservations on a port don't leave enough bandwidth for the advertised stream, the normal situation (regardless of stream class) is to propagate the declaration as a Talker Failed [...]

Thanks what I wanted to confirm. Thank you !!
I'm fine to close this question. 

John

closed 2017-07-10 by andrew-elder