Avnu/OpenAvnu #656: Raspberrypi with AVB

dnpawate, 2017-07-17:
Hi,

I'm looking to run the simple talker and listener using Raspberrypi
Is this possible?
Basic Idea
(TALKER)RPi->I210-----<one to one> with I210-->RPI(Listener)

Please provide your inputs, to make the setup using RPI

andrew-elder, 2017-07-17:
I don't think a RPi supports hardware timestamping for gPTP, so it can't do AVB.

dnpawate, 2017-07-17:
Hi,

Thanks for the reply, for gptp timestamping what could be 
other alternative so that RPI can be used.

Regards,
Deepak

pinealservo, 2017-07-17:
The i210 is a PCI-E peripheral. The Raspberry Pi does not support the PCI-E bus. The SoC does not even have an onboard ethernet interface of any kind; the one on the board is interfaced via USB.

I can't think of any reason to try to make a Raspberry Pi an AVB endpoint. It would require enough hardware external to the RPi that any cost savings from using a RPi vs. a SoC with a more appropriate peripheral set would be more than erased.

Pick up something based on an i.MX 7 instead, like this: https://www.wandboard.org/products/android-things/PICO-PI-IMX7/

closed 2017-07-17 by pinealservo

ahogen, 2017-07-17:
While a Raspberry Pi probably can't/shouldn't be used as an AVB endpoint, it could be an AVB controller (running AVDECC software) which could have its uses. Certainly not the goal here, but I wanted to make a note of it for those who find this issue because of the "raspberry pi" keywords.

See:
- http://blog.numerix-dsp.com/2014/09/controlling-avb-network-from-raspberry.html
- https://github.com/jdkoftinoff/jdksavdecc-mcu

tschiemer, 2019-08-16:
To add a note regarding RPi (3) and PTP timestamping (wether rpis are suitable or not), seemingly software based timestamping should be possible once it's enabled in the kernel:

https://github.com/twteamware/raspberrypi-ptp

andrew-elder, 2019-08-19:
AVB requires 802.1AS support, not PTP. Timestamp accuracy has to be on the order of 50 nanoseconds for 802.1AS to work correctly. This is not possible with a software timestamping solution.

pinealservo, 2019-08-19:
Summary: You can't do AVB on a Raspberry Pi 4 either.

In case anyone was wondering if the new integrated gigabit ethernet on the Rasberry Pi 4 supports 802.1AS/1588 hardware timestamping--the answer is that it does not appear to. The linux driver is `bcmemac`, which indicates a fairly old Broadcom UniMAC IP core. It's possible there's a newer version of the core with MAC-level timestamping, but as there's no driver support in the kernel for it and no datasheet, any such silicon is essentially useless even if it exists. The PHY on the board doesn't list 1588 timestamping, although some related models do, so it's a pretty sure bet there's no PHY-level timestamping either.

The Raspberry Pi 4 actually has a single PCI-E lane, but it's in use for the USB 3.0 interface to which it's hardwired. So you still can't hook an i210 up to it either.

jagmeethanspal, 2019-08-20:
And what about the transit nodes (non end-points, non-talker, non-listener)
do they require an Boundary clock or Transparent clock with H/w
time-stamping will suffice?

pinealservo, 2019-08-20:
If you are talking about the network bridges, they have to support all the time sync, stream reservation, and traffic shaping protocols specified by the AVB standards (see 802.1AS instead of IEEE1588 for time sync; they are related but not exactly the same) if you want an AVB network. Nodes that don't participate in streaming can co-exist on the network and only have to support basic Ethernet.

I am not sure what that has to do with Raspberry Pi though?

jagmeethanspal, 2019-08-20:
It was a lateral question, not RP sorry. I should've probably asked in a
separate thread.

tschiemer, 2020-10-22:
Maybe just to mention here for reference, seemingly the [Raspberry Pi Compute Module 4](https://www.raspberrypi.org/products/compute-module-4/?variant=raspberry-pi-cm4001000) has freed the occupied PCIe and also seemingly the ethernet phy has IEEE 1588v2 support (yes, not 802.1AS, but still interesting) using [bcm54210](https://www.broadcom.com/products/ethernet-connectivity/phy-and-poe/copper/gigabit/bcm54210)

Didn't yet have a deeper look, but this makes the module somewhat interesting

ValentinPintat, 2025-05-01:
Hello everyone,
I saw that the Raspi 5 does have a PCI express.
Does that means that it would be now possible to put ABV on Raspi ?