Avnu/OpenAvnu #661: Is the kernel module in this repository needed to run the /lib/igb code?

simue, 2017-07-20:
Since I am confused heavily after digging into the library-code:
Do you need to build and load the kernel module located in the _"kmod/igb"_ folder of this repository to use the functions provided in _"lib/igb/"_?

I have seen calls from the _lib/igb/igb.c_ file where registers of i210 were written (from userspace?!) directly, but also a call to ioctl with a parameter that is caught in the files of _kmod/igb/..._
So I am not sure whether you must have both or can just pull the _lib/igb_-files into your application project and start programming.

Any help is welcome! :-)

pinealservo, 2017-07-20:
Yes, the special igb driver needs to be loaded in order to use the igb library, and it does indeed allow direct control of some i210 registers from userspace. But it still has to collaborate with the driver via ioctls so that the regular kernel-driven functionality still works and so that kernel resources can be made available to the userspace element of the driver.

This is not a situation that's acceptable to the mainline kernel, which has to remain general-purpose and maintain the kernel/userspace boundaries, but it is not uncommon for highly latency-sensitive networking applications to use a custom driver or driver framework for doing user-mode network drivers. If you have a platform and application in which you can meet your performance needs without the userspace driver element, you might want to look at the various methods used by the avtp_pipeline library to get reasonable performance from the standard kernel interfaces.

Just to be clear, the igb library is only strictly necessary to access the i210's special traffic shaping support (both per-packet timed launch and the credit-based shaper); the PTP functionality is accessed through standard kernel interfaces. The avtp_pipeline code can run without the igb library on any platform with kernel PTP support, but performance may not meet your needs and you will probably have to tweak configuration files and overall system configuration to get latency down to reasonable levels.

There are several parties working on getting aspects of AVB aside from PTP into standard kernel interfaces, but progress and collaboration here seem to be frustratingly slow. If you are working on this and want to collaborate, please let us know.

simue, 2017-07-20:
Thanks for the quick reply!
Our application is such a highly latency-sensitive networking application and the main feature we would like to use is the time-sensitive send mechanism of i210.
The question is mainly in the context of analyzing already existing solutions and see which one is the best to base our work on. So I can't say anything about jumping in and collaborating in this very project right away. But since Linux and Open-Source are always an option... who knows what the future will bring? ;-)

closed 2017-07-20 by simue